zephyrproject-rtos/zephyr #48863: hawkbit subsystem - prints garbage if debug enabled and no update pending

gkujawsk, 2022-08-10:
I believe I've just hit a bug in hawkbit subsystem. When the hawkbit client debug is enabled (CONFIG_HAWKBIT_LOG_LEVEL_DBG=y)  and when there's no update pending on the hawkbit server side the debug may  print garbage to the serial console. However, this seems random behaviour and is not reproducible deterministically. 

When response_cb() function is parsing HAWKBIT_PROBE the return state is only checked for errors. json_obj_parse() expects to parse out config and _links tokens out of the server response. When there's no _links token parsed the following fields stays uninitialized: hawkbit_results.base._links.deploymentBase.href, hawkbit_results.base._links.cancelAction.href and hawkbit_results.base._links.configData.href. When hawkbit_dump_base() is called the LOG_DBG expects thet deploymentBase.href, cancelAction.href and configData.href points to valid c-string. That's possibly the reason why I get garbage on the serial console. 

Json response when there's update pending:
{"config":{"polling":{"sleep":"00:00:30"}},"_links":{"cancelAction":{"href":"https://ota.dot1.pl/default/controller/v1/qwerty/cancelAction/1"},"configData":{"href":"https://ota.dot1.pl/default/controller/v1/qwerty/configData"}}}

Json response when there's no update pending:
{"config":{"polling":{"sleep":"00:00:30"}}}

**To Reproduce**
Steps to reproduce the behavior: build hawkbit sample, do not provide any updates on hawkbit server side

**Expected behavior**
No garbage displayed when debug enabled and no update pending. However, this seems random behaviour based on particular build and is not reproducible deterministically. 

**Impact**
Possible application crash. 

**Logs and console output**
[00:03:05.146,000] <inf> hawkbit: No update found
[00:04:10.353,000] <inf> hawkbit: Polling target data from Hawkbit
[00:04:10.529,000] <dbg> hawkbit: hawkbit_dump_base: config.polling.sleep=00:01:00
[00:04:10.539,000] <dbg> hawkbit: hawkbit_dump_base: _links.deploymentBase.href=
�O�n�&���
        o��ƞ%�(uM=�����e��A����?����0Ī��)n�Ks{BW [AӟȊ0n�,�Y)�2��j��#�ٚH�Jҋ��7��y  [00:29:42.313,000] <inf> hawkbit: No update found
hawkbit>MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
Booting Zephyr OS build zephyr-v3.1.0-212-g010f61729612

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: zephyr-v3.1.0-212-g010f61729612

** Additional context **
 - custom board based on STM32F7
 - zephyr-v3.1.0-212-g010f61729612

ycsin, 2022-08-17:
Thanks for the report, sounds like this is related to #46072

mmahadevan108, 2022-09-09:
@gkujawsk, can you help test the PR and confirm it fixes the issue you reported.

gkujawsk, 2022-09-10:
@mmahadevan108  Yes. It fixed the issue.

ycsin, 2022-09-11:
> @mmahadevan108 Yes. It fixed the issue.

Could you please approve the PR if it fixes the issue?

PR #49232 (2022-09-19): mgmt/hawkbit: Fix null pointer access
Pending testing and approval(s) by issue reporter(s)

Fixes #48863