zephyrproject-rtos/zephyr #48880: BLE notifications on custom service not working anymore: <wrn> bt_gatt: Device is not subscribed to characteristic

trembel, 2022-08-10:
**Describe the bug**
Using the nRF Connect App and the appended code, whenever I try to send a notification to all connections, an error is returned by `notify_cb`. It seems that this error originates by the change in https://github.com/zephyrproject-rtos/zephyr/pull/46136. When line 2251 in subsys/bluetooth/host/gatt.c is commented out, it works again.

**To Reproduce**
Using attached service

**Expected behavior**
Notifications are being send whenever a phone has subscribed to it using the nRF Connect App

**Logs and console output**
```
[00:01:21.360,839] <wrn> bt_gatt: Device is not subscribed to characteristi
```

**Environment (please complete the following information):**
 - OS: Linux
[rssi_service.zip](https://github.com/zephyrproject-rtos/zephyr/files/9301279/rssi_service.zip)

jori-nordic, 2022-08-11:
Hi, i'm on it. I think it's the assumptions on where the CCC is located after the characteristic that don't hold true anymore for your service (which is fine, problem is on our end). The issue is in `bt_gatt_is_subscribed`. I'll try to make a fix PR asap.

jori-nordic, 2022-08-11:
@trembel can you please checkout the linked PR and check if that fixes your issue?

trembel, 2022-08-11:
@jori-nordic thanks for the fast fixing, it solves my issue.

closed 2022-08-14 by carlescufi

Aeneas32, 2023-03-06:
Can this bt_gatt message be disabled from flooding the Term output ?
When I try to disable it, I see requirements to use <choice BT_GATTP_LOG_LEVEL_CHOICE>,
which does not seem to be documented.

jori-nordic, 2023-03-06:
It is not supposed to flood the term output, unless the application is spamming notifications. You're supposed to get one message per gatt fn call. Can you post a sample of your app code, or a small description of what fns you're calling (e.g. if it's in a loop, etc). Do the notifications (where the warning is posted) reach the other device?

To get rid of the warning, you will want to change the GATT log level to `ERR`, since the message has a `WRN` severity:
set `CONFIG_BT_GATT_LOG_LEVEL_ERR=y` in your `prj.conf`.

I do see that there's one place that doesn't honor `CONFIG_BT_GATT_ENFORCE_SUBSCRIPTION=n` and that is in `gatt_notify_multiple_verify_params`. I created https://github.com/zephyrproject-rtos/zephyr/pull/55448 to remedy that.

Aeneas32, 2023-03-06:
Thanks for the offer but I am pretty sure this code I am working on Is spamming ntf's.
That will be fixed later.
CONFIG_BT_GATT_ENFORCE_SUBSCRIPTION=n  does work to generate an initial
"No buffer available to send notification" error then stops further
"not subscribed" messages -- great.
However  CONFIG_BT_GATT_LOG_LEVEL_ERR=y   in prj.conf for 2.2.0 tools and sdk 
for BLE nrf52840  results in  "Unknown symbol CONFIG_BT_GATT_LOG_LEVEL_ERR"  error
and " CONFIG_BT_GATTP_LOG_LEVEL_OFF was assigned the value y, but got the value n.
 Missing dependencies: <choice BT_GATTP_LOG_LEVEL_CHOICE> "   error.
This choice feature is not documented anywhere.

jori-nordic, 2023-03-06:
This seems like a nRF Connect SDK issue (`CONFIG_BT_GATTP` is from the sdk-nrf repo). Could you open a ticket on https://devzone.nordicsemi.com/ ?

I guess this log option is only present in NCS 2.3, which just got tagged a few days ago. It is definitely present and selectable in upstream zephyr (this repo).

I agree that the documentation for logging could be better. We have a logging macro that generates those options from the module name (ie gatt, l2cap, etc). This is used throughout zephyr; it looks like this in the Kconfig files:
```
module = BT_GATT
legacy-debug-sym = BT_DEBUG_GATT
module-str = "Bluetooth Generic Attribute Profile (GATT)"
source "subsys/bluetooth/common/Kconfig.template.log_config_bt"
```
The rule of thumb is for a given c file, look at the top, next to the `LOG_MODULE_REGISTER`, and you will likely see the kconfig that selects the log level. Then append the `_DBG`, `_INF`, `_WRN`, `_ERR` or `_OFF` to select the desired log level.

NB: when you open the ticket on devzone, please include the whole log (you can remove your confidential stuff) so we can effectively debug the issue. Setting `CONFIG_BT_DEBUG_GATT=y` will also give us more insight.

ujur007, 2023-03-16:
Was this issue solved?
I have the same issue where I have subscribed to a characteristic via phone but there is no notification sent when the characteristic status change.

AvinashKailas, 2023-03-20:
Currently, even I am facing the same issue (bt_gatt: Device is not subscribed to characteristic) this is what I am getting in nRF Terminal while working with nRF52840 DK and nRF Connect SDK version 2.3.0.
Kindly let me know how to get rid of this problem?

PR #48910 (2022-08-14): Bluetooth: gatt: allow the CCCD to be in any location
Per spec, the CCCD doesn't necessarily have to be located immediately after
the characteristic value. This commit fixes that assumption when checking
for subscriptions.

Fixes #48880.
commit 5b0ff0b3 (2022-08-14): Bluetooth: gatt: support having the CCCD in any location

Per spec, the CCCD doesn't necessarily have to be located immediately after
the characteristic value. This commit fixes that assumption when checking
for subscriptions.

Fixes #48880.