zephyrproject-rtos/zephyr #48886: Cleanup Campaigns: Documenting the process for tree wide cleanups

nashif, 2022-08-10:
How to deal with tree wide changed and changes to platforms and SoCs on a wide scale, for example in Kconfig, device tree, device structures.

The process should cover:
- Documentation of changes and rationale behind changes and if needed, the decision or vote leading to this.
- Announcement of changes to the wider community
- Dealing with tree wide changes in PRs and how we make sure PRs are reviewed by the right people and that "owners" of platforms or subsystems changed are given enough time to review and adapt to the changes.

glarsennordic, 2022-08-11:
I agree that adding some design documentation processes for cleanup campaigns would be an excellent idea.

In fact, I think Zephyr would benefit from generalized design and design documentation processes, cleanup or otherwise, and so I sort of view this issue, as well as https://github.com/zephyrproject-rtos/zephyr/issues/48887 as two sides of the same three-sided coin. As such, I'll just write my thoughts in this single comment.

I don't want to presume to advocate a specific process. I think that needs to be discussed at length with the Zephyr community.

Instead, I will just highlight a couple of case studies that I think such processes, ideally, should be designed to prevent.

First, some important caveats: I have tried not to use language that is too strong or demanding in the below, but just in case I failed at that, I'd like to acknowledge a couple of things:

1. These are just **_my_** opinions, and I do not presume them to be universally correct. Instead, I simply hope they will be useful for discussion
2. I do not presume to represent the needs and interests of all of Zephyr. I expect others here will have differing opinions, and I hope that they share those opinions too

### `log_strdup` and the V2 logging API

Recently, the V1 logging API was deprecated, making the V2 logging API One of the nice things about the new API is that transient strings no longer need to be wrapped in `log_strdup`.

This is very convenient, but it is totally unclear how this works, or under what circumstances it works. Regrettably, these details were never documented on the Zephyr logging API docs (AFAICT).

It turns out that there **_is_** a rare circumstance under which it fails (when deferred logging is active, and one tries to print `uint_8[]` or other non-char string types), which I accidentally encountered. The only reason I figured it out at all was thanks to this issue: https://github.com/zephyrproject-rtos/zephyr/issues/44996 which, mercifully, the author decided to leave open for exactly that reason.

To me, this is a pretty important caveat to our new logging system for it not to be documented. Ideally, whatever system we arrive at would have at least reasonable odds of preventing situations like this

### Removal of Device Tree node label props

Missing documentation for key features is not the only thing we should aim to prevent, though, IMO. On a much broader scale, we lack systems by which to track, democratize, and document the underlying design decisions that drive Zephyr.

For a recent example of this, see the recent decision to remove, or at least stop relying upon, node label props and the Device Tree macros that relied upon them. 

I don't personally have an opinion on the design decision itself; I frankly don't understand Device Tree well enough to comment. What I find concerning is that I only ever heard about this design change by word of mouth, long after significant portions of the NCS and Zephyr codebases had been refactored to accommodate it.

None of the PRs affecting the changes that reflect this design decision had any kind of metadata linking them to the original design decision. Without that metadata, it was impossible for me to read and understand the design rationale behind it, and to this day it is still unclear to me precisely why the decision was made, and what impact that might have on I should approach interacting with Device Tree. This makes it very difficult to ensure the integrity of the code that I produce. I genuinely desire to write code that interfaces with Device Tree correctly, but without understanding the design decisions that yielded its current API, how can I do that?

It turns out that this design decision was, apparently, discussed on the Zephyr discord, and this discussion was apparently somewhat scattered between several channels. Furthermore, PRs were merged implementing the design decision without a PR for documenting it first having been merged, so even bleeding-edge docs were not sufficient to understand the new methodology.

In short, this incident highlighted the following potential issues with our current processes:
 
1. We do not have a process that ensures public, searchable, archived, and community-inclusive discussion of large-scale design decisions
1. We have no way of linking PRs and their commits for large-scale design decisions to the underlying design decision that inspired them
1. We do not have any kind of guarantees that large-scale design decisions are documented before their changes are merged
1. We do not have any kind of system by which large-scale design decisions which must be implemented over the course of multiple PRs are guaranteed to be grouped and organized.

Some scattered thoughts on each of these problems:

#### No public discussion process
Discord is a fantastic platform for live communication, and I understand its important role in the Zephyr development community, but once a major design goal has been identified, it is no longer appropriate (in my opinion) to continue to have that discussion on Discord.

This is for a variety of reasons:
1. Discord conversations tend to be scattered, and are hard to follow or fully track down, possibly even occurring across multiple channels, and interspersed by other conversations over great periods of time.
1. Discord conversations require a discord account to observe. They are, therefore, not truly public.
1. Discord conversations are not archived
[... cut ...]
I think that we should require that large-scale design changes include documentation commits alongside implementation commits. For large-scale decisions that need to be split across multiple PRs, there should be a documentation PR, and it should, at a minimum, be in rough-draft stage before any of the related implementation PRs are merged, and all of these PRs should be required to link to each other.

#### Multi-PR design decisions are not always grouped
Sometimes, multi-PR design changes do not successfully cross-link their PRs / commits. I think we should require this, somehow, as part of the review process.

gmarull, 2022-08-12:
One thing I'd really like is to receive a weekly (maybe monthly or bi-weekly) mail with Zephyr development highlights, collected from activity in meetings and PRs/issues. I'd expect content like:

- Ongoing discussions about multi-API devices
- Removal of label property (with a summary on why this is done, and what is the way forward)
- Introduction of Kconfig DT helpers, move to select vs. depends
- Any new API proposals
- etc.

Same as we have release managers, we could have a person assigned to do this on each release cycle. Nowadays, one needs to be quite involved in the project to follow what is going on.

mbolivar-nordic, 2022-08-12:
> edit: I heard @mbolivar-nordic started doing something like that, but I can't find the links.

The links are in the mailing lists if you look, but I stopped doing it because it was simply too much work as the project scaled up. Maybe this is something that the new community manager can take up, but manually scanning the commits like I did when I wrote those newsletters won't work anymore; it's too much to do. Rather I think the community manager should work with subsystem maintainers to ask them what sort of big changes are coming down the pipe, to spread out the effort.

galak, 2022-08-12:
I'd like to keep the scope to tree wide cleanup changes.

Some initial thoughts
* Email communication on dev list before changes are started to communicate what is going on, why, etc.  Similar to a breaking API change communication
* how to balance getting such changes done in a timely manor

mbolivar-nordic, 2022-08-12:
> Email communication on dev list before changes are started to communicate what is going on, why, etc. Similar to a breaking API change communication

Email is fine for discussion, but I think we need an RFC or enhancement issue too if we're going to address @glarsennordic 's points:

>  linking PRs and their commits for large-scale design decisions to the underlying design decision that inspired them

Solve this by linking to the main GH issue from each PR

> system by which large-scale design decisions which must be implemented over the course of multiple PRs are guaranteed to be grouped and organized

Solve this with e.g. a checklist in the GH issue description describing subtasks and linking to PRs

In addition, email tends to wander as discussion evovles, and we can't edit it. A GH issue description can be updated as we refine designs.

nashif, 2022-08-12:
> * doc such changes in release notes

Yes, we need those, but in some cases this is too late to be serving as a communication to anyone tracking the development tree.

henrikbrixandersen, 2022-08-14:
I think requiring a RFC umbrella issue for tree-wide changes sounds like a good idea. It’s a well-known process, doesn’t impose much overhead, allows for linking from PRs, and still allows us to get the needed changes in at a fast pace.

gregshue, 2022-08-15:
Interesting thread.  Here are a few thoughts that came to my mind:

- the process of applying a "tree-wide change" need to be appropriate to apply to the all code following the Zephyr directory and/or code organization, no matter which repository it exists in (including downstream, non-"zephyrproject-rtos" modules and forks of modules).
- Once user needs and system requirements are captured, design decisions and justification for each will have to trace back to higher level requirements. This will change whatever process we come up with.  How big do you want that change to be?

mbolivar-nordic, 2022-08-15:
> need to be appropriate

"Need" is too strong for my taste. You can do whatever you want in your own repositories if you have a good reason to, and upstream really shouldn't care what that is, beyond providing good examples with rationales when possible.

> Once user needs and system requirements are captured, design decisions and justification for each will have to trace back to higher level requirements. This will change whatever process we come up with. How big do you want that change to be?

IMO we need to start small and iterate this process until people are happy with it, not try to do a big design up front for how we're going to do requirements traceability, et cetera. Zephyr moves too quickly in some areas for this to be practical, I believe.

gregshue, 2022-08-15:
> You can do whatever you want in your own repositories if you have a good reason to

I can do whatever I want in my own repositories, whether or not I have a good reason to.  

In this particular case, I want to **reuse** Zephyr open source and licensed modules and **efficiently** apply all the tools, patterns, and practices of upstream Zephyr to my private modules and patches to non-private modules.  I may not be able to **efficiently** do that unless the transformation process is described. 

> upstream really shouldn't care what [is done in downstream repositories], beyond providing good examples with rationales when possible.

Actually, it should care (as indicated by the feedback from downstream users).  In fact, it already must care, because `example-application` is a repository downstream of the Zephyr repo.

> not try to do a big design up front for how we're going to do requirements traceability, et cetera.

No one has asked for or implied that here.

mbolivar-nordic, 2022-08-17:
Process WG:

Rough consensus on:

1. Person making the change must communicate it via an RFC issue on GitHub
2. Process WG recommends Architecture meeting as default place for go/no-go decisions on tree wide cleanups, with escalation to TSC if consensus cannot be reached
3. Open, but email to devel is a likely minimum; questions remain about tracking at a project

----------------

- @galak 3 questions: 1. person making the change needs to communicate it. 2. how do we decide to approve or deny this cleanup? 3. what is the mechanism we use to tell developers/maintainers about tree-wide cleanups that everyone needs to know?
- @galak there's been a few such changes in this release. E.g. I cleaned up a lot of devicetrees, changing conventions and deprecating things. This opens a broader question of communication mechanisms and documentation that needs to happen in a more standardized way to manage things like this. We want to find a balanced process to manage the conversation about changes while still being able to make them in a timely manner.
- @galak looking back, we've gone through "driver reworks" (pinmux -> pinctrl as an example) with an issue that documents the change and tracks changes. 
- @mbolivar-nordic consensus in the thread seems to be than RFC issue is required for changes like this
- @dleach02 Agreed
- @gregshue Fine for short term, but not in the long term. Eventually this needs to be tracked in requirements outside of GH issues.
- @mbolivar-nordic the question then becomes if we need RFC format changes, right?
- @galak yeah -- @gmarull mentioned having a 'zephyr weekly news' being something that would be nice in addition. So having an RFC makes sense, but then is there some other way to make developers aware of these big changes without constantly polling?
- @stephanosio how about announcement emails?
- @mmahadevan108 what about adding them to the release milestones (https://github.com/zephyrproject-rtos/zephyr/projects/9#column-17808165)?
- @stephanosio people aren't aware of that area
- @mbolivar-nordic that's fixable though
- @dleach02 we used to review this project almost every week. It's not a bad thing that we have it; we just got out of practice of reviewing them in the triage meeting
- @MaureenHelm my early vision for this project was a preview of the release notes -- high level features planned for the release. I think @mmahadevan108 's plan is a good one.
- @stephanosio makes sense, but people still have to poll that. Should send an email or discord announcement
- @galak maybe whenever there's an addition to the release plan, send an email. Not sure when approval happens / how things get added or removed. Maybe when things get added or removed we can send an announcement.
- @mbolivar-nordic not sure if we have a process for managing each release's swim lane in that project though
- @dleach02 it was ad hoc, but was managed by the release team previously. Want to be able to ask developers if they're really going to make it. Think you should leave it up to the release managers.
- @mbolivar-nordic so should release managers be responsible for sending announcements when milestones are added/removed?
- @dleach02 question about when community does things, e.g. pinctrl, have been decided in different meetings. Same thing for cleanups, just pull it into an already established meeting.
- @mbolivar-nordic so pick a meeting to decide when these RFCs are agreed upon or punted?
- @galak: would like to put a pin in that for now -- want to close on communication. How do we communicate these changes? We discussed adding them to the release milestones, which I think makes sense. Does it then make sense to add a responsibility for the release managers to communicate changes to the release plan?
- @dleach02 find it hard to discuss this idea since the community has disparate people working on their own goals; people work on things without involving the release plan at all.
- @galak I'm OK if it's not using the release plan, but I want to know how I, as a developer, should know that I shouldn't use DT label properties etc.
- @stephanosio should be checking devel email

- @fabiobaltieri It would be cool to have a presubmit where we could somehow find these changes. E.g. for DTS, some check to make sure the label property deprecation is flagged in a PR. If it's not automated, reviewers will have to do it manually.
- @dleach02 Deprecations in particular need documentation that there's a new standard, and a CI check flagging new uses.
- @fabiobaltieri I think we have a gap for that for DT and Kconfig. E.g. can we deprecate a Kconfig variable and have a warning?
- @galak I think @tejlmand added something
- @fabiobaltieri the other problem is that checks are going to have to be perfect since they will block PRs. I think we are lacking something between something that can be skipped by developers and something that is a hard failure.
- @stephanosio PRs to upstream should always be blocking
- @galak we need something "less than" deprecated. The issue is we have to clean up the whole tree before marking something as deprecated, or we break CI. E.g. there are DT labels in tree; if I mark it deprecated, existing labels fail CI. Need a "deprecated for new".
- @fabiobaltieri we have linters on our code base that can trigger on newly introduced or changed code; not sure how it works.
- @MaureenHelm do we fail CI on DT warnings?
[... cut ...]
- @dleach02 TSC is always there as the final arbitrator, but we want the community to be able to make decisions. I don't think option 1 is "take it to the TSC to vote". I don't want to see a litany of these changes coming to the TSC; it's the last option
- [.... discussion ...]
- Rough consensus on recommending architecture meeting as a place to decide go vs no-go on tree-wide cleanups that force most developers into changing their behavior. Escalation to the TSC always remains possible as with any other technical question.

galak, 2022-08-22:
> Open, but email to devel is a likely minimum; questions remain about tracking at a project

I do think a label for issues, PRs would be useful and is pretty low hanging fruit.  Not saying this is the ONLY means of communication, but one that should be used.

mbolivar-nordic, 2022-08-24:
Process WG recommendations:

0. Change "cleanup" term to "treewide"
1. The person proposing a treewide change shall write an RFC issue
2. The architecture WG shall include the issue on the agenda and discuss whether the project will accept or reject the cleanup, with escalation to the TSC if consensus is not reached
3. The person proposing a treewide change shall email devel about the RFC
4. Release engineers shall not merge pull requests for treewide changes until the community has has time to review (e.g. 4 days)
5. The project GitHub shall add a 'treewide' label for use on issues and PRs that are treewide cleanups

mbolivar-nordic, 2022-08-24:
@galak since you brought this up, would be good to get your ACK that you are happy with the above

galak, 2022-08-24:
> Process WG recommendations:
>
>     0. Change "cleanup" term to "treewide"
>
>     1. The person proposing a treewide change shall write an RFC issue
>
>     2. The architecture WG shall include the issue on [...]

When, after approval or before?  Is this to communicate "asking for input" or "here's the change, be aware of it"?
> 
>     4. Release engineers shall not merge pull requests for treewide changes until the community has has time to review (e.g. 4 days)

I have some concern on this one.  I think this should only be for cases in which its done in a single PR.  Once the arch group approves a change, it might be handled in multiple PRs over time as the cleanup might NOT need to occur atomically.

> 
>     5. The project GitHub shall add a 'treewide' label for use on issues and PRs that are treewide cleanups

0, 1, 2, and 5 are good. with me.  Need some clarification on 3 & 4.

mbolivar-nordic, 2022-08-25:
> When, after approval or before? Is this to communicate "asking for input" or "here's the change, be aware of it"?

Good question -- do you have a preference?

> I have some concern on this one. I think this should only be for cases in which its done in a single PR. Once the arch group approves a change, it might be handled in multiple PRs over time as the cleanup might NOT need to occur atomically.

That's a fair point. How about using this language instead:

> Release engineers should use good judgement in deciding when to merge approved treewide changes, with the goal of giving the community enough time for review.

mbolivar-nordic, 2022-09-28:
Process WG:

> When, after approval or before? Is this to communicate "asking for input" or "here's the change, be aware of it"?

After approval in the Architecture WG.

> 4. Release engineers shall not merge pull requests for treewide changes until the community has has time to review (e.g. 4 days)

Change to: "the architecture working group shall define what the merge process will be for a particular treewide change"

---------------

- @nashif: we do need approvers to review
- @fabiobaltieri: current 4 days/2 days/etc are hard to understand. If we want to enforce, maybe we should have the bot enforce this with a DNM tag being applied.
- @galak : is there a need for CI / a bot make sure that individual required approvers approve before merge? Or if nobody is available to be assigned, e.g. an unmaintained board, we shouldn't block the PR on those.
- @nashif there has to be judgement calls applied. But because anyone with merge access can merge with +2 and CI easily, having the treewide label is a hint that we need to ping the relevant people or let things sit for longer because the assignee is unresponsive.
- @galak I'm thinking of automation more generally
- @stephanosio you're talking about getting the assignee to approve?
- @galak right
- @nashif we do need to preserve override power due to nonresponsive assignees. But also need to review the MAINTAINERS file -> GitHub assignee process; there are cases where the scope of the MAINTAINERS file is being applied too broadly (e.g. every kernel change goes to @andyross, every doc change goes to @carlescufi , every DT binding goes to @mbolivar-nordic )
- @fabiobaltieri I'm also thinking that if APIs change, it's not uncommon that there are changes throughout the codebase, so we'd need approval from all the owners. How does Linux handle this?
- @galak real API changes are most likely done in a single PR, so you'll have a ton of reviewers and the assignee needs to be whatever subsystem owner is involved in the treewide change.
- @nashif we don't have processes for updating tests/docs when we change an API. This can apply even to cases of adding APIs.
- @mbolivar-nordic looks like the problem is that we don't have a process for identifying the required approvers, right?
- @stephanosio could use maintainers script and select multiple assignees
- @mbolivar-nordic if that's the approach, we need to fix the issues in the script identified above -- so are we talking about sub-areas?
- @stephanosio could be simpler than that; if we have some sort of specificity metric in MAINTAINERS, we can try to pick assignees from "most specific" areas
- @nashif that's something that started happening in a recent change from Gerard. Documentation was split into infrastructure and general guidelines. Everything else needs to be part of the respective subsystem. E.g. if maintaining I2C, I'm responsible for i2c.rst. If I'm maintaining Bluetooth, it's part of my responsibility to review Bluetooth documentation. Another issue is that some of us have accumulated a lot of areas over the years; I'm looking at giving away some areas. For example testing is mostly driven by Yuval and the Google team working on the new ztest framework, which could be assigned to him or someone on his team. Need to fix the MAINTAINER file and have reviewers speak up.
- @stephanosio I think people refrain from adding themselves to MAINTAINERS since we don't have "sub-areas" for things that are e.g. single files in lib/os.
- @mbolivar-nordic so for the process requirement, it would remain "all assignees must approve". We will fix the MAINTAINERS issues in parallel; in the meantime, mergers continue to exercise caution on a best effort basis.
- @dkalowsk I think that would slow you down
- @mbolivar-nordic assignees are different than reviewers though
- @nashif I don't think it makes sense to pick multiple assignees; having people exercise caution remains
- @stephanosio that's not a well defined process, though.
- @nashif other problems exist, like board owners not having triage rights or not being able to be assigned as a reviewer
- @mbolivar-nordic seems like we may need to drop this requirement since we can't converge
- @nashif not sure about that; we could drop from 4 days to 2 days, for example.
- @stephanosio that's kind of random -- if you only have 1 day per week to work on zephyr, that might not work
- @nashif we want to move fast as a project, though; we want to avoid merging without reviewers. I could even be OK with 1 day
- @fabiobaltieri might as well keep the '2 days' or 'couple of days' requirement for standard patches
- @galak I go back to the scope of this. If I'm doing a treewide change that is just to move a Kconfig symbol to DT_HAS_FOO, waiting 2 days for every one of those PRs doesn't make sense to me. If we use language "the first PR shall not be merged for 4 days" then that's OK with me, since it allows merging trivial incremental progress towards a treewide change. Subsequent PRs follow the normal process.
- @galak for example, removal of `soc.h` would go through this entire process, where we have an RFC, a meeting, etc. -- ample opportunity for you to find out about the change.
- @nashif the question for me is why do we think anyone has the authority to remove `soc.h`. In my view we should be telling people if we approve that, the individual owners should be responsible for that.
- @galak I think anybody who's willing to do the change should be able to, not just soc maintainers
[... cut ...]
- @galak at times, a number of things may depend on other PRs. If it's one big monolithic PR, I have no issue. It's specifically about the case where we can do things bit by bit, and we've already agreed to do the change. If the first PR waits a few days, then fine, but why should subsequent PRs have to wait?
- @nashif I think we can solve this in the review process 
- @galak I'm fine with this if we want the wording to be "the architecture working group defines what the merge process will be for a particular treewide change"

mbolivar-nordic, 2022-09-28:
Next steps:

@mbolivar-nordic to clean up the language for the final proposal and go for TSC approval. If approved, I can update the documentation

mbolivar-nordic, 2022-09-28:
Final language:

- This process covers how to make "treewide" changes. The project GitHub shall add a 'treewide' label for use on issues and PRs that are treewide changes.
- The person proposing a treewide change shall write an RFC issue describing the change.
- The architecture WG shall include the issue on the agenda and discuss whether the project will accept or reject the cleanup, with escalation to the TSC if consensus is not reached.
- The person proposing a treewide change shall email devel about the RFC if it is accepted by the architecture WG.
- The architecture WG shall specify the procedure for merging any PRs associated with each treewide change.

mbolivar-nordic, 2022-11-02:
Approved at the TSC. Pending  documentation merge. Closing.

closed 2022-11-02 by mbolivar-nordic