zephyrproject-rtos/zephyr #48888: Deprecate `emul_get_binding`

alevkoy, 2022-08-10:
**Is your enhancement proposal related to a problem? Please describe.**
`EMUL_DT_GET` will provide a mechanism to statically get a reference to a `struct emul` based on the device tree. Generally, it should be possible and preferable to get emulator references statically. `emul_get_binding` currently provides a mechanism to get them dynamically.

**Describe the solution you'd like**
Deprecate `emul_get_binding`. Provide a function `struct emul *emul_get_device(struct device *)`, which will dynamically get the emulator corresponding to a specified device.

**Describe alternatives you've considered**
* Continue to use `emul_get_binding`: This is potentially time-consuming and can fail at runtime.
* Something about not having `emul_get_device`, to be elaborated by @galak.

**Additional context**
See request from @galak in https://github.com/zephyrproject-rtos/zephyr/pull/44883#issuecomment-1209981429.

@galak Did I miss anything? Is that what you were thinking for `emul_get_device`? What is the use case for `emul_get_device`?

galak, 2022-08-17:
> @galak Can you flesh out how you wanted `emul_get_device` to work?

I think something like:

```
const struct emul *emul_get_dev(const struct device *dev)
{
	const struct emul *emul_it;

	for (emul_it = __emul_list_start; emul_it < __emul_list_end; emul_it++) {
		if (emul_it->dev == dev) {
			return emul_it;
		}
	}

	return NULL;
}
```

galak, 2022-08-17:
I think we can mark `emul_get_binding` deprecated now.  Its only if you guys need a runtime mechanism do we need to add `emul_get_dev`

alevkoy, 2022-08-18:
@yperess I think you had thoughts on this previously, but I can't remember what they were.

yperess, 2022-08-18:
> @yperess I think you had thoughts on this previously, but I can't remember what they were.

I'm good with this, had a chat with Kumar offline and we have a good path forward if we end up needing a dynamic lookup later on