zephyrproject-rtos/zephyr #48911: sample.drivers.flash.shell: Failed on atmel targets

erwango, 2022-08-11:
**Describe the bug**
Seen here: https://github.com/zephyrproject-rtos/zephyr/runs/7769981753?check_suite_focus=true

`samples/drivers/flash_shell/sample.drivers.flash.shell ` is failed on following targets:
- sam4s_xplained
- sam4l_ek
- sam4e_xpro
- arduino_due

**Log Error**
```
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_get_page':
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:88:18: error: 'IFLASH_PAGE_SIZE' undeclared (first use in this function); did you mean 'IFLASH0_PAGE_SIZE'?
   88 |  return offset / IFLASH_PAGE_SIZE;
      |                  ^~~~~~~~~~~~~~~~
      |                  IFLASH0_PAGE_SIZE
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:88:18: note: each undeclared identifier is reported only once for each function it appears in
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_wait_ready':
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:108:13: error: 'EEFC_FSR_FLERR' undeclared (first use in this function); did you mean 'EEFC_FSR_FRDY'?
  108 |   if (fsr & EEFC_FSR_FLERR) {
      |             ^~~~~~~~~~~~~~
      |             EEFC_FSR_FRDY
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_write':
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:204:27: error: 'IFLASH_PAGE_SIZE' undeclared (first use in this function); did you mean 'IFLASH0_PAGE_SIZE'?
  204 |    eop_len = -(offset | ~(IFLASH_PAGE_SIZE - 1));
      |                           ^~~~~~~~~~~~~~~~
      |                           IFLASH0_PAGE_SIZE
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_erase_block':
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:255:4: error: 'EEFC_FCR_FCMD_EPA' undeclared (first use in this function); did you mean 'EEFC_FCR_FCMD_EA'?
  255 |    EEFC_FCR_FCMD_EPA;
      |    ^~~~~~~~~~~~~~~~~
      |    EEFC_FCR_FCMD_EA
In file included from /local/mcu/zephyrproject/zephyr/include/zephyr/device.h:29,
                 from /local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c:13:
/local/mcu/zephyrproject/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_erase':
/local/mcu/zephyrproject/zephyr/twister-out/arduino_due/samples/drivers/flash_shell/sample.drivers.flash.shell/zephyr/include/generated/devicetree_unfixed.h:2883:34: error: 'DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000_P_erase_block_size' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000_P_write_block_size'?
 2883 | #define DT_N_INST_0_soc_nv_flash DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

**To Reproduce**
`twister -b -s samples/drivers/flash_shell/sample.drivers.flash.shell -p sam4s_xplained`

**Impact**
Blocking CI

**Environment (please complete the following information):**
zephyr-v3.1.0-3289-gc93361a5bf

nandojve, 2022-08-11:
Possible duplicated: #48516
PR to Fix: #48762

erwango, 2022-08-11:
I confirm #48762 fixes current issue

erwango, 2022-08-12:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/48762

closed 2022-08-12 by erwango