zephyrproject-rtos/zephyr #48913: net: Add pointer member to net_mgmt_event_callback struct to pass user data to the event handler.

savinweera, 2022-08-11:
**Is your enhancement proposal related to a problem? Please describe.**
We are porting an application to Zephyr which needs access to network change events. Our existing codebase is structured to be as OS/platform agnostic as possible, so we want to avoid injecting a new network event handler function into that code which conforms to the Zephyr specific net_mgmt_event_callback template.
Instead we would like to implement the Zephyr specific net_mgmt_event_callback function in a Zephyr specific code layer which then calls our existing handlers.

**Describe the solution you'd like**
We would like to pass a function pointer to the network management event handler through the net_mgmt_event_callback structure. This requires the addition of a pointer member (e.g. void * user) to the net_mgmt_event_callback struct.

**Describe alternatives you've considered**
We have attempted to use the info and info_length members in the net_mgmt_event_callback struct, enabled using the CONFIG_NET_MGMT_EVENT_INFO option. However, these members are updated internally in Zephyr with information related to the specific network event.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

jukkar, 2022-08-11:
> We would like to pass a function pointer to the network management event handler through the net_mgmt_event_callback structure. This requires the addition of a pointer member (e.g. void * user) to the net_mgmt_event_callback struct.

I suppose that could be done, for example change `net_mgmt_init_event_callback()` and add a `user_data` pointer at the end of the signature. And then pass that pointer to the event handler when the event happens. This would be an API change, so would need to be documented properly.

mniestroj, 2022-08-11:
@savinweera Did you try to embed `struct net_mgmt_event_callback` into your own structure? That way you could access your data similar to:
```
struct my_data {
	struct net_mgmt_event_callback mgmt_cb;
	void *user_data;
};

static void my_handler(struct net_mgmt_event_callback *cb,
		       uint32_t mgmt_event, struct net_if *iface)
{
	struct my_data *data =
		CONTAINER_OF(cb, struct my_data, mgmt_cb);
	void *user_data = data->user_data;
}

static struct my_data my_data;

static void some_other_function(void)
{
	struct my_data *data = &my_data;

	net_mgmt_init_event_callback(&data->mgmt_cb,
				     my_handler,
				     NET_EVENT_L4_CONNECTED |
				     NET_EVENT_L4_DISCONNECTED);
}
```

savinweera, 2022-08-12:
@mniestroj We did consider this as an alternative.
In our use case, the `some_other_function` will be accessed by multiple threads to receive notifications. So this would require making access to `static struct my_data mydata` thread-safe. It is a possible solution, however we felt that having a `user_data` pointer as part of `struct net_mgmt_event_callback` would be a cleaner solution.

```
struct my_data {
	struct net_mgmt_event_callback mgmt_cb;
	my_handler_fn *fn;
};

static void net_mgmt_event_handler(struct net_mgmt_event_callback *cb,
				uint32_t mgmt_event, struct net_if *iface)
{
	struct my_data *data = cb->user_data;
	data->fn(.., data);
}

void some_other_function(struct my_data ** pdata, my_handler_fn * fn)
{
	struct my_data *data = (struct my_data *) malloc(sizeof(struct my_data));
	data->fn = fn;
	data->mgmt_cb.user_data = (void *)data;
	net_mgmt_init_event_callback(&data->mgmt_cb,
						my_handler,
						NET_EVENT_L4_CONNECTED |
						NET_EVENT_L4_DISCONNECTED);
	net_mgmt_add_event_callback(&data->mgmt_cb);
	*pdata = data;
}
```

mniestroj, 2022-08-12:
>In our use case, the some_other_function will be accessed by multiple threads to receive notifications. So this would require making access to static struct my_data mydata thread-safe.

Does the fact about being thread-safe or thread-unsafe change anything? I just don't see how it affects the need for `user_data`.

Could you elaborate on the example and show how your code would look like if `user_data` is introduced and how would it be impossible/harder to implement with `CONTAINER_OF` approach? From what I can see you could just replace `net_mgmt_event_handler()` body with:
```
static void net_mgmt_event_handler(struct net_mgmt_event_callback *cb,
				uint32_t mgmt_event, struct net_if *iface)
{
	struct my_data *data =
		CONTAINER_OF(cb, struct my_data, mgmt_cb);
	data->fn(.., data);
}
```
and it would work right away. Additionally you could drop `data->mgmt_cb.user_data = (void *)data;` line from `some_other_function()`. Is there something I am missing still?

I'm asking because of 2 main reasons:
1. I was pretty happy with embedding `struct net_mgmt_event_callback mgmt_cb;` into my structs and using `CONTAINER_OF` for several projects in the past, never run into the need for `user_data`,
2. adding `user_data` to `struct net_mgmt_event_callback mgmt_cb;` would increase its size from 12 to 16 bytes, which isn't that bad in total, but in % it makes a difference already; starts to be noticeable with lots of events in the system, which is a must if you don't  want to lose any information (e.g. when scanning for WiFi networks, each discovered WiFi network is a new event).

jukkar, 2022-08-12:
@mniestroj proposal should work fine, bonus points that we avoid embedding a pointer to net_mgmt struct that most users probably might not need.

savinweera, 2022-08-17:
@mniestroj We've tested out your proposed solution and it works for us. Thanks

closed 2022-08-17 by savinweera