zephyrproject-rtos/zephyr #48950: west: string quotes are removed from extra_kconfig_options.conf

xudongzheng, 2022-08-11:
**Describe the bug**

When string configuration options are passed via `west build` and `prj.conf` changes, the necessary quotes get removed from 
`build/zephyr/misc/generated/extra_kconfig_options.conf`.

**To Reproduce**

```
cd samples/bluetooth/peripheral_hids
west build -b nrf52840dk_nrf52840 -- -D'CONFIG_BT_DEVICE_NAME="hello world"'

echo >> src/main.c
west build # builds correctly

echo >> prj.conf
west build # no longer builds
```

This results in the following error message:

```
zephyr/samples/bluetooth/peripheral_hids/build/zephyr/misc/generated/extra_kconfig_options.conf:2: warning: malformed string literal in assignment to BT_DEVICE_NAME (defined at subsys/bluetooth/host/Kconfig:579). Assignment ignored.
```

**Expected behavior**
I would expect the string quotes to remain `extra_kconfig_options.conf`.

**Impact**
Minor annoyance as it requires the `build` directory to be manually deleted.

mbolivar-nordic, 2022-08-17:
@xudongzheng thank you for the detailed report.

Honestly, I initially thought this must be user error, but I can reproduce and I think we should adjust priority and reassign.

With this patch:

```
--- a/cmake/modules/kconfig.cmake
+++ b/cmake/modules/kconfig.cmake
@@ -182,6 +182,10 @@ foreach (name ${cache_variable_names})
   if("${name}" MATCHES "^${KCONFIG_NAMESPACE}_")
     # When a cache variable starts with the 'KCONFIG_NAMESPACE' value, it is
     # assumed to be a Kconfig symbol assignment from the CMake command line.
+    message(STATUS "
+
+    eval(name)=${name} eval(eval(name))=${${name}}
+")
     set(EXTRA_KCONFIG_OPTIONS
       "${EXTRA_KCONFIG_OPTIONS}\n${name}=${${name}}"
       )
```

I see the following on the first `west build`:

```
eval(name)=CONFIG_BT_DEVICE_NAME eval(eval(name))="hello world"
```

and this after touching prj.conf and re-building:

```
eval(name)=CONFIG_BT_DEVICE_NAME eval(eval(name))=hello world
```

Further, in build/CMakeCache.txt, I see:

```
CONFIG_BT_DEVICE_NAME:STRING=hello world
```

so for some reason, cmake itself seems to be:

- stripping the quotes from the string literal in the cache file
- preserving the quotes in memory for the first cmake run
- using the cache file in subsequent re-runs, which has discarded the quotes

I am tempted to call this a bug in cmake itself, but let's call it high for now and see if there is a known fix. I'm surprised no one has seen this before. We may need to document this as a limitation if there's no way around this. Nothing I tried seems to force cmake to keep the quotes in the cache file.

nordicjm, 2022-08-17:
> I am tempted to call this a bug in cmake itself, but let's call it high for now and see if there is a known fix. I'm surprised no one has seen this before. We may need to document this as a [...]

It's been known for a long time, I've certainly seen it for as long as I can remember. It becomes very annoying with sysbuild however because any change to via menuconfig will break the project and you won't be able to build or configure it again, you have to delete the whole build folder and start over. This was very much a hacky bit of code I threw together to work around the issue in cmake/modules/kconfig.cmake on line 182:
```
# Support assigning Kconfig symbols on the command-line with CMake
# cache variables prefixed according to the Kconfig namespace.
# This feature is experimental and undocumented until it has undergone more
# user-testing.
unset(EXTRA_KCONFIG_OPTIONS)
get_cmake_property(cache_variable_names CACHE_VARIABLES)
foreach (name ${cache_variable_names})
  if("${name}" MATCHES "^${KCONFIG_NAMESPACE}_")
get_property(result CACHE ${name} PROPERTY TYPE)
message(STATUS "${name} = ${${name}} = ${result}")
if(${result} MATCHES "STRING")
if(${${name}} MATCHES "y" OR ${${name}} MATCHES "n")
    set(EXTRA_KCONFIG_OPTIONS
      "${EXTRA_KCONFIG_OPTIONS}\n${name}=${${name}}"
      )
else()
string(REPLACE "\"" "" varout ${${name}})
    set(EXTRA_KCONFIG_OPTIONS
      "${EXTRA_KCONFIG_OPTIONS}\n${name}=\"${varout}\""
      )
endif()
else()
    set(EXTRA_KCONFIG_OPTIONS
      "${EXTRA_KCONFIG_OPTIONS}\n${name}=${${name}}"
      )
endif()
    # When a cache variable starts with the 'KCONFIG_NAMESPACE' value, it is
    # assumed to be a Kconfig symbol assignment from the CMake command line.
  endif()
endforeach()
```

mbolivar-nordic, 2022-08-18:
Is there some way that we can ask kconfiglib if a particular option has string type at cmake time, before we generate extra_kconfig_options.conf? If so, then we could manually add quotes each time they are missing.

tejlmand, 2022-08-19:
@xudongzheng please find a fix here: https://github.com/zephyrproject-rtos/zephyr/pull/49267

closed 2022-08-26 by mbolivar-nordic

Yaxit, 2024-08-23:
This seems to still be an issue on Windows?
`west build -b nrf5340dk/nrf5340/cpuapp ..\zephyr\samples\hello_world\ -d build_hello -- '-DCONFIG_BOOTLOADER_MCUBOOT=y' '-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE="bootloader/mcuboot/root-rsa-2048.pem"'`

gives Kconfig malformed string literal warning, where the command is stripped of the quotes

`FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/andre/.pyenv/pyenv-win/versions/3.9.13/python.exe '-BC:\Users\andre\ws_samples\sample_usbfull\build_hello' -GNinja -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=bootloader/mcuboot/root-rsa-2048.pem '-SC:\Users\andre\ws_samples\zephyr\share\sysbuild' '-DAPP_DIR:PATH=C:\Users\andre\ws_samples\zephyr\samples\hello_world'`

PR #49267 (2022-08-26): cmake: kconfig: preserved quotes for Kconfig string values
Fixes: #48950

Kconfig requires quoted strings in its configuration files, like this:
> CONFIG_A_STRING="foo bar"

But CMake requires that strings are without additional qoutes,
and therefore quotes are stripped when loading Kconfig config filers
into CMake.

This is particular important when the string in Kconfig is a path to a
file. In this case, not stripping the quotes leads to an error as the
file cannot be found.

When users pass a string to Kconfig through CMake, they are expected to
pass it so that qoutes are correct seen from  [... cut]
commit 4cb2ef83 (2022-08-26): cmake: kconfig: preserved quotes for Kconfig string values

Fixes: #48950

Kconfig requires quoted strings in its configuration files, like this:
> CONFIG_A_STRING="foo bar"

But CMake requires expects that strings are without additional qoutes,
and therefore qoutes are stripped when loading Kconfig config filers
into CMake.

This is particular important when the string in Kconfig is a path to a
file. In this case, not stripping the quotes leads to an error as the
file cannot be found.

When users pass a string to Kconfig through CMake, they are expected to
pass it so that qoutes are correct s [... cut]