zephyrproject-rtos/zephyr #48977: kernel: mem_protect: mimxrt11xx series build failure

hakehuang, 2022-08-12:
**Describe the bug**
build failure

Please also mention any information which could help others to understand
the problem you're facing:
```
see: /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/mimxrt1170_evk_cm7/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/build.log
INFO    - Total complete:   11/  20  55%  skipped:   10, failed:    0
ERROR   - mimxrt1170_evk_cm7        tests/kernel/mem_protect/stackprot/kernel.memory_protection.stackprot FAILED: Build failure
ERROR   - see: /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/mimxrt1170_evk_cm7/tests/kernel/mem_protect/stackprot/kernel.memory_protection.stackprot/build.log
```

**To Reproduce**
Steps to reproduce the behavior:
```
 scripts/twister -p mimxrt1170_evk_cm7 --build-only -T tests/kernel/mem_protect/
```

**Expected behavior**
build pass

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**

```
FAILED: modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/caam/fsl_caam.c.obj
ccache /home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCPU_MIMXRT1176DVMAA_cm7 -DFSL_SDK_ENABLE_DRIVER_CACHE_CONTROL -DKERNEL -DTC_RUNID=b497ff30139cffe3653f0807cb7375cd -DXIP_BOOT_HEADER_DCD_ENABLE=1 -DXIP_BOOT_HEADER_ENABLE=1 -DXIP_EXTERNAL_FLASH -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/common/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/drivers/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/CMSIS/Core/Include/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/caam/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/gpt/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/boards/evkmimxrt1170/xip/. -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/mimxrt1170_evk_cm7/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/zephyr/include/generated -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/soc/arm/nxp_imx/rt -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/testsuite/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/testsuite/ztest/include -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/cmsis/CMSIS/Core/Include -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176 -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/drivers -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/xip -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/caam -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/gpt -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7 -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/boards/evkmimxrt1170 -isystem /home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/libc/minimal/include -isystem /home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -fno-strict-aliasing -Os -imacros /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/mimxrt1170_evk_cm7/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/ubuntu/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/kernel/mem_protect/stack_random=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/caam/fsl_caam.c.obj -MF modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/caam/fsl_caam.c.obj.d -o modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/caam/fsl_caam.c.obj -c /home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/caam/fsl_caam.c
/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/caam/fsl_caam.c:18:2: error: #warning "DCACHE must be set to write-trough mode to safely invalidate cache!!" [-Werror=cpp]
   18 | #warning "DCACHE must be set to write-trough mode to safely invalidate cache!!"
      |  ^~~~~~~
cc1: all warnings being treated as errors
[220/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/ssl_msg.c.obj
[221/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/x509_crl.c.obj
[222/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/x509_csr.c.obj
[223/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/rsa.c.obj
[224/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/ssl_srv.c.obj
[225/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/x509.c.obj
[226/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/sha1.c.obj
[227/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/ssl_tls.c.obj
[228/270] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dir/home/shared/disk/zephyr_project/zephyr_test/modules/crypto/mbedtls/library/x509_crt.c.obj
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  zephyr-v3.1.0-3420-ga41f4f30b5

hakehuang, 2022-08-12:
[mimxrt1170_evk_cm7.zip](https://github.com/zephyrproject-rtos/zephyr/files/9313630/mimxrt1170_evk_cm7.zip)
@dleach02  @EmilioCBen  is is ok when we upgrade the hal.

hakehuang, 2022-08-12:
@CristXu  find this issue in #48927

hakehuang, 2022-08-12:
bisect found this commit trigger this issue

d556a0c8a6699cc430ff2464b612a1ea9d25e948 is the first bad commit
commit d556a0c8a6699cc430ff2464b612a1ea9d25e948
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Wed Aug 10 14:22:45 2022 -0500

    drivers: entropy: Add entropy driver for MCUX CAAM

    Added entropy shim driver for MCUX CAAM and Kconfig symbol

    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>

:040000 040000 9e53c1916d525b55f344466d085a6f6882ee1617 e28aca1a26af00fa01ebf49f36ec48b18eb4f5db M      drivers
:100644 100644 a1245a7ade0d1c924abacd6c40f8b4738b288038 62d3f3dae5aead62f4f56c53a8373ddf76aef86c M      west.yml

mmahadevan108, 2022-08-12:
cc @decsny

PR #49033 (2022-08-12): soc: rt: CAAM requires cache write-through mode
The use of the RT11xx CAAM requires
cache write-through mode,
otherwise some tests give a compiler warning.

Add to the soc CMake to do this

Fixes #48977
closed 2022-08-12 by galak