zephyrproject-rtos/zephyr #48992: qemu_leon3: tests/posix/common/portability.posix.common fails

stephanosio, 2022-08-12:
**Describe the bug**

The `tests/posix/common/portability.posix.common` test fails on the `qemu_leon3` target:

```
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1764
	aborted _current back from dead
```

This assertion failure occurs after the test successfully completes and the test thread is aborted (fails to swap):

https://github.com/zephyrproject-rtos/zephyr/blob/6cfb18686e1c494d0011aec87bbf24ab530d3a34/kernel/sched.c#L1761-L1765

**To Reproduce**

Build `tests/posix/common/portability.posix.common` for `qemu_leon3` and run inside QEMU.

**Expected behavior**

Test passes.

**Impact**

CI reports a failure.

**Logs and console output**

```
Running TESTSUITE posix_apis
===================================================================
START - test_nanosleep_0_1
 PASS - test_nanosleep_0_1 in 0.020 seconds
===================================================================
START - test_nanosleep_0_1000000000
 PASS - test_nanosleep_0_1000000000 in 0.001 seconds
===================================================================
START - test_nanosleep_0_1001
 PASS - test_nanosleep_0_1001 in 0.019 seconds
===================================================================
START - test_nanosleep_0_500000000
 PASS - test_nanosleep_0_500000000 in 0.510 seconds
===================================================================
START - test_nanosleep_0_n1
 PASS - test_nanosleep_0_n1 in 0.001 seconds
===================================================================
START - test_nanosleep_1_0
 PASS - test_nanosleep_1_0 in 1.010 seconds
===================================================================
START - test_nanosleep_1_1
 PASS - test_nanosleep_1_1 in 1.020 seconds
===================================================================
START - test_nanosleep_1_1001
 PASS - test_nanosleep_1_1001 in 1.020 seconds
===================================================================
START - test_nanosleep_NULL_NULL
 PASS - test_nanosleep_NULL_NULL in 0.001 seconds
===================================================================
START - test_nanosleep_NULL_notNULL
 PASS - test_nanosleep_NULL_notNULL in 0.001 seconds
===================================================================
START - test_nanosleep_n1_0
 PASS - test_nanosleep_n1_0 in 0.001 seconds
===================================================================
START - test_nanosleep_n1_n1
 PASS - test_nanosleep_n1_n1 in 0.001 seconds
===================================================================
START - test_nanosleep_notNULL_NULL
 PASS - test_nanosleep_notNULL_NULL in 0.001 seconds
===================================================================
START - test_nanosleep_notNULL_notNULL
[... 193 lines cut ...]
E:  #3   4000fcb0   00000000
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x400166a0 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.14.2
 - Commit SHA: 0226e53e239f8562b0636a2a4e42e99b0894f784

**Additional context**

Bisected to 9a4b5e1d908575a552f15204a51c92c9732f3d84; however this commit, in itself, does not seem to be doing anything wrong and the failure is likely due to a more serious underlying issue.

This failure is likely due to an alignment-related issue:

https://github.com/zephyrproject-rtos/zephyr/blob/9a4b5e1d908575a552f15204a51c92c9732f3d84/subsys/testsuite/include/zephyr/tc_util.h#L56-L59

Changing the number of `=`s by -7 or +7 makes the assertion failure go away.

Related discussion on Discord: https://discord.com/channels/720317445772017664/733037890514321419/1007518292732289034

p.s. That this test also fails when compiled using the Zephyr SDK 0.15.0, so the issue is not toolchain or QEMU version specific.

stephanosio, 2022-08-12:
For now, the quickest workaround is to specify `-mfaster-structs`, which forces all structs to have 8-byte alignment.

The root cause still needs to be investigated and fixed.

andyross, 2022-08-12:
Took a quick look.  Agree that the alignment guess seems close.  After checking to be sure this wasn't a stack overflow (seriously: always check the stacks on weird stuff like this) I got it localized to a particular test case that isn't doing anything but creating a thread and joining it synchronously.  See notes in this patch: https://gist.github.com/andyross/d2edbd77463000e57b4c7a39cac31ed7

I thought it was a race at first, so I added a delay between the steps and that "fixed" it, as did a yield.  But so did a busywait, which is more suspicious (that shouldn't generally cause a context switch), and so did a busy wait of 1us, which was weird.  And so did a hand-written delay.  And so did the **same** hand written delay with a loop count of 1 that adds only four instructions and a stack word!  (And indeed, AFAICT the spawned thread never runs before the join in that final "working" case, it's not a race between them at all).

But hand-adding four NOPs did not fix the problem, nor did my attempts to expand the stack frame by a word via other means.  So no root cause yet.

But this definitely looks like a glitch in the platform context switch code.  That pthread_join() is just going to call arch_switch() synchronously.  My guess is that there's an edge case in there that gets something wrong based on some oddity about the stack frame + instruction pointer state.

But it could totally be a bug in the pthread code too that just happens to fail here, or in the kernel for that matter.  But my intuition points at the arch code.

PR #63397 (2024-01-02): posix: timer: Fix timer_obj alignment
This makes sure the heap `posix_timer_slab` provides objects aligned compatible with the type `timer_obj`. It was previously set to align at 4 bytes. One example where this failed was on the SPARC which requires access to `int64_t` to be 8-byte aligned.
    
In particular, `struct timer_obj` contains fields of type `k_timer_t` and `struct _timeout` which can have 64-bit fields.
    
With this commit we now get the information on required alignment for `struct timer_obj` from the compiler by using `__alignof__()`.

Before adding this patch, the run-time con [... cut]
closed 2023-03-20 by github-actions[bot]