zephyrproject-rtos/zephyr #49004: unexpected eof in qemu_cortex and mps2

1xlibiao, 2022-08-12:
**Describe the bug**
On qemu_cortex and mps2 platforms, tests/subsys/logging/log_backend_init falied.

Please also mention any information which could help others to understand
the problem you're facing:
 - qemu_cortex and mps2
 - What have you tried to diagnose or workaround this issue?
 - Comment out the LOG_PANIC() of 943 lines under subsys()   in https://github.com/zephyrproject-rtos/zephyr/blob/f5378424f7f492a8ae0d173a015bc1186fcc7127/subsys/testsuite/ztest/src/ztest_new.c#L943

**To Reproduce**
Steps to reproduce the behavior:
1. For example on qemu_cortex_m0 : twister -p qemu_cortex_m0 -T tests/subsys/logging/log_backend_init/ -vv
2. See error
INFO    - 462/462 qemu_cortex_m0            tests/subsys/logging/log_backend_init/logging.log_backend_initialization FAILED unexpected eof (qemu 0.176s)
INFO    - /__w/zephyr/zephyr/twister-out/qemu_cortex_m0/tests/subsys/logging/log_backend_init/logging.log_backend_initialization/handler.log
ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-3289-g0066254f5650  ***
Running TESTSUITE log_backend_init
===================================================================
START - test_log_backends_initialization
 PASS - test_log_backends_initialization in 2.217 seconds
===================================================================
TESTSUITE log_backend_init succeeded
===================================================================
RunID: ca515183eafc8f3faad76f3742dcb209
PROJECT EXECUTION SUCCESSFUL

INFO    - /__w/zephyr/zephyr/twister-out/qemu_cortex_m0/tests/subsys/logging/log_backend_init/logging.log_backend_initialization/handler.log
INFO    - 4 Iteration:
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 -OS: Linux
 -Toolchain - SDK 14.1
 -Commit SHA  https://github.com/zephyrproject-rtos/zephyr/pull/48848

nashif, 2022-08-23:
The log is showing a successful run, so I am not sure what the issue is.

1xlibiao, 2022-08-24:
> The log is showing a successful run, so I am not sure what the issue is.

log error update

mengxianglinx, 2022-08-24:
Not aware of why "LOG_PANIC()" in ztest_new.c, but that need all log_backend implement "panic" interface, if not, access to NULL pointer will cause crash,  "unexpected eof" will come up.

hwang37, 2022-09-06:
@1xlibiao @mengxianglinx  could you continue updating and working on this bug?

mengxianglinx, 2022-09-07:
By implementing panic() interface for log_backend, the "unexpected eof" issue of some qemu platform disappears. close it.

closed 2022-09-07 by mengxianglinx