zephyrproject-rtos/zephyr #49008: ESP32:  net_buf_get() FAILED

sam131208, 2022-08-12:
**Describe the bug**
Add the hog.c to the project "samples\subsys\mgmt\mcumgr\smp_svr", and enable CONFIG_BT_SMP=y.
While connected to it, get the error "ASSERTION FAIL [buf] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:2371"

**Logs and console output**

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:5296
load:0x40093000,len:13748
ho 0 tail 12 room 4
load:0x4009b800,len:6336
entry 0x40093434
[esp32] [INF] [boot] chip revision: 1
[esp32] [INF] [boot_comm] chip revision: 1, min. bootloader chip revision: 0
[esp32] [INF] Enabling RNG early entropy source...
[esp32] [INF] *** Booting MCUboot build v1.9.0-1669-g3b79128 ***
[esp32] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Scratch: magic=bad, swap_type=0x0, copy_done=0x2, image_ok=0x2
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [INF] DRAM segment: start=0x76ac, size=0x18b8, vaddr=0x3ffbdb60
[esp32] [INF] IRAM segment: start=0x8f64, size=0x11dc4, vaddr=0x40080000
[esp32] [INF] start=0x400908f4
▒*** Booting Zephyr OS build zephyr-v3.1.0-3288-g236484456f30  ***
<inf> mcumgr_zephyr_grp: Registering Zephyr basic mgmt group
<inf> littlefs: LittleFS version 2.4, disk version 2.0
<inf> littlefs: FS at flash-controller@3ff42000:0x250000 is 6 0x1000-byte blocks with 512 cycle
<inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
<inf> littlefs: /lfs1 mounted
<inf> esp32_bt_adapter: BT controller compile version [1342a48]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .data initialise [0x3ffae6e0] <== [0x4000d890]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb0000] - [0x3ffb09a8]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb2730] - [0x3ffb6388]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb8000] - [0x3ffb9a20]
<dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffbdb28] - [0x3ffbdb5c]
<inf> smp_bt_sample: Bluetooth initialized
<inf> smp_bt_sample: Advertising successfully started
<inf> smp_sample: build time: Aug 12 2022 17:38:22
<inf> smp_bt_sample: Connected
uart:~$ ASSERTION FAIL [buf] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:2371

Backtrace:0x4008BBC7:0x3FFE75D0 0x400DAB8B:0x3FFE75E0 0x400D1CAF:0x3FFE7620

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr SDK 14.0.2
 - latest main version
 
**Project file
[smp_svr.zip](https://github.com/zephyrproject-rtos/zephyr/files/9315442/smp_svr.zip)

sam131208, 2022-08-18:
Today,  i update zephyr to latest and build , got the error "region `dram0_1_seg' overflowed by 1240 bytes". under is the build log.

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr
-- Cache files will be written to: D:/zephyrproject/zephyr/.cache
-- Zephyr version: 3.1.99 (D:/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: esp32
-- Found host-tools: zephyr 0.14.2 (C:/Users/sam/zephyr-sdk-0.14.2)
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (C:/Users/sam/zephyr-sdk-0.14.2)
-- Found BOARD.dts: D:/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/dts.cmake

warning: BT_CTLR_DATA_LENGTH_MAX (defined at subsys/bluetooth\controller/Kconfig:379) was assigned
the value '251' but got the value ''. Check these unsatisfied dependencies: BT_CTLR_DATA_LENGTH
(=n), BT_CTLR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_DATA_LENGTH_MAX and/or look up
BT_CTLR_DATA_LENGTH_MAX in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing D:/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/prj.conf'
Merged configuration 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/overlay-bt.conf'
Configuration saved to 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/.config'
Kconfig header saved to 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build
-- west build: building application
[273/283] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && C:\Users\sam\zephyr-sdk-0.14.2\xtensa-espressif_esp32_zephyr-elf\bin\xtensa-espressif_esp32_zephyr-elf-gcc.exe   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/subsys/mgmt/mcumgr/libsubsys__mgmt__mcumgr.a  zephyr/subsys/mgmt/mcumgr/zephyr_grp/libsubsys__mgmt__mcumgr__zephyr_grp.a  zephyr/subsys/dfu/boot/libsubsys__dfu__boot.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/bluetooth/libdrivers__bluetooth.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/littlefs/libmodules__littlefs.a  modules/mcuboot/libmcuboot_util.a  modules/zcbor/libmodules__zcbor.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/users/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/10.3.0"  -LD:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -TD:/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp32/ld/esp32.peripherals.ld  -Wl,-- [... cut ...] ot fit in region `dram0_1_seg'
c:/users/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/10.3.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.exe: region `dram0_1_seg' overflowed by 1240 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\zephyrproject\zephyr\samples\subsys\mgmt\mcumgr\smp_svr\build'

sylvioalves, 2022-09-21:
@sam131208 Can you add the following: `CONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE=4096`? This will make room for dram0_1 and will fix the error. Also, can you check whether this issue is still valid? Otherwise we can close this.

sam131208, 2022-09-22:
@sylvioalves Thanks, now it can be compiled and passed.

closed 2022-09-22 by sam131208