zephyrproject-rtos/zephyr #49016: intel_adsp smoke test fails with CONFIG_LOG_MIPI_SYST_USE_CATALOG=y

kv2019i, 2022-08-12:
**Describe the bug**
Enabling CONFIG_LOG_MIPI_SYST_USE_CATALOG=y for SOF Zephyr builds results in a binary that does not boot on Intel cavs2.5 platform.

I reproduce the issue with plain smoketest for intel_adsp, with following small diff:
```
--- a/tests/boards/intel_adsp/smoke/prj.conf
+++ b/tests/boards/intel_adsp/smoke/prj.conf
@@ -5,3 +5,8 @@ CONFIG_SCHED_CPU_MASK=y
 CONFIG_IPM=y
 CONFIG_IPM_CAVS_HOST=y
 CONFIG_IPM_CAVS_IDC=n
+CONFIG_LOG=y
+CONFIG_LOG_OUTPUT=y
+CONFIG_LOG_MODE_IMMEDIATE=y
+CONFIG_LOG_MIPI_SYST_ENABLE=y
+CONFIG_LOG_MIPI_SYST_USE_CATALOG=y
```

The resulting run on TGL cavs2.5 platform results in:

```
===================================================================
START - test_host_ipc
Simple message send...
Return message request...
Return message request 2...
Return message request, async...
Synchronous message send...
 PASS - test_host_ipc in 0.257 seconds
===================================================================
START - test_clock_calibrate
CLOCK: 39048471 Hz
[00:00:02.579,000] <err> probe: ASSERTION FAIL [z_spin_lock_valid(l)] @ ZEPHYR_BASE/include/zephyr/spinlock.h:142
        Recursive spinlock 0x9e02d6c8
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [z_spin_lock_valid(l)] @ ZEPHYR_BASE/include/zephyr/spinlock.h:142
        Recursive spinlock 0x9e02d6c8
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [z_spin_lock_valid(l)] @ ZEPHYR_BASE/include/zephyr/spinlock.h:142
        Recursive spinlock 0x9e02d6c8
...
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129
ASSERTION FAIL [plen >= 0] @ ZEPHYR_BASE/subsys/logging/log_msg.c:129

Backtrace:0xBE01416F:0xBE024894  |<-CORRUPTED
```

**To Reproduce**
Steps to reproduce the behavior:
1. apply patch above
2. west build -b intel_adsp_cavs25 tests/boards/intel_adsp/smoke
3. sign, copy and run smoke on target

**Expected behavior**
Test passes.

**Impact**
Unable to enable Sys-T catalog logging in SOF.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA: 5ceda9fe7d8a27a0226ea41be412cef4ab55ed5b

kv2019i, 2022-08-12:
FYI @nashif @dcpleung -- syst works, but when I enable catalog, we seem to hit some spinlock issues.

dcpleung, 2022-08-15:
* You will need `CONFIG_LOG_PRINTK=y` to get output in Sys-T format as the test subsys uses `printk()` by default.
* Also need https://github.com/zephyrproject-rtos/zephyr/pull/49082 to fix an issue on Xtensa.
* Lastly, catalog would not work with `printk()` due to how `printk()` is programmed. Catalog only works with `LOG_*()`. This means most of the test output will not in catalog format (unless `LOG_*()` is called without those tests).
* There seems to be some issue with the trace buffer where output are cut short, so `systprint` cannot parse them.

kv2019i, 2022-08-18:
@dcpleung Thanks. Let me test more. I added following to smoke test prj and indeed it doesn't crash anymore:

tests/boards/intel_adsp/smoke/prj.conf
CONFIG_LOG=y
CONFIG_LOG_OUTPUT=y
CONFIG_LOG_BACKEND_ADSP=y
CONFIG_LOG_BACKEND_ADSP_OUTPUT_SYST=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_MIPI_SYST_ENABLE=y
CONFIG_LOG_MIPI_SYST_USE_CATALOG=y
CONFIG_LOG_PRINTK=y

Build fails with XCC, but GCC build with Zephyr SDK is ok. XCC fail is:
```zephyr/lib/os/cbprintf_packaged.c:20:2: error: #error "CONFIG_CBPRINTF_PACKAGE_SUPPORT_TAGGED_ARGUMENTS " "requires toolchain to support _Generic!"```

kv2019i, 2022-08-22:
@dcpleung That's a bit complicated. That will be possible for some newer products, but for older ones we are using a fixed toolchain per board and these are older XCC versions.

dcpleung, 2022-08-22:
Hm... due to how the logging subsys is constructed, there is no way to get away from using _Generic.

dcpleung, 2022-08-30:
With the changes to ADSP HDA logging backend, and the following kconfigs, sys-t catalog messages worked on my end:

```
CONFIG_DMA=y
CONFIG_LOG=y
CONFIG_LOG_OUTPUT=y
CONFIG_LOG_PRINTK=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_MIPI_SYST_ENABLE=y
CONFIG_LOG_MIPI_SYST_USE_CATALOG=y
CONFIG_LOG_BACKEND_ADSP_HDA=y
CONFIG_LOG_BACKEND_ADSP_OUTPUT_SYST=y
```

closed 2022-09-05 by carlescufi