zephyrproject-rtos/zephyr #49019: Fix multiple issues with adxl372 driver

nordicjm, 2022-08-12:
**Describe the bug**
The adxl372 driver has many issues, it uses a Kconfig to select if it uses SPI or I2C interface which leads to undefined operation, this should be handled in the dts instead. It also does not support multiple instances, nor can it cope with 2 devices on SPI and I2C simultaneously, therefore the driver needs to be updated and fixed to support this and prevent firmware from crashing at start up.

**Impact**
Showstopper

nordicjm, 2022-08-12:
@mhennerich ping

mhennerich, 2022-08-12:
Well. When this driver was mainlined. The issues you mentioned were the way zephyr worked. Coming from Linux I was wondering a lot how Devicetree was    used in zephyr...
Anyways do you have a fix we can review? Otherwise I need to see who will have a look on our side.

Anyways thanks for pointing out!

nordicjm, 2022-08-12:
We do not have a fix for this unfortunately, no

MaureenHelm, 2022-08-30:
> Well. When this driver was mainlined. The issues you mentioned were the way zephyr worked. Coming from Linux I was wondering a lot how Devicetree was used in zephyr... Anyways do you have a fix we [...]

@mhennerich Any update finding someone at ADI that can help fix this driver?

mhennerich, 2022-09-05:
> @mhennerich Any update finding someone at ADI that can help fix this driver?

Yes @amiclaus will take care of this.

MaureenHelm, 2022-09-06:
> > @mhennerich Any update finding someone at ADI that can help fix this driver?
> 
> Yes @amiclaus will take care of this.

Thank you!

PR #50265 (2022-09-21): drivers: sensor: adxl372: update driver
Handle SPI/I2c interface in the dts.

Support multiple instances.

Tested with nucleo_f446re.

Fixes #49019
closed 2022-09-21 by fabiobaltieri