zephyrproject-rtos/zephyr #49026: Add a CI check on image file sizes (specifically around boards)

galak, 2022-08-12:
The Zephyr repo grows quite large and one of the main reasons is the image that exist as part of the board docs.  We should check that images aren't too large >100K or 200K per file as a CI check.

stephanosio, 2022-08-12:
Not sure if checking the size is the best approach.

I think the following criteria would be reasonable (for the image files under `boards/`):

* Ensure that image file type is JPEG (i.e. not PNG/lossless)
* Ensure that the width of the image does not exceed 750px.

What do you think? @gmarull @galak

galak, 2022-08-12:
I'm fine with that.  The JPEG requirement should get a little broader discussion.  I do think a file size check would also make sense:

https://github.com/marketplace/actions/lfs-warning

Might be able to do it for us.

For example, we shouldn't have let in:

 boards/arm/ast1030_evb/doc/ast1030_evb.png  at 12M (obviously being JPEG instead would improve that).

stephanosio, 2022-08-12:
> boards/arm/ast1030_evb/doc/ast1030_evb.png at 12M (obviously being JPEG instead would improve that).

Hmm .. that is beyond anything I imagined. I guess it would not harm anyone to have a file size criteria, as long as it is set at a fairly high limit.

gmarull, 2022-08-12:
What I'd do:

1. Limit images to JPEG/SVG
2. Limit image width up to 750px
3. Limit file size to something reasonable, e.g. 200-300kb max?

PR #53308 (2022-12-22): compliance: add an image size check (and fix DevicetreeBindingsCheck)
Add a compliance check to validate the size of images added in the repository, trying to limit the size of board images in particular.

Bumped into an issue with DevicetreeBindingsCheck as well, fixed that one too.

Closes #49026
closed 2022-12-22 by carlescufi