zephyrproject-rtos/zephyr #49027: Regulator support for gpio-leds

Snevzor, 2022-08-12:
** Issue **
I have some LEDs that are controlled using GPIOs connected to mosfet gates because the LEDs are connected to a dedicated fixed power regulator.
I can't seem to find a way to automatically turn on the regulator when using the LEDs in my application. 

**Describe the solution you'd like**
It seems that "vin-supply" is what I need, but it appears not implemented for gpio-leds.
Actually in my case "supply-gpios" is probably better as the LEDs are the only consumers of the regulator.

**Describe alternatives you've considered**
The alternative is to control the enable pin of the regulator before using the LEDs (using an inferred binding in /zephyr,user perhaps), but as we have an application for two different HW setups (with only one setup using the regulator) this would require some ugly #ifdef constructions.
Are there other solutions already available?

Snevzor, 2022-08-16:
My apologies. I forgot to underline that I need the regulator turned off when the LEDs are not used due to power restrictions.

henrikbrixandersen, 2022-08-16:
> My apologies. I forgot to underline that I need the regulator turned off when the LEDs are not used due to power restrictions.

I would likely write a custom LED driver for that, then. The driver could keep track of LEDs being on and turn off the regulator once the last LED is turned off. The LED driver itself could utilize the regulator API for controlling the regulator.

Snevzor, 2022-08-16:
Thanks for the suggestion @henrikbrixandersen. Could you perhaps direct me to a sample where the regulator API is currently used? I'm trying to look for "vin-supply"/"vin_supply" without any result.

henrikbrixandersen, 2022-08-16:
> Thanks for the suggestion @henrikbrixandersen. Could you perhaps direct me to a sample where the regulator API is currently used? I'm trying to look for "vin-supply"/"vin_supply" without any result.

I don't think we have any in-tree users of https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/regulator.h except for the tests in https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/regulator/fixed