zephyrproject-rtos/zephyr #49032: espi saf testing disabled

galak, 2022-08-12:
Because of CI build failures we disabled CONFIG_ESPI_SAF in:

https://github.com/zephyrproject-rtos/zephyr/pull/48962.

This should probably be fixed.

albertofloyd, 2022-09-29:
@jvasanth1 could you take a look into this? 
Even after setting espi_saf0 device to okay, generic error occurs.

Change on top of mainline.
+&espi_saf0 {
+       status = "okay";
+};

/home/jamezaar/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `saf_read':
/home/jamezaar/ecwork/zephyr_mchp/samples/drivers/espi/src/main.c:665: undefined reference to `__device_dts_ord_188'

Looks like the  nested device cannot be obtained via current macro 
static const struct device *const espi_saf_dev =
	DEVICE_DT_GET(DT_NODELABEL(espi_saf0));

albertofloyd, 2022-10-06:
Fixed #50799

closed 2022-10-06 by albertofloyd