zephyrproject-rtos/zephyr #49036: soc: telink_b91: ROM region section overlap

stephanosio, 2022-08-12:
**Describe the bug**

The Telink B91 SoC linker script defines its own `ROM_INIT` memory region that overlaps the `ROM` memory region defined by the [RISC-V arch common linker script](https://github.com/zephyrproject-rtos/zephyr/blob/5281b58d298066230b47b161d8926e3691fac22a/include/zephyr/arch/riscv/common/linker.ld#L79):

https://github.com/zephyrproject-rtos/zephyr/blob/5281b58d298066230b47b161d8926e3691fac22a/soc/riscv/riscv-privilege/telink_b91/linker.ld#L16

The intent of this seems to have been to place the `.init.*` sections at the beginning of the ROM memory region, and it achieves exactly that with the Binutils 2.35 included in the Zephyr SDK 0.14.2.

With the Binutils 2.38 included in the Zephyr SDK 0.15.0, however, the linker attempts to place these sections at the same memory address (because there are two separate overlapping memory regions) and outputs an error message saying that there are overlapping sections:

```
ld.bfd: section vector LMA [0000000020000000,0000000020000017] overlaps section vector LMA [0000000020000000,00000000200000bb]
```

**This is a bug, although it works for Binutils 2.35, because defining overlapping memory regions should not have been done in the first place and this working in the older versions of Binutils was purely by luck.**

The "correct" way of doing this would be to use the `zephyr_linker_sources(ROM_START ...`; but, the RISC-V linker script currently (questionably) places the `ROM_START` sections after the `vector` sections, which is not what this SoC wants to do:

https://github.com/zephyrproject-rtos/zephyr/blob/5281b58d298066230b47b161d8926e3691fac22a/include/zephyr/arch/riscv/common/linker.ld#L110-L126

**To Reproduce**

Build any test for `tlsr9518adk80d`.

```
scripts/twister -v -N -M -p tlsr9518adk80d -s tests/kernel/common/kernel.common
```

**Expected behavior**

Linker script does not define any overlapping memory regions.

**Impact**

Any boards based on the `telink_b91` SoC cannot be built with the Zephyr SDK 0.15.0 and above.

**Logs and console output**

```
/opt/zephyr-sdk-0.15.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: section vector LMA [0000000020000000,0000000020000017] overlaps section vector LMA [0000000020000000,00000000200000bb]
```

**Environment (please complete the following information):**

 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: 2c301fe22f53468261246fb7cfd88b2092372d78

stephanosio, 2022-08-12:
p.s. This can be easily fixed if the RISC-V common linker script is updated to place the `ROM_START` sections before the "vectors"; not sure what led to the decision to place these sections after the `vectors` for the RISC-V arch and if doing so is actually necessary.

cc @kgugala @pgielda @tgorochowik @tejlmand

fkokosinski, 2022-08-24:
Hi @andy-liu-telink! I wrote to you on Discord, but I'm not sure how often you check it. Are you perhaps working on this issue and/or will be creating a PR addressing it soon, or would you like some help with this issue?

commit 6c888b2d (2022-09-02): soc: telink_b91: Fix ROM region section overlap

Place vector sections after ROM_START sections.
Also add init.ld script that will prevent overlapping .init sections
in telink_b91 SoC.

Fixes #49036.
closed 2022-09-02 by carlescufi