zephyrproject-rtos/zephyr #49046: Cannot use devices behind I2C mux (TCA9548A)

jacobw56, 2022-08-13:
Our device is an nRF9160 attached to a TCA9548A I2C multiplexer with eight LSM6DSL IMUs attached, one per mux channel.  Following the tca954x test overlay, here is our overlay:

    &i2c1 {
        compatible = "nordic,nrf-twim";
        status = "okay";
        sda-pin = <26>;
        scl-pin = <27>;
    
        mux: tca9548a@70 {
            compatible = "ti,tca9548a";
            reg = <0x70>;
            status = "okay";
            label = "i2c_mux";
            #address-cells = <0x1>;
            #size-cells = <0x0>;
    
            ch0: mux_i2c@0 {
                compatible = "ti,tca9546a-channel";
                label = "i2c1_mux_0";
                reg = <0x0>;
                status = "okay";
                #address-cells = <0x1>;
                #size-cells = <0x0>;
    
                lsm6dsl0: lsm6dsl@6a {
                    compatible = "st,lsm6dsl";
                    label = "LSM6DSL32_0";
                    reg = <0x6a>;
                };
            };
    
            ch1: mux_i2c@1 {
                compatible = "ti,tca9546a-channel";
                label = "i2c1_mux_1";
                reg = <0x1>;
                status = "okay";
                #address-cells = <0x1>;
                #size-cells = <0x0>;
    
                lsm6dsl1: lsm6dsl@6a {
                    compatible = "st,lsm6dsl";
                    label = "LSM6DSL32_1";
                    reg = <0x6a>;
                };
            };
    
            ... for six more channels/sensors
        };
    };

From this issue, I also include the following in our `prj.conf`

    CONFIG_I2C=y
    CONFIG_SENSOR=y
    CONFIG_LSM6DSL=y
    CONFIG_I2C_TCA954X=y
    CONFIG_I2C_TCA954X_ROOT_INIT_PRIO=61
    CONFIG_I2C_TCA954X_CHANNEL_INIT_PRIO=62

Then in our init code, again following the tca954x test code, we have (just trying to test getting things up and running)

    #if DT_NODE_HAS_STATUS(DT_NODELABEL(ch0), okay)
    #define I2C_0_CTRL_NODE_ID DT_NODELABEL(ch0)
    #define I2C_0_CTRL_DEV_NAME DT_LABEL(I2C_0_CTRL_NODE_ID)
    #else
    #error "I2C 0 controller device not found"
    #endif
    
    ...
    
    #if DT_NODE_HAS_STATUS(DT_NODELABEL(lsm6dsl0), okay)
    #define SENSOR_0_DEV_NAME DT_LABEL(DT_NODELABEL(lsm6dsl0))
    #else
    #error "Sensor 0 device not found"
    #endif
    
    ...

    const struct device *sensor[8];
    const struct device *ch[8];

    ch[0] = device_get_binding(I2C_0_CTRL_DEV_NAME);
    ch[1] = device_get_binding(I2C_1_CTRL_DEV_NAME);
    ch[2] = device_get_binding(I2C_2_CTRL_DEV_NAME);
    ch[3] = device_get_binding(I2C_3_CTRL_DEV_NAME);
    ch[4] = device_get_binding(I2C_4_CTRL_DEV_NAME);
    ch[5] = device_get_binding(I2C_5_CTRL_DEV_NAME);
    ch[6] = device_get_binding(I2C_6_CTRL_DEV_NAME);
    ch[7] = device_get_binding(I2C_7_CTRL_DEV_NAME);

    for (int i = 0; i < 8; i++)
    {
        if (ch[i] == NULL)
        {
            LOG_WRN("Could not get I2C mux channel %d", i);
        }
    }

    sensor[0] = device_get_binding(SENSOR_0_DEV_NAME);
    sensor[1] = device_get_binding(SENSOR_1_DEV_NAME);
    sensor[2] = device_get_binding(SENSOR_2_DEV_NAME);
    sensor[3] = device_get_binding(SENSOR_3_DEV_NAME);
    sensor[4] = device_get_binding(SENSOR_4_DEV_NAME);
    sensor[5] = device_get_binding(SENSOR_5_DEV_NAME);
    sensor[6] = device_get_binding(SENSOR_6_DEV_NAME);
    sensor[7] = device_get_binding(SENSOR_7_DEV_NAME);

    for (int i = 0; i < 8; i++)
    {
        if (sensor[i] == NULL)
        {
            LOG_WRN("Could not get LSM6 device %d", i);
        }

        uint8_t buffer[1];
        i2c_reg_read_byte(ch[i], 0x6a, 0x0f, buffer);
        if (buffer[0] != 0x6a)
        {
            LOG_WRN("Bad WHOAMI from sensor %d", i);
        }
    }

Here, the calls to `i2c_reg_read_byte` and the `device_get_binding` of each of the mux channels works great.  However, the calls to `device_get_binding` for the sensors all return NULL, and worse, on the oscilloscope we can see that they don't try to change the channel on the mux before calling to the sensors (which is why they fail).  Luckily the mux starts with the channels all disabled, because obviously if it initialized with one of them enabled, it would see an LSM6DSL at the end of it, even though it might be the wrong one, but I digress.

Is there some other secret to getting the fw to switch the mux channel appropriately for the corresponding device that we are missing?

mmahadevan108, 2022-08-16:
cc @teburd

teburd, 2022-08-16:
I think this is more of a device tree issue than an i2c issue if I'm reading this right cc @mbolivar

closed 2022-08-18 by galak

nyameaama, 2024-06-21:
@jacobw56 Where you able to fix your issue? I'm experiencing the same problems

PR #49236 (2022-08-18): drivers: i2c_tca954x: Fix transfer function
This is a follow-up to commit e1c0a494b3dc697d65a5cfab92b3bb8711ab0862.

The `tca954x_transfer()` function cannot call `i2c_transfer_dt()`,
because the I2C device address used in the transaction must be the one
passed as the `addr` parameter, not the address of the TCA954xA switch
itself. Hence, this commit restores the call to `i2c_transfer()`.

Fixes #49046.