zephyrproject-rtos/zephyr #49047: LORAWAN Devicetree sx1262 setup on rak4631_nrf52840 board

nilmonto, 2022-08-13:
**Configured dio1-gpios on rak4631_nrf52840 devicetree leads to tx timeouts**

Hi all, I've been trying to properly use the zephyr rtos with lorawan on my rak4631 development board for the past days but got stucked on the fact that I could send join requests but always got the error message *lorawan: MlmeConfirm failed : Tx timeout*.

Here are more info about my setup:
 - Platform: Wiscore Rak4631 based on nrf52840
 - Running Zephyr sample subsys/lorawan/class_a
 - Using VSCode and Platformio as IDE
 - Debugging with a FTDI232H using Openocd and configuration file board/nordic_nrf52_ftx232.cfg
 - Ubuntu

**Impact**
sx1262 lora modem does not work as expected;

**Possible solution**
After some research, found that the polarity defined at devicetree for this board is inverted. The below code on the file board/arm/rak4631_nrf52840 seems to solve the problem:

```
&spi1 {
	compatible = "nordic,nrf-spi";
	status = "okay";
	sck-pin = <43>;
	mosi-pin = <44>;
	miso-pin = <45>;
	cs-gpios = <&gpio1 10 GPIO_ACTIVE_LOW>;

	lora: sx1262@0 {
		compatible = "semtech,sx1262";
		reg = <0>;
		label = "sx1262";
		reset-gpios = <&gpio1 6 GPIO_ACTIVE_LOW>;
		busy-gpios = <&gpio1 14 GPIO_ACTIVE_HIGH>;
		tx-enable-gpios = <&gpio1 7 GPIO_ACTIVE_LOW>;
		rx-enable-gpios = <&gpio1 5 GPIO_ACTIVE_LOW>;
		dio1-gpios = <&gpio1 15 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>; // dio1-gpios = <&gpio1 15 GPIO_ACTIVE_LOW>; 
		dio2-tx-enable;
		dio3-tcxo-voltage = <SX126X_DIO3_TCXO_3V3>;
		tcxo-power-startup-delay-ms = <5>;
		spi-max-frequency = <1000000>;
	};
};
```

I don't know if that is the correct way to inform/suggest this correction (sorry if not) but I believe it could help someone in the future not to waste some time to find it out.

RomainPelletant, 2022-08-14:
Hi @nilmonto 

Could you show us the inverted polarity line please? Are you talking about mosi/miso pinout?

Thanks

nilmonto, 2022-08-15:
> Hi @nilmonto
> 
> Could you show us the inverted polarity line please? Are you talking about mosi/miso pinout?
> 
> Thanks

Its about the *dio1-gpios* configuration, as it is on repository (&gpio1 15 GPIO_ACTIVE_LOW) the driver keeps giving tx timeout as mentioned. After some trials, I found that the configuration of *(GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)* is the correct one for the same gpio1 15 pin., so maybe its just a matter of gpio configuration on devicetree file.  

Looking at SX1262 datasheet, this dio1 pin refers to a gpio, and seems to be used by chipset firmware as a loopback to zephyr loramac driver, not sure as I am new to the framework.. If anyone with this development kit could do some test to verify the sugested change on .dts it would be nice..

RomainPelletant, 2022-08-15:
According to SX1262 datasheet you are right : dio (irq) is active high (not low as written in initial dts file)
Based on sx127x, GPIO pull down is required:
[image]

Will you open a pull request with your modification?

PR #49084 (2022-09-19): boards: arm: rak4631_nrf52840: sx1262 dio1-gpios setup change
Change on rak4631_nrf52840 devicetree dio1-gpios definition to properly interface with sx1262 lorawan chipset.
Fixes: #49047
closed 2022-09-19 by carlescufi