zephyrproject-rtos/zephyr #49054: STM32H7 apps are broken in C++ mode due to HAL include craziness

dberlin, 2022-08-15:
**Describe the bug**

Trying to build a C++ based zephyr app that works fine on plenty of boards, fails miserably on STM32H7 based boards that require the HAL. 
This is with zephyr-sdk-0.14.2

Let's take nucleo_h745zi_q_m7 for example.

Any C++ source code that includes anything that ends up requiring code from the HAL (IE most of it), will issue all sorts of horrible errors.

What is happening is simple:
All of the stm32h7 hal .h files have extern "C" blocks, then include system headers *inside* those blocks.
Which is honestly insane.

So for example, stm32h7xx_hal_def.h includes stddef.h and math.h inside the extern "C" block.
math.h now includes cmath in C++ mode
cmath includes some experimental math functions, which start include C++ header files like algobase and such.
Which, being inside an extern "C" block, explodes horribly into 10000 errors.

I imagine this worked at some point out of luck - it will only work if no system header has any C++ code, *in C++ mode*.

*None* of these header files should be included header files inside an extern "C" block.
The extern block should always be after the includes.

FRASTM, 2022-08-16:
@dberlin thanks for reporting this. 
It seems that all the stm32Cube files are written in that form 
```
#ifdef __cplusplus
 extern "C" {
#endif
```
followed by `#include`

Do you have a simple piece of code to build and check the pb.
Is it just a question of moving 
#include <stddef.h>
#include <math.h>
out of the 
```
#ifdef __cplusplus
 extern "C" {
#endif 
```
block ?

dberlin, 2022-08-16:
I'll make you a piece of code.
There are two options, one of which fixes the code to be right, and the
other doesn't require code mods and seems to be well-supported by gcc but
requires toolchain update to do it "right".

1. The "fix-the-code" way is to move all the extern C blocks to not cover
the includes, as you say.
To test this I wrote a simple python parsing script to do this perfectly
for single-include-block files (for STM32H7, this is most of them), and
mostly-right for multi-include-block-files.

By "multiple-include-block", i meant that they have includes further in the
header file somewhere, usually guarded by a define.
stm32h7xxx_hal_conf.h is a good example.  Right now the parser i wrote
doesn't track nested ifdefs, so it will accidentally place the end of the
new extern "C" block inside the define sometimes in these cases. I fixed
these by hand for testing.
Totally fixable in the script

This works when done, but obviously it's a large patch against upstream to
carry, which is why i started writing something that could automate it/not
require a patch be maintained.

2. The "workaround-in-toolchain" way takes advantage of the fact that main
offender is the math.h->cmath->lots of C++ include that happens.  If you
#define _GLIBCXX_INCLUDE_NEXT_C_HEADERS 1, g++ will include the C version
of math.h without including cmath and adding C++ functions to it.

I can't tell if this will be deprecated at some point - libstdc++ release
notes for gcc 12 have a discussion of the c header modes and says one of
them will be deprecated, but i haven't figured out if that affects this or
not.

Assuming it's not being deprecated, this should happen in the toolchain,
but for testing i've just made a math.h header file that looks like this:
```
#ifndef _LIB_STL_THREADED_MATH_H_
#define _LIB_STL_THREADED_MATH_H_
#undef GLIBCXX_INCLUDE_NEXT_C_HEADERS
#define _GLIBCXX_INCLUDE_NEXT_C_HEADERS 1
#include_next <math.h>

#endif /* _LIB_STL_THREADED_MATH_H_ */
```

This  works without any changes to the STM32 HAL (On both 0.14.2 and
0.15.0-rc2 SDK).
This is not guaranteed to work forever, of course. Or with other compiler
toolchains.
The entire stm32 hal (ie not just h7) uses math.h and a few other things
inside header files. I hand-checked those other system headers, and they do
*not* seem to include C++ compat headers by default.

So just fixing the math.h issue seems like it will work without modifiying
the code, and at least successfully kick the can down the road for a few
more years.

dberlin, 2022-08-16:
So, digging a lot further into option 2 - this should already be happening - cmath already defines that.

What is really happening is some kind of weird include-guard issue causing it to not include sys-include/math.h in the right place all the time, which ends up missing all the global functions it needs to make the extern "C" block mess they have work.

The good news it this may be easier to fix. I'll spend the day staring at preprocessor dumps and include trees.

dberlin, 2022-08-17:
This is my fault in the end - the includes are very fragile to ordering because of how the include guards are structured (and they are re-entrant in some cases), and apparently some compiler option I added ended up screwing this up.

Without that option, I can get it to work.

Their code is still wrong, of course, but i also doubt they will fix it :)

closed 2022-08-17 by dberlin

dberlin, 2022-10-11:
Oh, one other thing.
The upshot of #2 is that #include <math.h> in C++ will not, by default,
include the extra tr1/etc transcendental functions that are part of C++ but
not C (For some versions of C++ and some versions of C).

I suspect this is fine in practice.