zephyrproject-rtos/zephyr #49056: STM b_u585i_iot02a MCUboot crash

JiiZR, 2022-08-15:
**Describe the bug**
MCUboot crashes with latest Zephyr codes.

I: OSPI flash config is OPI / DTR
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0xff0000cc
E: r0/a1:  0x00000000  r1/a2:  0x2000002c  r2/a3:  0x00000000
E: r3/a4:  0x00800000 r12/ip:  0x00000001 r14/lr:  0x08003825
E:  xpsr:  0x69000000
E: Faulting instruction address (r15/pc): 0x08005b8c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200001b8 (unknown)
E: Halting system

First I tried to disable OSPI driver from menuconfig (STM32 Octo SPI Flash driver)
but there is some linking problems:
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: zephyr/libzephyr.a(flash_map_default.c.obj):(.rodata.default_flash_map+0x5c): undefined reference to `__device_dts_ord_97'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Then I disabled chip, \bootloader\mcuboot\boot\zephyr\dts.overlay:

&mx25lm51245 {
	status = "disabled";
};

 and after this MCUboot boots OK. This is workaround for this crash.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b b_u585i_iot02a boot\zephyr
2. west flash

**Expected behavior**
boot without crash

*** Booting Zephyr OS build zephyr-v3.1.0-3576-g6825d8412340  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
E: Unable to find bootable image

**Impact**

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

FRASTM, 2022-08-24:
Could be a bootloader/mcuboot/boot/zephyr/boards/b_u585i_iot02a.overlay file made of
```
&mx25lm51245 {
 status = "disabled";
};
```

This overlay file is found but still get an error : 
_devicetree error: Label 'slot1_partition' appears on /soc/flash-controller@40022000/flash@8000000/partitions/partition@30000 and on /soc/flash-controller@40022000/flash@8000000/partitions/partition@78000_

erwango, 2022-08-31:
After anaylis of the crash, it appears this is a duplicated of https://github.com/zephyrproject-rtos/zephyr/issues/49715:
After initialization of the OSPI driver, the stack is corrupted with elements of the SFDP table:
```
z_sys_init_run_level (level=0, level@entry=2)
    at /local/mcu/zephyrproject/zephyr/kernel/device.c:73
73			if (dev != NULL) {
(gdb) 
CPU in Non-Secure state
CPU in Non-Secure state
0x08005b5e	73			if (dev != NULL) {
(gdb) 
CPU in Non-Secure state
CPU in Non-Secure state
77				if (rc != 0) {
(gdb) p dev
$3 = (const struct device *) 0xff0000c0

```

closed 2022-08-31 by erwango