zephyrproject-rtos/zephyr #49073: SOC_FLASH_LPC vs SOC_FLASH_MCUX

fabiobaltieri, 2022-08-15:
**Describe the bug**
Hey folks, seems like we have two drivers that can handle `compatible = "nxp,lpc-iap"`:

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/soc_flash_mcux.c#L35
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/soc_flash_lpc.c#L21

These used to be enabled manually on few platforms but since https://github.com/zephyrproject-rtos/zephyr/pull/48154 we started enabling them based on the dt setting, which means that both gets enabled at the same time.

This is causing build failures in CI for patches touching those platforms, something like

```
/usr/local/google/home/fabiobaltieri/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h: In function 'iap_entry':
/usr/local/google/home/fabiobaltieri/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h:109:24: error: 'FSL_FEATURE_SYSCON_IAP_ENTRY_LOCATION' undeclared (first use in this function)
  109 |     ((FLASHIAP_ENTRY_T)FSL_FEATURE_SYSCON_IAP_ENTRY_LOCATION)(cmd_param, status_result);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/google/home/fabiobaltieri/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h:109:24: note: each undeclared identifier is reported only once for each function it appears in
[7/20] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_lpc.c.obj
```

For SoCs that are missing that definition and should use the other driver.

I think that the solution would be to add some strategic `depends on SOC_LPC...` in https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/Kconfig.lpc and https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/Kconfig.mcux, though I'm not familiar enough with the families. Could you NXP folks look into it?

**To Reproduce**
Steps to reproduce the behavior:
$ ./scripts/twister -v -n -M -p lpcxpresso11u68 -p lpcxpresso51u68 -p lpcxpresso54114_m0 -p lpcxpresso54114_m4 -p lpcxpresso55s06 -p lpcxpresso55s16 -p lpcxpresso55s28 -p lpcxpresso55s36 -p lpcxpresso55s69_cpu0 -p lpcxpresso55s69_cpu1 -p lpcxpresso55s69_ns -T samples/drivers/flash_shell

^^ bunch of platforms fail

**Expected behavior**
Pass everywhere.

**Impact**
CI broken...

**Logs and console output**
```
$ ./scripts/twister -v -n -M -p lpcxpresso11u68 -p lpcxpresso51u68 -p lpcxpresso54114_m0 -p lpcxpresso54114_m4 -p lpcxpresso55s06 -p lpcxpresso55s16 -p lpcxpresso55s28 -p lpcxpresso55s36 -p lpcxpresso55s69_cpu0 -p lpcxpresso55s69_cpu1 -p lpcxpresso55s69_ns -T samples/drivers/flash_shell
Keeping artifacts untouched
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.1.0-3590-g91e4ae35f7e9
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - JOBS: 128
INFO    - 1 test scenarios (11 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  1/11 lpcxpresso55s28           samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s28/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    -  2/11 lpcxpresso11u68           samples/drivers/flash_shell/sample.drivers.flash.shell SKIPPED (runtime filter)
INFO    -  3/11 lpcxpresso55s36           samples/drivers/flash_shell/sample.drivers.flash.shell SKIPPED (runtime filter)
INFO    -  4/11 lpcxpresso51u68           samples/drivers/flash_shell/sample.drivers.flash.shell SKIPPED (runtime filter)
INFO    -  5/11 lpcxpresso54114_m4        samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso54114_m4/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    -  6/11 lpcxpresso54114_m0        samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso54114_m0/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    -  7/11 lpcxpresso55s69_cpu1      samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s69_cpu1/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    -  8/11 lpcxpresso55s16           samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s16/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    -  9/11 lpcxpresso55s06           samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s06/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    - 10/11 lpcxpresso55s69_cpu0      samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s69_cpu0/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    - 11/11 lpcxpresso55s69_ns        samples/drivers/flash_shell/sample.drivers.flash.shell FAILED Build failure (build)
ERROR   - see: /usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/twister-out/lpcxpresso55s69_ns/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: anything after a529c4511d (running on 6fc4bb9486248b2d2a9cc8eff505f365a4006ef1 right now)

fabiobaltieri, 2022-08-15:
cc @mmahadevan108

dleach02, 2022-08-16:
This has been fixed with #48832

fabiobaltieri, 2022-08-16:
Actually one of the tests is still failing:

`$ ./scripts/twister -v -n -M -p lpcxpresso55s69_cpu1 -T samples/drivers/flash_shell` <- this one

Looks like the cpu1 still enables `SOC_FLASH_MCUX`, but not `HAS_MCUX_IAP` (from https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nxp_lpc/lpc55xxx/Kconfig.soc#L77).

Should there be a `select HAS_MCUX_IAP` under `SOC_LPC55S69_CPU1`? Happy to add a patch to the PR where I'm hitting this test if it helps.

fabiobaltieri, 2022-08-16:
Guess if cpu1 is not supposed to access the flash controller I can add a

```patch
+&iap {
+       status = "disabled";
+};
```
to lpcxpresso55s69_cpu1.dts, that would disable the flash driver and runtime skip the test.

mmahadevan108, 2022-08-16:
cc @ yvanderv

fabiobaltieri, 2022-08-17:
Filed https://github.com/zephyrproject-rtos/zephyr/pull/49189 to propose disabling the iap node. Happy to change it if that's not the right thing to do.

closed 2022-08-25 by galak