zephyrproject-rtos/zephyr #49100: STM b_u585i_iot02a  NOR flash and OSPI_SPI_MODE, erase failed

JiiZR, 2022-08-16:
**Describe the bug**
Running twice samples/drivers/spi_flash test with altered data, test fails.

It seems that flash erase is failing. When using OPI mode ja DTR speed everything is OK.

**To Reproduce**
Steps to reproduce the behavior:
1. modify \zephyr\boards\arm\b_u585i_iot02a\b_u585i_iot02a-common.dtsi

	mx25lm51245: ospi-nor-flash@0 {
	...
		spi-bus-width = <OSPI_SPI_MODE>;
		data-rate = <OSPI_STR_TRANSFER>;
		
2.  west build -p auto -b b_u585i_iot02a_ns samples/drivers/spi_flash
3. west flash
4. See prints:
*** Booting Zephyr OS build zephyr-v3.1.0-3616-g5af932f6e788  ***

ospi-nor-flash@0 SPI flash testing

Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read matches data written. Good!!

5. modify \zephyr\samples\drivers\spi_flash\src\main.c
	const uint8_t expected[] = { 0x55, 0xaa, 0x66, 0x99 };
	
	to
	
	const uint8_t expected[] = { 0x55, 0xaa, 0x66, 0x22 };
6. west flash
7. see prints
*** Booting Zephyr OS build zephyr-v3.1.0-3616-g5af932f6e788  ***

ospi-nor-flash@0 SPI flash testing

Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read does not match data written!!
000ff000 wrote 55 read 55 match
000ff001 wrote aa read aa match
000ff002 wrote 66 read 66 match
000ff003 wrote 22 read 00 MISMATCH

**Expected behavior**
Test should every time even if written data is changed.

**Impact**
One bit mode is not fully working.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2022-08-16:
With f2c49619d2698a1abbeab62e5596a676938c863f, 
west build -p auto -b b_u585i_iot02a samples/drivers/spi_flash
gives the following output in the conditions of this issue:
```
OSPI flash config is SPI|DUAL|QUAD / STR                                       
*** Booting Zephyr OS build zephyr-v3.1.0-3609-gf2c49619d269  ***              
                                                                               
ospi-nor-flash@0 SPI flash testing                                             
==========================                                                     
                                                                               
Test 1: Flash erase                                                            
Flash erase succeeded!                                                         
                                                                               
Test 2: Flash write                                                            
Attempting to write 4 bytes                                                    
 Wr : 55, aa, 66, 22                                                           
Data read matches data written. Good!!                                         
 Rd : 55, aa, 66, 22  
```

JiiZR, 2022-08-16:
Did you try all steps from 1 to 7 to reproduce problem ?

FRASTM, 2022-08-16:
Yes, the pb comes from the AddressSize when eraing in SPI mode : should be 24 bits but is uncorrectly set to 32bits.
Looking in  the SDFP table when processing the data to configure the `spi_nor_process_bfp_addrbytes`

JiiZR, 2022-08-17:
Fix corrects the erase problem, thanks.

closed 2022-08-25 by fabiobaltieri