zephyrproject-rtos/zephyr #49102: hawkbit - dns name randomly not resolved

gkujawsk, 2022-08-16:
**Describe the bug**
Domain resolution fails randomly with DNS_EAI_FAIL returned by function z_impl_z_zsock_getaddrinfo_internal()
(subsys/net/lib/sockets/getaddrinfo.c). The error is returned because ai_flags & AI_NUMERICHOST condition is met.
In the very same time hawkbit module do not set AI_NUMERICHOST explicitly anywhere in the code. It seems that struct addrinfo hints inside start_http_client() function is not properly initialized. It's a local variable thus it requires initialization before use but it's not initialized at all. This would explain a kind of random behavior I'm observing. Adding memset(&hints, 0, sizeof(struct addrinfo)) seems to alleviate the problem. 

**To Reproduce**
The behavior is random. I hit the bug when added SMP support to the hawkbit sample. The first call to hawkbit() probe works correctly eg. domain name is resolved. Subsequent calls fails. 

**Expected behavior**
DNS name resolved properly. 

**Impact**
Cannot upgrade firmware remotely. 

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit: Zephyr v3.1

carlescufi, 2022-08-16:
Note: I do not think that the networking stack supports SMP at this point. @rlubos and @jukkar can confirm.

ycsin, 2022-08-17:
> It's a local variable thus it requires initialization before use but it's not initialized at all. This would explain a kind of random behavior I'm observing. Adding memset(&hints, 0, sizeof(struct addrinfo)) seems to alleviate the problem.

I think you are right that at least the `ai_flags` of `hints` should be initialized, but I'm not sure if it should be 0 (if memset), maybe @rlubos @jukkar have any idea?

jukkar, 2022-08-17:
> I think you are right that at least the `ai_flags` of `hints` should be initialized, but I'm not sure if it should be 0 (if memset)

I think it is a safe to init structs to 0 when needed. If something else is expected, then a separate init macro or function should be provided and used for initialization.

PR #49181 (2022-08-18): subsys/mgmt/hawkbit: Fix `hints` uninitialized
Fixes #49102
closed 2022-08-18 by fabiobaltieri