zephyrproject-rtos/zephyr #49112: lack of support for lpsram cache

tmleman, 2022-08-16:
**Describe the bug**
z_xtensa_cache_flush_inv_all() doesn't flush lpsram.

**To Reproduce**
Steps to reproduce the behavior:
1. write something into memory area started from address 0xA0000000,
2. call z_xtensa_cache_flush_inv_all(),
3. check if changes are visible under physical memory starting from address 0x40000000

**Expected behavior**
Cache is flushed.

edersondisouza, 2022-08-25:
I'm not sure this classifies as a bug - indeed, although it's called "flush_inv_all", there's no support currently for flushing other cache areas than a default one. I'll change the label to "Enhancement" instead of "bug".

edersondisouza, 2022-10-07:
> I'm not sure this classifies as a bug - indeed, although it's called "flush_inv_all", there's no support currently for flushing other cache areas than a default one. I'll change the label to "Enhancement" instead of "bug".

Heh, it turned out to be actually a bug, PR at #51090

PR #51090 (2022-10-07): arch/xtensa: Fix cache invalidate/flush functions
- Fix cache invalidate/flush functions that were using the wrong instructions
- Add a test for them. Note that test is at board level (intel_adsp) to avoid assuming some memory area can be written without side effects.

Fixes #49112
commit c28d3820 (2022-10-07): arch/xtensa: Fix z_xtensa_cache_[flush|inv|flush_inv]_all functions

They were basically using the wrong instructions to traverse the cache
by index: `dhi`, `dhwb` and `dhwbi` instead of `dii`, `diwb` and `diwbi`
variants.

Fixes #49112.
closed 2022-10-07 by nashif