zephyrproject-rtos/zephyr #49121: Update 'west blob' to support source files with binary data

ifyall, 2022-08-16:
## Introduction

Zephyr supports pulling in binary blobs using 'west blob'. This mechanism supports .bin (and similar files) as well as pre-compiled libraries (i.e. .a files). This proposal extends the blob support to allow for C/C++ files containing only binary data encoded as an array of bytes (or similar mechanisms).

### Problem description

Acquiring .bin files and encoding them in C/C++ code is a use case that is already supported by the tooling. However, some silicon vendors have already produced libraries with the binary data in a C/C++ file. If we don't support the acquisition of these files, then vendors will have to re-implement something they already have, resulting in additional testing and risk of defect introduction.

### Proposed change

The proposal is to update https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/contribute/bin_blobs.rst to allow for C/C++ files that encode binary data. These files would be required to follow the same rules as documented for pre-compiled libraries: https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/contribute/bin_blobs.rst#precompiled-library-specific-requirements

## Detailed RFC

Two changes would be needed to fulfil this:
1. Documentation changes to add the rule allowing C/C++ files.
2. Updates to the zephyr_module.py rules for blobs to all img, lib and src.

### Proposed change (Detailed)

The code changes required will be in https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/zephyr_module.py, changing this:

`type:
required: true
type: str
enum: ['img', 'lib']`

to this:

`type:
required: true
type: str
enum: ['img', 'lib', 'src']`

Developers who want to add blobs to be acquired via 'west blog' will then need to add to their module.yml entries for their source files using the 'src' type.

### Dependencies

None.

### Concerns and Unresolved Questions

Known concerns are around developers abusing this by adding pre-compiled code that provides and API and violates security or other similar concerns.

## Alternatives

Acquiring .bin files and encoding them in C/C++ code is a use case that is already supported by the tooling. However, some silicon vendors have already produced libraries with the binary data in a C/C++ file. If we don't support the acquisition of these files, then vendors will have to re-implement something they already have, resulting in additional testing and risk of defect introduction.

quic-egmc, 2022-08-17:
> This proposal extends the blob support to allow for C/C++ files containing only binary data encoded as an array of bytes (or similar mechanisms).

But why?  If we can have tools take a binary and expose it via a named symbol is that not functionally equivalent?

ifyall, 2022-08-17:
@quic-egmc, my primary concern is that the files Infineon has are coming from a separate team as a self-contained and validated package. Changing where the files live, such as moving binary files and their C encoding to another repository and fetching them via the blob mechanism doesn't introduce many concerns. However, throwing away the C file and regenerating it using a different tooling system and then revalidating introduces risk that something is broken in a previously working package.

One way I can ameliorate this concern is to use the Zephyr binary conversion tools and then do a md5 or equivalent to be sure that the regenerated C file is functionally equivalent to what is generated by the other team. I will take a look at that prior to next TSC.

sjg20, 2022-08-17:
Here are some points against this proposal:

1. We have tools for dealing with binary files, dumping them in hex, extracting strings, copying, etc. Dealing with a binary file in C format seems like a pain. How do I get the binary out?
2. There is no need to convert a binary file to C just to include it in the build. We can use 'incbin' in an assembly file to do this, for example, which allows symbols to be added. I'm sure there are other ways.
3. The C format is a derived format, not the original file, since as mentioned above there is tooling needed to convert from binary to C. We should try to have (closer to) original files in Zephyr, not things processed by tools that we don't have. Vendors don't really need to re-implement something, just stop running that extra step.
4. Different vendors may format the C code differently, complicating any effort to ingest / convert things to another format later, but there is no ambiguity as to the format of a binary file
5. This adds complexity since the only way to obtain the binary file is to compile a board within Zephyr! I am not suggesting that the file would contain C code or preprocessor elements, but that format is *far* more complex than is needed for a simple binary file
6. People might use #ifdef to change the content of the binary depending on the board being built. I hope not!
7. This 'binary blob' thing is provided to some extent as a courtesy to people who don't want to publicise the source code for the blobs. We should make this accommodation as easy as possible on the Zephyr project and its contributors. A plain binary file seems best.

BTW there is 'binman firmware packager' if we want to add more sophistication in this area to Zephyr.

ifyall, 2022-08-17:
@sjg20 We do use #ifdef to change the firmware image, since there are two choices, normal image or manufacturing test image. See here for instance: https://github.com/Infineon/wifi-host-driver/blob/master/WiFi_Host_Driver/resources/firmware/COMPONENT_43012/resources.h.

I assume making this decision (normal vs mfg image) could be configured via KConfig.

romkell, 2022-08-18:
### C encoded bin blobs
To me, from the contents nature it clearly is of "Allowed Types:Firmware images" encoded in C or even double encoded in ihex or srec or other and encoded in C and not a pre-compiled library.
From that perspective it is rather a special case (subtype) of "Allowed Types:Firmware images" which would have to be described/regulated there and not a new type.

I can see the advantage of allowing "C encoded binaries" especially when your receive it in that format from a third-party. On the other hand regulating all kind of special cases makes the handling complicated and therefore I would rather vote for keeping the interface small.

@ifyall: I feel, you should convince your separate team to provide the images as C encoded files (if still used somewhere else) **and** as *.bin files to be included via "incbin" in your package. I guess they must have them somewhere as bin.

### Licensing
It is very important that, as a user of Zephyr, I do not run into the situation that I have to disclose my companies IP, because I did not recognize linking against non-permissively licensed code / binary parts.

Hence what ever change we do, that should be "guaranteed".

To my understanding that is covered by
- as described in "Tainting" and
- "Precompiled library-specific requirements:Interface header files"

already.

mbolivar-nordic, 2022-08-26:
> @ifyall: I feel, you should convince your separate team to provide the images as C encoded files (if still used somewhere else) **and** as *.bin files to be included via "incbin" in your package. I guess they must have them somewhere as bin.

This.

Asking zephyr to open up arbitrary source files as .bin files just because you do not want to perform the conversion of a single array in a C file to a .bin file and validate the results for yourself is asking rather too much of upstream in my opinion.

henrikbrixandersen, 2022-08-26:
> Asking zephyr to open up arbitrary source files as .bin files just because you do not want to perform the conversion of a single array in a C file to a .bin file and validate the results for yourself is asking rather too much of upstream in my opinion.

I agree.

sjg20, 2022-08-31:
> @sjg20 We do use #ifdef to change the firmware image, since there are two choices, normal image or manufacturing test image. See here for instance: [...]

That makes it even worse, IMO. We have to decode the file with a C preprocessor to get the contents!

d1g2w3, 2022-09-13:
It's also worth noting outside of this use case, there are Machine Learning models that often come in C Array format. You look at various partners who do ML Training and AutoML in the TinyML community, they often output a C Array

henrikbrixandersen, 2022-09-13:
> It's also worth noting outside of this use case, there are Machine Learning models that often come in C Array format. You look at various partners who do ML Training and AutoML in the TinyML community, they often output a C Array 

That's not an issue unless you'll need to ship one of these models in order to support a SoC in upstream Zephyr (unlikely).

keith-zephyr, 2023-09-06:
Ping from the Process WG: @ifyall - Is this still something you would like Zephyr to adopt?  Or can we close this?

marc-hb, 2023-09-06:
Warning: I'm quoting different people.

> Vendors don't really need to re-implement something, just stop running that extra step.

Even simpler: stop omitting the original .bin file from their validated release package. The generated C file can stay in the package for any non-Zephyr user if needed.

> However, throwing away the C file and regenerating it using a different tooling system and then revalidating introduces risk that something is broken in a previously working package.

Oh, wait: so the original binary _is_ already in the release package in your case? So there's actually nothing to ask from your vendor?

> We have tools for dealing with binary files, dumping them in hex, extracting strings, copying, etc.

Could you share some pointers?

> ...  is to use the Zephyr binary conversion tools and then do a md5 or equivalent to be sure that the regenerated C file is functionally equivalent to what is generated by the other team. I will take a look at that prior to next TSC.

Good idea: you could find and report bugs in the vendor's tooling :-)

Maybe the easiest is to compile the two generated C files together and compare them with `memcmp()`.

> Different vendors may format the C code differently, complicating any effort to ingest / convert things to another format later,

The list of restrictions in https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/contribute/bin_blobs.rst makes it indeed pretty clear that binaries are more "tolerated" than welcome for a number of maintenance reasons. Example:
> Precompiled library blobs must be in a data format which is compatible with and can be linked by a toolchain supported by the Zephyr Project. This is required for maintainability and usability.

There is also a very important introduction at the top of this document:
> It is important to note that all the information in this section applies only to [upstream (vanilla) Zephyr](https://github.com/zephyrproject-rtos/zephyr).
> There are no limitations whatsoever (except perhaps license compatibility) in the support for binary blobs in forks or third-party distributions of Zephyr.

As soon as you distribute your own Zephyr branch, you can add to it absolutely whatever you want.

ifyall, 2023-09-06:
This was discussed previously and voted against. It is no longer relevant.

closed 2023-09-06 by ifyall

dleach02, 2023-11-22:
Just a comment on this closed RFC, I'm assuming the vote was to not allow a hex array C source file as a binary blob. Unless I'm missing this, shouldn't we document this point in the blobs section of the documentation?

We list the allowed types and the second one, firmware images, doesn't specify the actual format. It just says: 

- "An image containing the executable code for a secondary processor or CPU". 

When advising an internal team, they also have been delivering a C hex array as part of their deployment for the last several years. My initial feedback was that it wasn't explicitly called out that they needed to convert this to a binary file (my reading of the above line). But now I found  this discussion thread from the process working group. I'm guessing I might have missed this set of meetings but with the benefit of some hindsight, I think we should clearly articulate the rule/restriction around a source file for both libraries and firmware images submitted as a binary blob.