zephyrproject-rtos/zephyr #49136: L2CAP ecred test cases failed.

yajunzhi, 2022-08-17:
When I test L2CAP cases with PTS. There are 3 cases failed. 
L2CAP/ECFC/BV-11-C, L2CAP/ECFC/BV-13-C and L2CAP/ECFC/BV-15-C. 
From the test log output, the issue is that when send le_ecred_conn_resp packet to PTS. The PTS needs including valid CID value. But the l2cap of zephyr response with "CID 0x0000". 

It looks reasonable, but I want to know is zephyr passed these PTS cases or not. 

If zephyr projecct have some CICD job for these test cases?
Thank you.

hermabe, 2022-08-17:
There is an open PTS CASE for these tests (CASE0072485). PTS does not conform to the latest spec, where `0x0000` is the expected CID when the connection request is rejected.

Also note that L2CAP/ECFC/BV-15-C is category D, so it is not needed for qualification.

yajunzhi, 2022-08-17:
Ok, got it.
Now, I find the issue is processing and release a patch about the PTS tool. If the result of the connection is not success, the CID should be 0x0000. So, I think the zephyr code and your comment is correct.
Thank you.

kruithofa, 2022-09-21:
This is an issue with PTS, see  CASE0072485

closed 2022-09-21 by kruithofa