zephyrproject-rtos/zephyr #49148: Asnchronous UART API triggers Zephyr assertion on STM32WB55

r2r0, 2022-08-17:
**Describe the bug**
Zephyr assertion is triggered by UART reception when asynchronous API is used.
The event flow is as described below:
1. System timer ISR starts handling IDLE timeout (submitred from UART driver as delayed work).
2. In the `sys_clock_announce` (work) timeout is removed from the timeouts system queue and then interrupts are being reenabled (`k_spin_unlock`).
3. UART (IDLE) interrupt comes here and ISR reschedules delayed work (`k_work_reschedule`->`unschedule_locked`) and this reschedule causes attachement of timeout to the system timeouts queue (**timeout associated with this work is already being handled by system timer**).
4. Then control returns to the timer ISR ( `sys_clock_announce`) and there is called timeout function `t->fn(t)` which calls `work_timeout` where **`K_WORK_DELAYED_BIT` flag is cleared** (for the timeout being in the meantime added to the system timeouts queue) and then work associated with timeout is submitted do the system work queue (this work in executed as `uart_stm32_async_rx_timeout` and it doesn't change anything because work timeout cannot be effectively cancelled because `K_WORK_DELAYED_BIT` flag is cleared).
5. After short time another UART IDLE interrupt occurs and ISR tries to reschedule again IDLE timeout (delayed work) and this cannot be done properly because `K_WORK_DELAYED_BIT` flag is already cleared and because of this timeout is not removed from the timeouts queue (`z_abort_timeout` is not called).
6. In the next step `k_work_reschedule` calls `schedule_for_queue_locked` which calls `z_add_timeout` which requires that timeout being added cannot be already in the system timeouts queue and this causes crash (assertion is triggered).

Please also mention any information which could help others to understand
the problem you're facing:
 - target MCU: STM32WB55
 - Reverting LPUART and LPTIMER interrupts priorities make problem gone (hides it)

**To Reproduce**
Steps to reproduce the behavior:
1. Configure LPUART driver to use asynchronous API and implement appropriate user code.
8. Try to send relatively big amout of data from host to device (frames of about 400 bytes, delay between frames few milisesonds , serial speed 115200).
9. Zephyr assertion (`__ASSERT(!sys_dnode_is_linked(&to->node), "")`) can be observed triggered from `z_add_timeout` function.

**Expected behavior**
Assertion should not be triggered.

**Impact**
showstopper

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 9315aef26d79197ccdc293a72a9f8491b0b7b18b

**Additional context**
MCU is setup to use LPUART with LSE as a system clock, TIMESLICING is not configured.

FRASTM, 2022-08-23:
This seems that PM activation, let  the stm32 enter low power stop 0, 1, 2  modes and will also stop the DMA clocks.
Testing the tests/drivers/uart/uart_async_api/ on the lpuart of the stm32wb55 with PM_CONFIG=y, some testcases like 
 _test_chained_write_ or - _test_long_buffers_ fail.
When preventing the system  to enter low power stop mode  (with uart_stm32_pm_policy_state_lock_get) during  DMA transfers (on uart_stm32_dma_tx_enable and uart_stm32_dma_rx_enable) will make the transfer complete ( uart_stm32_pm_policy_state_lock_put when disabling)

See https://github.com/FRASTM/zephyr/pull/new/wb55_uart_async_pm

FRASTM, 2022-08-24:
from step 4 to 5 do you see that (re-starting) the data->dma_rx.timeout_work comes  (in step 5) _after_ the previous work has already expired  (expires too early from step 4)

r2r0, 2022-08-26:
@FRASTM 
Does it mean that it is a more generic problem?
I mean - Zephyr allows to change already handled timeout in system timer ISR.

r2r0, 2022-09-07:
Any progress for that?
I observed similar problem associated with using delayed work from interrupt.
We use separate HW timer and ISR for timer interrupt and depending on interrupt priority problems are visible (when timer interrupt priority is higher than LPTIM interrupt) or not.

r2r0, 2022-11-22:
It may be related to #51872.

erwango, 2023-01-05:
> It may be related to #51872.

@r2r0, if you're confident, I propose to close this issue. Tbh, we'll won't do nothing on current issue until #51872 is fixed.
Issue could be re-open later if required

r2r0, 2023-01-05:
Need to be checked if fix for #51872 also resolves this issue.

closed 2023-01-05 by r2r0