zephyrproject-rtos/zephyr #49154: SDMMC driver with STM32 U575

JA-vega, 2022-08-17:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
STM32 SDMMC driver isn't compatible with the u575 (?), I have defined the dts/dtsi from the SoC and board with the given pins, register and IRQ but I'm not able to use it. 

**Describe why you are asking for this support?**
I have added and configured the sdmmc for the U575 but it isn't working as expected.

- Error is coming from the HAL_SD_Init
- SDMMC Clock: 10 MHz
- SDMMC and FatFS working with same configuration in bare metal

**Additional context**
stm32u5.dtsi:
[image]
board .dts:
[image]
Zephyr console output:
[image]

FRASTM, 2022-08-17:
On a nucleo_u575zi_q with the DTS you show (except for the <cd-gpios>) it looks like a SDMMC clock pb.
By default the SDMMC clock selection is on ICLK (set by the RCC_CCIPR2)  bit 14 is 0. 
This ICLK is the Intermediate clock source selected by the RCC CCIPR1. By default the ICLK is on HSI48.

JA-vega, 2022-08-18:
Thanks a lot, the HSI48 was disabled yes

closed 2022-08-18 by JA-vega