zephyrproject-rtos/zephyr #49164: samples/arch/smp/pi/sample.smp.pi fails on a number esp32 platforms

galak, 2022-08-17:
`./scripts/twister -a xtensa -s samples/arch/smp/pi/sample.smp.pi -l`

Gets:

```
ERROR   - olimex_esp32_evb          samples/arch/smp/pi/sample.smp.pi                  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/olimex_esp32_evb/samples/arch/smp/pi/sample.smp.pi/build.log
INFO    - Total complete:    9/  19  47%  skipped:    8, failed:    0
ERROR   - esp_wrover_kit            samples/arch/smp/pi/sample.smp.pi                  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/esp_wrover_kit/samples/arch/smp/pi/sample.smp.pi/build.log
INFO    - Total complete:   10/  19  52%  skipped:    8, failed:    0
ERROR   - heltec_wifi_lora32_v2     samples/arch/smp/pi/sample.smp.pi                  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/heltec_wifi_lora32_v2/samples/arch/smp/pi/sample.smp.pi/build.log
INFO    - Total complete:   11/  19  57%  skipped:    8, failed:    0
ERROR   - esp32                     samples/arch/smp/pi/sample.smp.pi                  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/esp32/samples/arch/smp/pi/sample.smp.pi/build.log
INFO    - Total complete:   12/  19  63%  skipped:    8, failed:    0
ERROR   - odroid_go                 samples/arch/smp/pi/sample.smp.pi                  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/odroid_go/samples/arch/smp/pi/sample.smp.pi/build.log

```

```
_go/samples/arch/smp/pi/sample.smp.pi/zephyr && /usr/bin/cmake -E echo
/opt/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/10.3.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `noinit' will not fit in region `dram0_1_seg'
/opt/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/10.3.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: region `dram0_1_seg' overflowed by 17456 bytes
collect2: error: ld returned 1 exit status
```

sylvioalves, 2022-08-17:
@galak, well, no room to run this sample. Should we add esp32 family into ignore list as in https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/arch/smp/pktqueue/sample.yaml#L18?

galak, 2022-08-17:
> @galak, well, no room to run this sample. Should we add esp32 family into ignore list as in https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/arch/smp/pktqueue/sample.yaml#L18?

We have other tests/samples that fail due to size an those get ignored.  @nashif how does twister detect that?  Might need some updating for esp32.

sylvioalves, 2022-08-17:
If twister can catch this, then we can clean this ignore list.  Let's see if anything is needed on soc side.

PR #49356 (2022-08-22): twister: Add esp32 section name to overflow check
esp32 platforms can overflow their dram0_1_seg so add that section name
to the overflow check in twister.  We were seeing this with the
samples/arch/smp/pi/sample.smp.pi test and would get:

 zephyr_pre0.elf section `noinit' will not fit in region `dram0_1_seg'
 region `dram0_1_seg' overflowed by 17456 bytes

Fixes #49164
commit f16e2a9d (2022-08-22): twister: Add esp32 section name to overflow check

esp32 platforms can overflow their dram0_1_seg so add that section name
to the overflow check in twister.  We were seeing this with the
samples/arch/smp/pi/sample.smp.pi test and would get:

 zephyr_pre0.elf section `noinit' will not fit in region `dram0_1_seg'
 region `dram0_1_seg' overflowed by 17456 bytes

Fixes #49164
closed 2022-08-22 by nashif