zephyrproject-rtos/zephyr #49166: samples/drivers/flash_shell/sample.drivers.flash.shell fails to build on a few nxp platforms

galak, 2022-08-17:
`./scripts/twister -s samples/drivers/flash_shell/sample.drivers.flash.shell -p lpcxpresso55s69_cpu1 -p hexiwear_kw40z`

On hexiwear:
```
ftfx_cache.c.obj.d -o modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.c.obj -c /home/gala
k/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.c
In file included from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_features.h:21,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_controller.h:13,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.h:13,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.c:10:
/home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_adapter.h:290: error: "SIM_FCFG1_REG" redefined [-Werror]
  290 | #define SIM_FCFG1_REG           SIM->FCFG1
      | 
In file included from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/devices/MKW40Z4/./fsl_device_registers.h:44,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/common/./fsl_common.h:22,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_features.h:18,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_controller.h:13,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.h:13,
                 from /home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flash/fsl_ftfx_cache.c:10:
/home/galak/git/modules/hal/nxp/mcux/mcux-sdk/devices/MKW40Z4/./MKW40Z4.h:6829: note: this is the location of the previous definition
 6829 | #define SIM_FCFG1_REG(base)                      ((base)->FCFG1)
      | 
cc1: all warnings being treated as errors
```

on LPC:
```
In file included from /home/galak/git/zephyr/drivers/flash/soc_flash_lpc.c:17:
/home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h: In function 'iap_entry':
/home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h:109:24: error: 'FSL_FEATURE_SYSCON_IAP_ENTRY_LOCATION' undeclared (
first use in this function)
  109 |     ((FLASHIAP_ENTRY_T)FSL_FEATURE_SYSCON_IAP_ENTRY_LOCATION)(cmd_param, status_result);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/modules/hal/nxp/mcux/mcux-sdk/drivers/flashiap/fsl_flashiap.h:109:24: note: each undeclared identifier is reported only once for each function it appears in
```

mmahadevan108, 2022-08-17:
`lpcxpresso55s69_cpu1` should be excluded from the test as it does not have many features enabled.

mmahadevan108, 2022-08-17:
@galak, any idea why CI is building this test for` lpcxpresso55s69_cpu1`. The `.yaml` for this target has very limited features defined.

galak, 2022-08-17:
> @galak, any idea why CI is building this test for` lpcxpresso55s69_cpu1`. The `.yaml` for this target has very limited features defined.

The only filter is `    filter: CONFIG_FLASH_HAS_DRIVER_ENABLED` for the sample, so features defined in board yaml don't have any impact.

dleach02, 2022-08-17:
This is a duplicate of #48912. I will close the other one since this one has actual comments to retain.

danieldegrasse, 2022-08-17:
@galak #49197 should fix the LPC5569 CI failure. As for the kw40z, is that platform currently blocking CI? The root cause here is that the flash controller node for NXP SOCs is not set to `status="disabled"` by default, so the changes to make driver enablement depend on devicetree has effectively enabled the flash driver on untested platforms. If the kw40z is blocking CI, I can create a hotfix, but I would prefer to correctly disable the flash node by default, and reenable it on tested platforms as a long term fix for this issue.

galak, 2022-08-17:
> @galak #49197 should fix the LPC5569 CI failure. As for the kw40z, is that platform currently blocking CI? The root cause here is that the flash controller node for NXP SOCs is not set to [...]

I don't think the KW40Z is holding anything up right now.  So needs to get fixed before release, but not super urgent.  Looking at the error, seems like a HAL issue with defines getting set twice for KW40Z vs the flash driver for it.

closed 2022-08-25 by galak