zephyrproject-rtos/zephyr #49169: v2m_musca_s1_ns fails to build several tfm related samples

galak, 2022-08-17:
`./scripts/twister -T samples/tfm_integration/  -p v2m_musca_s1_ns`

Gets:
```
ERROR   - v2m_musca_s1_ns           samples/tfm_integration/psa_protected_storage/sample.tfm.protected_storage FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/psa_protected_storage/sample.tfm.protected_storage/build.log
INFO    - Total complete:    4/  16  25%  skipped:    3, failed:    0
ERROR   - v2m_musca_s1_ns           samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition/build.log
INFO    - Total complete:    5/  16  31%  skipped:    3, failed:    0
ERROR   - v2m_musca_s1_ns           samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition.ipc FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition.ipc/build.log
INFO    - Total complete:    6/  16  37%  skipped:    3, failed:    0
ERROR   - v2m_musca_s1_ns           samples/tfm_integration/tfm_ipc/sample.tfm_ipc     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/build.log
INFO    - Total complete:    7/  16  43%  skipped:    3, failed:    0
ERROR   - v2m_musca_s1_ns           samples/tfm_integration/psa_crypto/sample.psa_crypto FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/psa_crypto/sample.psa_crypto/build.log

```

Seems related to:
```
commit c51505f66167ee39033ab50d5c8ad67f4e955f25
Author: Joakim Andersson <joakim.andersson@nordicsemi.no>
Date:   Wed Mar 9 10:43:34 2022 +0100

    modules: tfm: Exclude non-secure TF-M application from build
```

microbuilder, 2022-08-17:
The first of these issues is already fixed via https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/74 ... but that doesn't explain the following errors, which I also get with `mps2_an521_ns`:

```
[2/5] Building C object CMakeFiles/TargetConfigGen.dir/Users/kevintownsend/Linaro/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage/tfm/tf-m-tests/app/psa_api_tests/targetConfigGen.c.o
FAILED: CMakeFiles/TargetConfigGen.dir/Users/kevintownsend/Linaro/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage/tfm/tf-m-tests/app/psa_api_tests/targetConfigGen.c.o 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -I/Users/kevintownsend/Linaro/zephyr/modules/tee/tf-m/psa-arch-tests/api-tests/val/nspe -I/Users/kevintownsend/Linaro/zephyr/modules/tee/tf-m/psa-arch-tests/api-tests/val/common -I/Users/kevintownsend/Linaro/zephyr/modules/tee/tf-m/psa-arch-tests/api-tests/platform/targets/common/nspe -I/Users/kevintownsend/Linaro/zephyr/modules/tee/tf-m/psa-arch-tests/api-tests/platform/targets/common/nspe/crypto -I/Users/kevintownsend/Linaro/zephyr/modules/tee/tf-m/psa-arch-tests/api-tests/platform/targets/tgt_dev_apis_tfm_musca_s1/nspe -arch arm64 -o CMakeFiles/TargetConfigGen.dir/Users/kevintownsend/Linaro/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage/tfm/tf-m-tests/app/psa_api_tests/targetConfigGen.c.o -c /Users/kevintownsend/Linaro/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage/tfm/tf-m-tests/app/psa_api_tests/targetConfigGen.c
/Users/kevintownsend/Linaro/zephyr/twister-out/v2m_musca_s1_ns/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage/tfm/tf-m-tests/app/psa_api_tests/targetConfigGen.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
```

Looking at what broke.

microbuilder, 2022-08-17:
Looks like a cmake issue since it's defaulting to the system compiler in the error above (`/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc`), so the cross compiler isn't being propagated up properly to the TF-M build system from Zephyr.

galak, 2022-08-17:
> The first of these issues is already fixed via [zephyrproject-rtos/trusted-firmware-m#74](https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/74) ... but that doesn't explain the following errors, which I also get with `mps2_an521_ns`:

Have we not updated west.yml for that?

microbuilder, 2022-08-17:
@galak Minding testing against https://github.com/zephyrproject-rtos/zephyr/pull/49176 ?

closed 2022-08-17 by microbuilder