zephyrproject-rtos/zephyr #49178: subsys: pm:  stats:  typo causes build failure

sslupsky, 2022-08-17:
**Describe the bug**
There is a typo in pm_stats.c:
```
diff --git a/subsys/pm/pm_stats.c b/subsys/pm/pm_stats.c
index 9118e3d5bf..d2883be691 100644
--- a/subsys/pm/pm_stats.c
+++ b/subsys/pm/pm_stats.c
@@ -40,7 +40,7 @@ static int pm_stats_init(const struct device *dev)
                        snprintk(names[i][j], PM_STAT_NAME_LEN,
                                 "pm_cpu_%03d_state_%1d_stats", i, j);
                        stats_init(&(stats[i][j].s_hdr), STATS_SIZE_32, 3U,
-                                  STATS_NAME_INIT_PARMS(stats));
+                                  STATS_NAME_INIT_PARMS(pm_stats));
                        stats_register(names[i][j], &(stats[i][j].s_hdr));
                }
        }
```

carlescufi, 2022-08-17:
@sslupsky mind sending a PR?

sslupsky, 2022-08-17:
@carlescufi done.

closed 2022-08-18 by galak