zephyrproject-rtos/zephyr #49200: Intel CAVS: Failure in tests/kernel/interrupt

shuai1x, 2022-08-18:
**Describe the bug**
On intel_adsp_cavs15/18/25 platform, tests/kernel/interrupt failed.

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs15/18/25
 - tests/kernel/interrupt No Console Output(Timeout)

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/kernel/interrupt/ --no-skipped-report -vv
4. see error

**Logs and console output**
```
INFO:cavs-client:Monitor process
INFO:cavs-client:Sent:     start_log
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
--
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.1.0-3609-gf2c49619d269  (delayed boot 500ms) ***
Running TESTSUITE interrupt_feature
===================================================================
START - test_isr_dynamic
installing dynamic ISR for IRQ 0
ASSERTION FAIL [!z_soc_irq_is_enabled(irq)] @ WEST_TOPDIR/zephyr/arch/common/sw_isr_common.c:84
IRQ 0 is enabled
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
E:  **  PC 0xbe009bc2 VADDR (nil)
E:  **  PS 0x60420
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:4 CALLINC:2)
E:  **  A0 0xbe0098b0  SP 0xbe0116f0  A2 0x4  A3 0xbe011700
E:  **  A4 0xbe0116e0  A5 0x4  A6 0xbe00f711  A7 0xffffffff
E:  **  A8 0xbe008f38  A9 0xbe011600 A10 0x12 A11 (nil)
E:  ** A12 0x25 A13 0x12 A14 0xbe01162d A15 0x8
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR 0x4
Backtrace:0xBE009BBF:0xBE0116F0  |<-CORRUPTED
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0xbe014168 (test_isr_dynamic)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 14.1
 - Commit SHA: 2241fb0592117fd41207705d09f98be53cdf4d1c

nashif, 2022-08-23:
related to a change from me.

nashif, 2022-08-23:
bisected to 2241fb0592117fd41207705d09f98be53cdf4d1c

PR #49437 (2022-08-24): Revert "intel_adsp: cavs: irq: simplify code"
This reverts commit 2241fb0592117fd41207705d09f98be53cdf4d1c.

Fixes #49200
commit a04d0651 (2022-08-24): Revert "intel_adsp: cavs: irq: simplify code"

This reverts commit 2241fb0592117fd41207705d09f98be53cdf4d1c.

Fixes #49200
closed 2022-08-24 by nashif