zephyrproject-rtos/zephyr #49203: Intel CAVS: Failure in tests/boards/intel_adsp

shuai1x, 2022-08-18:
**Describe the bug**
On intel_adsp_cavs15 platform, tests/boards/intel_adsp failed.

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs15
 - tests/boards/intel_adsp No Console Output(Timeout)

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/boards/intel_adsp/ --no-skipped-report -vv
4. see error

**Logs and console output**
```
INFO:cavs-client:Monitor process
INFO:cavs-client:Sent:     start_log
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
--
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
E:  **  PC 0xbe00b5cf VADDR 0x71a62
E:  **  PS 0x60020
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
E:  **  A0 0xbe00c68a  SP 0xbe0165c0  A2 0xbe01309c  A3 0xbe011300
E:  **  A4 0xffff80ff  A5 0xbe014db0  A6 0x1  A7 0x9e017538
E:  **  A8 0x71a62  A9 0x80ff A10 0xbe01309c A11 0x800000
E:  ** A12 0xbe0112d4 A13 (nil) A14 0x9e0173a0 A15 0x60925
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR (nil)
Backtrace:0xBE00B5CC:0xBE0165C0 0xBE00C687:0xBE0165F0 0xBE00C729:0xBE016610 0xBE0094C3:0xBE016630 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x9e0173a0 (main)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 14.1
 - Commit SHA: b39d897b0972d26914dfc5614b5287d0476f7755

nashif, 2022-08-23:
You need to be specific about which test actually failed, this is a directory with many tests.

nashif, 2022-08-23:
bisected to b39d897b0972d26914dfc5614b5287d0476f7755, @ceolin

PR #49483 (2022-08-24): intel_adsp: gpdma: Fix driver initialiaziation
cavs15 does not need to request ownership of the dma controller.

Fixes #49203
commit 180f50fe (2022-08-24): intel_adsp: gpdma: Fix driver initialiaziation

cavs15 does not need to request ownership of the dma controller.

Fixes #49203
closed 2022-08-24 by nashif