zephyrproject-rtos/zephyr #49213: logging.add.log_user test fails when compiled with GCC 12

stephanosio, 2022-08-18:
**Describe the bug**

`logging.add.log_user` test fails when compiled with GCC 12, which is the toolchain included in the Zephyr SDK 0.15.0.

Note that it currently fails on all emulate-able platforms including, but not limited to, `qemu_x86`, `qemu_x86_64`, `qemu_riscv32`, `qemu_riscv64` and `mps2_an521`.

**To Reproduce**

`scripts/twister -v -N -M -s tests/subsys/logging/log_core_additional/logging.add.log_user`

**Expected behavior**

Test passes.

**Impact**

User mode logging is broken.

**Logs and console output**

_qemu_riscv32_

```
^MRunning TESTSUITE test_log_core_additional
^M===================================================================
^MSTART - test_log_from_user
^M[00:00:00.000,000] ESC[1;31m<err> os: ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:  mcause: 5, Load access faultESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:   mtval: 800244fcESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a0: 00000001    t0: 00000183ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a1: 00000000    t1: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a2: 00000000    t2: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a3: 00000000    t3: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a4: 00000001    t4: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a5: 80001122    t5: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a6: 00000010    t6: 00000000ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      a7: 2000801fESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      sp: 80022f70ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      ra: 80001132ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:    mepc: 80001136ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: mstatus: 00000080ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: Current thread: 0x800205f8 (test_log_from_user)ESC[0m
[00:00:00.020,000] ESC[1;31m<err> os: Halting systemESC[0m
```

_qemu_x86_

```
^MRunning TESTSUITE test_log_core_additional
^M===================================================================
^MSTART - test_log_from_user
^M[00:00:00.000,000] ESC[1;31m<err> os: Page fault at address 0x139020 (error code 0x4)ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: Linear address not present in page tablesESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: Access violation: user thread not allowed to readESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: PTE: not presentESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: EAX: 0x00000000, EBX: 0x0011a000, ECX: 0x00133fc8, EDX: 0x0000002bESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: ESI: 0x00000000, EDI: 0x00133fe8, EBP: 0x00133fb8, ESP: 0x00133f90ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: EFLAGS: 0x00000206 CS: 0x002b CR3: 0x0011c8a0ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: call trace:ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: EIP: 0x001008d2ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      0x00108451 (0xaaaaaaaa)ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os:      0x00101ef5 (0x11a000)ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0ESC[0m
[00:00:00.000,000] ESC[1;31m<err> os: Current thread: 0x11b120 (test_log_from_user)ESC[0m
[00:00:00.010,000] ESC[1;31m<err> os: Halting systemESC[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: 9a7e4b162d4650520532808f9c94fca257bb631b

**Additional context**

This test passes on all emulated platforms when compiled using the GCC 10.3 from Zephyr SDK 0.14.2.

gmarull, 2022-08-30:
changed priority to high since it is blocking some PRs due to CI failures

stephanosio, 2022-08-30:
> changed priority to high since it is blocking some PRs due to CI failures

This test has been disabled and should not be running in the CI:
https://github.com/zephyrproject-rtos/zephyr/commit/8e8b56881feb9ed68b3435aa67436abc8a2a8b0f

Something seems to be wrong with twister ...

gbarkadiusz, 2024-02-14:
It is still reproducible.

jonathonpenix, 2024-03-15:
Hi @gbarkadiusz and @stephanosio, was this intentionally closed or is this still an open issue? I think the snippet of code disabling the test is still present and I still see these failures if I re-enable it (my Zephyr is at ddb147d0a45df2b52af1065a49770566847ab975 and I'm using SDK v0.16.5). https://github.com/zephyrproject-rtos/zephyr/blob/ddb147d0a45df2b52af1065a49770566847ab975/tests/subsys/logging/log_core_additional/testcase.yaml#L18-L26

Additionally, I am also seeing similar failures when building with clang/llvm for a handful of the `qemu` boards mentioned above so I'd a) be interested in knowing why this was closed to see if I can similarly disable this test for clang or b) be willing to look into this if it is still an open issue. 

Thanks!

nashif, 2024-03-21:
@dcpleung  Can you please take a look at this, go a bit back in history and see why this failed with gcc 12 and why it now fails with other toolchains as well? Seems to be userspace related. I do not see how this could be a toolchain issue, it looks like now it is failing because of something else.

nashif, 2024-03-21:
we need to cleanup the testcase.yaml as well and stop excluding this for the wrong reasons.

dcpleung, 2024-03-21:
AFAICT, the tests are failing because the logging subsystem is trying to read global data stashed into sections like log_mpsc_pbuf_area or log_dynamic_area (basically the .data section), while still running in user mode. The old v1 had a bunch of `*_from_user` logging functions which eventually did syscalls. However, v2 does not seem to have this mechanism. For example, `log_generic()` eventually calls `z_log_msg_runtime_vcreate()` which is not a syscall, and is accessing kernel only data. So to properly fix this, logging subsys needs to support userspace as a whole and not just selected functions.

closed 2024-06-04 by github-actions[bot]