zephyrproject-rtos/zephyr #49216: Would like to ensure all data is sent over the air on bt_conn_disconnect.

evenl, 2022-08-18:
It would be nice to have a way to do a gracefully disconnect of a ble connection, by that I mean that any data queued in the stack should be sent over the air before the connection is disconnected. 

In our case we disconnects at regular interval because we use the radio to do some sync operation, when the sync operation is complete we start advertising again and then the client connects again. While connected we stream data and we want to ensure that all of the packets in the queue is sent over the air before we actually disconnect. 

I really don't care how this is implemented, it could be a parameter to bt_conn_disconnect to tell it to flush the data before returning, or a separate function to check the status/flush the data or a callback.

.even

evenl, 2022-08-19:
I created a solution where I count how many bytes I put into the stack, and then count how many bytes are sent on air using the complete callback that was suggested by alwa-nordic as a workaround. And it work as intended, I don't get any dropped packages any more. But it would be really nice if this was part of the API, and not something I had to implement my self.

jori-nordic, 2022-08-19:
I have more pressing bugs to tend to in the near future, but I'd gladly review a PR if you want to make one :)
Else, I think the ETA would be next year maybe? I can't make any promises yet.

evenl, 2022-08-19:
Sure, I can see if I can manage to find some time to create a patch