zephyrproject-rtos/zephyr #49251: STM32 HW TIMER + DMA + DAC

xkisahero, 2022-08-19:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/49202

<div type='discussions-op-text'>

<sup>Originally posted by **xkisahero** August 18, 2022</sup>
Hi everyone,

    I'm developing a medical device. I need to generate alarm waveforms that meet the Standard IEC 60601-1-8.

    I need to set the sampling frequency to 20000 Hz, so I need a hardware timer. It is preferable to use DMA function, then send alarm wave data from memory to DAC.

    I already have the alarm waveform data in memory , and I can use DAC. I checked a lot of information, but I still can't use the hardware timer. In particular, the combination of hardware timer and DMA and DAC makes me confused.

    I also tried stm32cube Hal library in zephyr. But I don't know how to combine Hal library interrupts with zephyr.

    We ported the code of STM32 from keil platform to zephyr platform because Nordic uses zephyr. We hope that the code can run on the same platform. Now, other functions have been ported, only this one left.

    Please help me, thank you.</div>

erwango, 2022-08-24:
@xkisahero 

From what I understood you managed to get your application running using HAL in STM32Cube environment (Btw can you specify the chip you're using ?). The missing link is how to get HAL callbacks called using Zephyr. 

If I'm correct, here is a way to achieve that:
```
void foo_hal_isr(void *arg)
{
      HAL_FOO_IRQHandler(FOO);
}

	IRQ_CONNECT(FOO_IRQn, 1, foo_hal_isr, 0, 0);
	irq_enable(FOO_IRQn);
```

Please let me know if not working or if I understood your request in a wrong way.

erwango, 2022-08-31:
@xkisahero did it help ?

xkisahero, 2022-09-06:
@erwango Hi erwango. I am use STM32G071RBT6, and hardware timer6. I tried your merhod also many other methods, but I still didn't receive the interrupt from timer6. How can I use just zephyr (not STM32Cube) receive timer6 interrupt？thank you.

erwango, 2022-09-06:
> @erwango Hi erwango. I am use STM32G071RBT6, and hardware timer6. I tried your merhod also many other methods, but I still didn't receive the interrupt from timer6. How can I use just zephyr (not STM32Cube) receive timer6 interrupt？thank you.

There are a lot of possible reasons. We'd need more information on the code you're using to answer.
Here are a couple of ideas:
- Are you able to get Timer6 IRQ with the piece of code I provided in a simple use case without using DAC/DMA ?
- Note that TIM6 is sharing IRQ line with DAC. This may explain why you're having issues in your use case.
I'd suggest testing other timer instance.
- You can also have a try to use TIM6 with PWM, just to verify timer6 is functional with Zephyr.

xkisahero, 2022-09-08:
> 

Hi erwango. Thank you for your reply. In fact, I only need a hardware timer example project with stm32. Then I can finish the rest of the work by myself. But I can't find such code on Google. I tried to refer to Nordic's code, but their code used their own library file. 

Can you provide a simple sample file? you can use any STM32 MCU and any hardware timer. Please help me, thank you.

xkisahero, 2022-09-08:
@erwango Hi erwango. Thank you for your reply. In fact, I only need a hardware timer example project with stm32. Then I can finish the rest of the work by myself. But I can't find such code on Google. I tried to refer to Nordic's code, but their code used their own library file. 

Can you provide a simple sample file? you can use any STM32 MCU and any hardware timer. Please help me, thank you.

erwango, 2022-09-08:
> @erwango Hi erwango. Thank you for your reply. In fact, I only need a hardware timer example project with stm32. Then I can finish the rest of the work by myself. But I can't find such code on [...]

You can look at `drivers/counter/counter_ll_stm32_timer.c` or `drivers/pwm/pwm_stm32.c` that both use timer IP.

closed 2022-09-16 by xkisahero

erwango, 2022-09-16:
@xkisahero Thank for closing the point. Did you manage to solve your point ?
If yes, can you share what was the blocking issue and how you solved it ?