zephyrproject-rtos/zephyr #49268: tests: samples/boards/stm32/power_mgmt/serial_wakeup failed on mec15xxevb_assy6853

hunterhu3000, 2022-08-19:
The daily twister tester will run samples/boards/stm32/power_mgmt/serial_wakeup on mec15xxevb_assy6853 and got a failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? -> mec15xxevb_assy6853
 
 - What have you tried to diagnose or workaround this issue?  
 Check the readme.rst in the test folder, and found this test case seems to be dedicated to stm32. Regarding to mec15xxevb_assy6853 has its own power_mgmt test case, this case should add "platform_alow: stm32" in the sample.yaml.

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - Not a regression. It looks like a small bug in the commit f5378424f7f492a8ae0d173a015bc1186fcc7127.

**To Reproduce**
Steps to reproduce the behavior:
twister --hardware-map $HOME/mec1.file --device-testing -T samples/boards/stm32/power_mgmt/serial_wakeup/ -vv

**Expected behavior**
The test should be skipped

**Impact**
1 test case fail in daily test

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-3848-g02b23f3733e0  ***
DEBUG   - DEVICE: Device ready
DEBUG   - DEVICE: Device is not wakeup capable
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.14.1
 - Commit SHA or Version used :02b23f3733e0e052775a7480ff1e7f6f7971f4e6

hunterhu3000, 2022-08-19:
@erwango Is this test case for STM32 only?
```
commit f5378424f7f492a8ae0d173a015bc1186fcc7127
Author: Erwan Gouriou <erwan.gouriou@linaro.org>
Date:   Tue Jun 28 16:54:49 2022 +0200

    samples: boards: stm32: Add a serial wakeup sample

    Add a sample to demonstrate use of (LP)U(S)ART with wakeup capability
    This sample is provided with 2 configurations
    - nucleo_wb55rg: Basic configuration that could be exercised in CI
    - stm32l562e_dk: "Full feature" configuration that require some set up
    configuration (be able to display console from lpuart) and allows to
    achieve lowest power consumption.

    Signed-off-by: Erwan Gouriou <erwan.gouriou@linaro.org>
```

FRASTM, 2022-08-19:
This is also seen on nucleo_wl55jc and b_u585i_iot02a and others (stm32L5, stm32L4, stm32G0 boards)

closed 2022-08-23 by fabiobaltieri