zephyrproject-rtos/zephyr #49313: NRF51822 sometimes hard fault on connect

korneliuszo, 2022-08-21:
**Describe the bug**

I write code https://github.com/korneliuszo/lumiax-controller/
and it sometimes hardfaults on connection

**To Reproduce**
./scripts/tlay2_server.py <ttyUSB> - to connect to my UART protocol
./scripts/aaa.py - for trigger loop

**Expected behavior**
not crashing

**Impact**
It makes my device unusable

**Logs and console output**

It breaks on assert:
https://github.com/zephyrproject-rtos/zephyr/blob/6fe1299c33994329ebfba5c4c94caae57f43b79f/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan.c#L1195

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK=0.14.2
 - Commit 6fe1299c33994329ebfba5c4c94caae57f43b79f

korneliuszo, 2022-08-22:
Tried to run bt code only after bt_ready callback from bt_enable function.
Didn't help.

Stack:
`
Thread #8 536877040 (Name: main, prio:0,useropts:1) (Suspended : User Request)	
	arch_system_halt() at fatal.c:34 0x1a2d8	
	k_sys_fatal_error_handler() at fatal.c:46 0x1a2e0	
	z_fatal_error() at fatal.c:131 0x1a2fa	
	z_arm_fatal_error() at fatal.c:63 0x172ca	
	z_arm_fault() at fault.c:1,094 0x1f6c	
	z_arm_hard_fault() at fault_s.S:102 0x1fa6	
	<signal handler called>() at 0xfffffff1	
	isr_rx_pdu() at lll_scan.c:1,195 0x107f8	
	isr_rx() at lll_scan.c:766 0x107f8	
	isr_radio() at radio.c:140 0x11344	
	radio_nrf5_isr_body() at lll.c:93 0xf45a	
	radio_nrf5_isr() at lll.c:87 0xf45a	
	<signal handler called>() at 0xfffffffd	
	0xe500e7ea	
	0x80fffe	
`

carlescufi, 2022-08-24:
@korneliuszo can you come up with a simple application that reproduces the problem? Your project is not trivial to set up.

korneliuszo, 2022-08-24:
[hang.zip](https://github.com/zephyrproject-rtos/zephyr/files/9417307/hang.zip)
Probably beacuse of this happens.
I use irqlock in fifo.hpp for atomically read 2 pointers.

closed 2022-11-17 by github-actions[bot]