zephyrproject-rtos/zephyr #49329: twister: frdm_k64f: test string mismatch

shaoanxli, 2022-08-22:
**Describe the bug**
Unknown failure occurs with the parameter `--retry-build-errors`, this bug can not reproduce.
Take cpp.main.newlib.set_promisc_mode_off failure as an example.

The test shows pass in handler.log, but failed in twister.log.

**Impact**
The pass rate drop, and some cases fail in twister.log.

**Logs and console output**
Can not fine the test case tests/subsys/cpp/cxx/src/main.cpp here, but tests/net/iface/src/main.c.
```
*** Booting Zephyr OS build zephyr-v3.1.0-3949-g5851a7b7de27  ***
Running TESTSUITE cxx_tests
===================================================================
START - test_global_static_ctor
PASS - test_global_static_ctor in 0.001 seconds
===================================================================
START - test_global_static_ctor_dynmem
PASS - test_global_static_ctor_dynmem in 0.001 seconds
===================================================================
START - test_new_delete
PASS - test_new_delete in 0.001 seconds
===================================================================
TESTSUITE cxx_tests succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [cxx_tests]: pass = 3, fail = 0, skip = 0, total = 3 duration = 0.003 seconds
- PASS - [cxx_tests.test_global_static_ctor] duration = 0.001 seconds
- PASS - [cxx_tests.test_global_static_ctor_dynmem] duration = 0.001 seconds
- PASS - [cxx_tests.test_new_delete] duration = 0.001 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 64640511c8a9a61490034639ae8449c6
PROJECT EXECUTION SUCCESSFUL

```
**Environment :**
 - OS: Linux,
 - Toolchain Zephyr SDK 0.14.2
 - Commit SHA or Version : 5851a7b7de27

nashif, 2022-08-23:
The bug description is not clear on how to reproduce or what the the bug exactly is. Please provide more information.

shaoanxli, 2022-08-24:
> The bug description is not clear on how to reproduce or what the the bug exactly is. Please provide more information.

This bug is unable to reproduce stably, but occurs with command "twister -p frdm_k64f --device-testing --hardware-map /home/ztest/map.yml -T tests/ -T samples/ --retry-failed 2 --retry-build-errors --no-skipped-report -e cmsis_dsp".

The failure is that the failure shows unknown failure is cpp.main.newlib.set_promisc_mode_off but the actual case should be net.iface.set_promisc_mode_off, so the twister judges that set_promisc_mode_off is the case below cpp.main.newlib is problematic.

hwang37, 2022-09-06:
@shaoanxli could you provide more clarification, reproduceable steps, etc?

mengxianglinx, 2022-09-07:
This is duplicate with #47127, will be resolved once we start getting testcase lists from binaries.

closed 2022-09-07 by mengxianglinx