zephyrproject-rtos/zephyr #49338: Antenna switching for Bluetooth direction finding with the nRF5340

roblittlejohns, 2022-08-22:
**Describe the bug**
Using the ant_patterns array in the direction_finding_connectionless_rx Bluetooth sample application it does not appear possible to achieve the desired switching sequence. I am using the nRF5340DK platform and using the zephyr-v3.1.0 tag.

The nRF5340 requires multiple writes to the SWITCHPATTERN register to set the switching pattern. The first write (index 0) sets the PDU RX/TX antenna pattern, second write (index 1) sets the guard and reference antenna pattern and the remaining indices (index 2+) set the switching pattern that follows. After the pattern list is exhausted it returns to index 2 of the switch pattern list.

The ant_patterns array provided in the main.c of the application is passed through to the radio_df_ant_switch_pattern_set function of subsys\bluetooth\controller\ll_sw\nordic\hal\nrf5\radio\radio_df.c. This sets index 0 of SWITCHPATTERN as the dfe-pdu-antenna device tree property value. Indices 1 to length(ant_patterns) of SWITCHPATTERN are set from the ant_patterns array. A last write is then performed (line 119) to the SWITCHPATTERN register which I believe isn't needed as this just adds an extra entry to the switching pattern. By removing this line I get the expected behaviour where the ant_patterns array is used to set the guard/reference antenna pattern and the antenna switching sequence.

**Expected behavior**
The ant_patterns array is used to set the guard/reference antenna pattern followed by the antenna switching sequence. The dfe-pdu-antenna device tree property value sets the PDU RX/TX antenna pattern.

**Impact**
Workaround by commenting out line 119.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 0.15.0
 - Zephyr Tag: zephyr-v3.1.0

ppryga-nordic, 2022-11-21:
Hello, I'm sorry for so long time to provide a feedback on the issue.

The described behavior is correct, meaning compliant with Bluetooth specification: BT 5.3. Vol 6, Part A  section 5.1 Antenna Switching General Description:

> If the pattern specified by the Host is exhausted before the last sample slot, it
shall be restarted from the beginning (this can happen more than once); that is,
the first antenna in the pattern is used in the sample slot following that used for
the last antenna in the pattern. If the pattern has not been completely used by
the end of the Constant Tone Extension, any remaining terms shall be ignored.

The specification says that complete switching pattern is used every time a pattern is exhausted, hence we have to get back to `SWITCHPATTERN[1]` (the guard\reference period antenna). The line 119 mentioned by @rob-lj was introduced in a PR #45432 to make the implementation Bluetooth specification compliant.

There is no bug in the implementation.

@rob-lj have you got any objections about this? If not then I'll close the issue.

carlescufi, 2022-11-23:
Closing as per @ppryga-nordic's response. Please reopen if required.

closed 2022-11-23 by carlescufi