zephyrproject-rtos/zephyr #49342: Zephyr hci_usb sample cannot use LE coded phy

eriklins, 2022-08-22:
Hey guys,

I built the hci_usb sample coming with Zephyr for the Laird BL654 USB dongle. Build process works and the dongle enumerates as  BT device on USB. Testing this on a VirtualBox running Ubuntu 20.04.

"hcitool dev" shows the device as "hci0"
"hcitool lescan" shows legacy 1MPhy BLE adverts.

However, switching to LE coded phy has some flaws.

It's not possible to switch phy to LE coded with btmgmt tool:

erik@erik-VirtualBox:~/zephyrproject/zephyr$ sudo btmgmt
[mgmt]# phy
Supported phys: LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX 
Configurable phys: LE2MTX LE2MRX LECODEDTX LECODEDRX 
Selected phys: LE1MTX LE1MRX 
[mgmt]# phy LECODEDTX LECODEDRX
Could not set PHY Configuration with status 0x0d (Invalid Parameters)

But it's possible to switch to LE coded phy with hcitool:

erik@erik-VirtualBox:~/zephyrproject/zephyr$ sudo hcitool cmd 08 32
< HCI Command: ogf 0x08, ocf 0x0032, plen 0
> HCI Event: 0x0f plen 4
  02 01 32 20 
erik@erik-VirtualBox:~/zephyrproject/zephyr$ sudo hcitool cmd 08 31 03 04 04
< HCI Command: ogf 0x08, ocf 0x0031, plen 3
  03 04 04 
> HCI Event: 0x0e plen 4
  01 31 20 00 

After doing so btmgmt shows the LE coded phy as selected

erik@erik-VirtualBox:~/zephyrproject/zephyr$ sudo btmgmt
[mgmt]# phy
Supported phys: LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX 
Configurable phys: LE2MTX LE2MRX LECODEDTX LECODEDRX 
Selected phys: LECODEDTX LECODEDRX 

However, hcitool lescan still just shows legacy 1MPhy adverts and no LE coded adverts.

On the other hand it's possible to set phy back to 1MPhy with btmgmt:

erik@erik-VirtualBox:~/zephyrproject/zephyr$ sudo btmgmt
[mgmt]# phy LE1MTX LE1MRX
PHY Configuration successfully set

I tried first with the default prj.conf and later added the below lines, which does not change anything.

CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_AUTO_PHY_UPDATE=y
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_LL_SW_SPLIT=y

All this on Ubuntu 20.04 VirtualBox (Windows 10 host), blueZ version 5.53-0ubuntu3.6.

The above might be related to issue #28145 (https://github.com/zephyrproject-rtos/zephyr/issues/28145) which desribes the same behaviour and was marked closed two years ago.

carlescufi, 2022-08-24:
@eriklins can you please capture a `btmon` trace and attach it here?

carlescufi, 2022-08-24:
cc @sjanc does this ring a bell?

eriklins, 2022-08-24:
Hello @carlescufi ,
see attached files. One is btmon trace, second are the commands issued. First switched phy to LE coded with hcitool, then did a scan, which only showed legacy 1MPhy adverts and then used btmgmt tool to check phy, switch back to 1MPhy (which works) and then again to LE coded phy (which does not work).

[btmon_220824.txt](https://github.com/zephyrproject-rtos/zephyr/files/9415671/btmon_220824.txt)
[commands_220824.txt](https://github.com/zephyrproject-rtos/zephyr/files/9415672/commands_220824.txt)

nordicjm, 2022-08-24:
@eriklins I do not see anywhere in the log where btmgmt is actually sending the commands to the device, that error looks to be coming from btmgmt itself, not from the module. And it seems that there is mention of this issue from many years ago https://www.spinics.net/lists/linux-bluetooth/msg76869.html might want to report this bug to blueZ

eriklins, 2022-08-24:
However, the hcitool commands issued seem to work and seem to properly set phy to LE coded. But hcitool lescan do not show any LE coded adverts just legacy 1MPhy adverts. So, should not be related to btmgmt.

nordicjm, 2022-08-24:
hcitool has been deprecated for many years, it should in no way, shape or form continue to be used, the blueZ maintainers have said you should migrate to bluetoothctl or other supported utilities

eriklins, 2022-08-24:
I confirm "phy LE1MTX LECODEDTX LE1MRX LECODEDRX" in btmgmt seem to work and does not throw an error. Would this enable both phys? Scanning with bluetoothctl afterwards still only shows 1MPhy adverts and no LE coded adverts.

nordicjm, 2022-08-24:
> I confirm "phy LE1MTX LECODEDTX LE1MRX LECODEDRX" in btmgmt seem to work and does not throw an error. Would this enable both phys? Scanning with bluetoothctl afterwards still only shows 1MPhy adverts and no LE coded adverts.

Enable yes, but to scan on the other PHY requires using the extended scanning commands, I would assume bluetoothctl does not have support for these, so is limited to scanning on 1M PHY only. Not sure if there are any libraries e.g. for python that would support scanning over LE coded PHY

eriklins, 2022-08-24:
So, the fact that btmgmt command "phy LECODEDTX LECODEDRX" to solely enable LE coded phy does not work would not be considered a bug?

nordicjm, 2022-08-24:
> So, the fact that btmgmt command "phy LECODEDTX LECODEDRX" to solely enable LE coded phy does not work would not be considered a bug?

That is not a bug, no, 1M PHY support is mandatory, LE coded is optional. You could try asking on the blueZ mailing list about support for extended scanning i.e. with LE coded PHY

eriklins, 2022-08-24:
Okay, so the issue doesn't seem to be related to the usb_hci implementation and hence issue could be closed. Thanks everybody.

closed 2022-08-24 by eriklins