zephyrproject-rtos/zephyr #49359: nrf5*: crash when Bluetooth advertisements and flash write/erase are used simultaneously

mniestroj, 2022-08-22:
**Describe the bug**

nrf52840 platform (but probably whole nrf5 family) crashes when both Bluetooth advertisements are enabled and there are some flash write/erase operations done from application thread (like DFU operation). There is a pointer dereferencing done in "ticker" callback (as part of nrf flash driver), which dereferences thread stack that might no longer be valid, which results in system crashes.

Please also mention any information which could help others to understand the problem you're facing:
 - What target platform are you using? - `nrf52840` based devices, but this whole `nrf5*` family might be affected
 - What have you tried to diagnose or workaround this issue? - workarounds are part of referenced branch that contains sample that allows to reproduce this issue; however provided workaround might not be enough, as the target application still crashes from time to time
 - Is this a regression? If yes, have you been able to "git bisect" it to a specific commit? - probably Zephyr 2.7.0 was already affected (could try to confirm if that information is valuable), but not sure about earlier versions; i.e. it does NOT look like a recent regression.

**To Reproduce**
Steps to reproduce the behavior:
1. `west init -m https://github.com/mniestroj/zephyr.git --mr nrf-flash-bt-issue` (1 commit includes `nrf_flash_bt_issue` sample to reproduce this issue easily; other commits just add logs/workaround for the issue)
2. `west build -b nrf52840dk_nrf52840 zephyr/samples/nrf_flash_bt_issue`
3. `west flash`
4. Attach console over UART to see logging messages
5. See warning messages from `soc_flash_nrf_ticker` module:

```
*** Booting Zephyr OS build zephyr-v3.1.0-3968-g5327d9c4dce5  ***
[00:00:00.254,211] <inf> bt: Bluetooth initialized
[00:00:00.254,730] <inf> bt: Advertising started
[00:00:00.255,035] <inf> main: Writing image
[00:00:03.225,189] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:07.756,134] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:12.634,948] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:19.488,311] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:19.813,903] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:22.749,450] <wrn> soc_flash_nrf_ticker: Unbalanced ticker take!
[00:00:31.069,976] <inf> main: Image written
[00:00:31.159,362] <inf> main: Image flushed
[00:00:31.159,362] <inf> main: Rebooting
```

Referenced branch already contains some workarounds and warning log messages that point out the problem and try to mitigate the problem. However checking out (git checkout ...) at the first commit in this branch, which is `samples: nrf_flash_bt_issue: reproduce BT+Flash ticker/mayfly issue` and rebuilding results in Zephyr crashes as shown below (those crashes are not as consistent as the warning messages generated, i.e. system crashes with approx 20% rate):

```
*** Booting Zephyr OS build zephyr-v3.1.0-3970-gb891918b9259  ***
[00:00:00.398,040] <inf> bt: Bluetooth initialized
[00:00:00.398,559] <inf> bt: Advertising started
[00:00:00.398,864] <inf> main: Writing image
[00:00:23.057,220] 00:23.057,250] <wrn> main: Failed to write image: -140
[00:00:23.058,807] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.058,807] <wrn> main: Failed to write image: -140
[00:00:23.060,333] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.060,333] <wrn> main: Failed to write image: -140
[00:00:23.061,614] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.061,828] <wrn> main: Failed to write image: -140
[00:00:23.063,110] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.063,110] <wrn> main: Failed to write image: -140
[00:00:23.064,636] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.064,636] <wrn> main: Failed to write image: -140
[00:00:23.066,070] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.066,070] <wrn> main: Failed to write image: -140
[00:00:23.067,352] <err> STREAM_FLASH: flash_write error -140 offset=0x000b6e00
[00:00:23.067,382] <wrn> main: Failed to write image: -140
[00:00:23.068,359] <err> os: ***** USAGE FAULT *****
[00:00:23.068,389] <err> os:   Illegal use of the EPSR
[00:00:23.068,389] <err> os: r0/a1:  0x20006ff4  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:00:23.068,420] <err> os: r3/a4:  0x20006fd0 r12/ip:  0x00000001 r14/lr:  0x00016fb7
[00:00:23.068,420] <err> os:  xpsr:  0x2000001b
[00:00:23.068,450] <err> os: Faulting instruction address (r15/pc): 0x00000002
[00:00:23.068,450] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:23.068,481] <err> os: Fault during interrupt handling

[00:00:23.068,481] <err> os: Current thread: 0x20000a08 (unknown)
[00:00:24.333,068] <err> os: Halting system
```

**Expected behavior**
There should be no warning messages about "unbalanced" k_sem_give/k_sem_take. Those messages just point out that one of the problems is that `time_slot_callback_work()` and/or `time_slot_delay()` are called even after `nrf_flash_sync_exe()` already finished execution. As a result, in body of `time_slot_callback_work()` callback, there is context pointer dereferenced as `struct flash_op_desc *op_desc` with `rc = op_desc->handler(op_desc->context);` code line. As `struct flash_op_desc` is created on thread stack (https://github.com/zephyrproject-rtos/zephyr/blob/fda3f54002408f7cb5a28285b89ce3fd9300f238/drivers/flash/soc_flash_nrf.c#L299 or https://github.com/zephyrproject-rtos/zephyr/blob/fda3f54002408f7cb5a28285b89ce3fd9300f238/drivers/flash/soc_flash_nrf.c#L317), which tries to write/erase flash, this stack variable might no longer be valid (stack might get unwound at this point).

**Impact**
Showstopper for using Bluetooth (at least in advertisement "state") and flash write/erase at the same time. This means that any device built around nrf5* board that uses Bluetooth and allows to operate on flash (like saving settings, doing firmware upgrade, etc) is potentially buggy.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0-beta2 (but issue was reproducible with older SDKs as well)
 - Commit SHA or Version used: tip of main (32a87215d7aa97191097357f602237f7da4a3536), 3.1.0; probably 3.0.0, 2.7.0 (did not confirm that)

**Additional context**
This issue references a minimal sample application that allows to reproduce this issue. This sample application does following things that seem important to reproduce this issue:
- Bluetooth advertisements enabled,
- Flash erase/write from application thread,
- Some (in this example `k_timer`) interrupts firing in the background.
- (optional, but improves reproducibility rate) `CONFIG_IMG_ERASE_PROGRESSIVELY=y` option enabled

The real problem is with downstream application, that uses nrf52840 + esp-at WiFi driver. System crashes when saving new firmware during DFU (Device Firmware Upgrade) downloaded over WiFi.

cvinayak, 2022-09-01:
@mniestroj I would like you to check somethingL

Use lower priorities for other SoC peripherals used in your application by having the below

#define NRF_DEFAULT_IRQ_PRIORITY 5

inside your application folder in dts/arm/nordic/override.dtsi file.

and then check the zephyr.dts file, the interrupt priorities should be 5 instead of 1.

Meanwhile, I will try to reproduce it at my end today.

cvinayak, 2022-09-01:
@mniestroj Thank you for providing a simple sample, it has helped me reproduce the issue, and here is the fix #49791. Please verify and revert back.

PR #49791 (2022-09-05): drivers: flash: nrf: Fix ticker stop synchronization
Fix usage fault due to spurious ticker timeout expiry post
enqueuing of ticker stop operation.

Use ticker operation callback to handle completion of ticker
stop operation and then give the semaphore to thread to
notifying the completion of flash operation.

Fixes #49359.
closed 2022-09-05 by carlescufi