zephyrproject-rtos/zephyr #49360: west boards doesn't print boards from modules

ktaborowski, 2022-08-22:
The `wset boards` command returns just a boards form zephyr/boards folder, but the ones form external modules aren't print.
e.g. there are no board form https://github.com/nrfconnect/sdk-nrf/tree/main/boards despite the folder is mention in https://github.com/nrfconnect/sdk-nrf/blob/main/zephyr/module.yml
I look into code and it looks like it should print board form external modules as well
https://github.com/zephyrproject-rtos/zephyr/blob/a202341958c074d4e10e576adf0bdc67ff49cbea/scripts/west_commands/boards.py#L80

fabiobaltieri, 2022-08-23:
This should be fixed upstream.

closed 2022-08-23 by fabiobaltieri

carlescufi, 2022-08-23:
@ktaborowski this was fixed by this PR:
https://github.com/zephyrproject-rtos/zephyr/pull/48811