zephyrproject-rtos/zephyr #49390: shell_rtt thread can starve other threads of the same priority

julian-yeoh-setec, 2022-08-23:
Under certain circumstances the shell_rtt thread can cause other threads of the same priority to stop running. An example buildable example project that that demonstrates the problem can be found at the following link:
[https://github.com/julian-yeoh-setec/shell_rtt_starves_other_threads](https://github.com/julian-yeoh-setec/shell_rtt_starves_other_threads)

Please also mention any information which could help others to understand the problem you're facing:
 - Tested on NRF52832_XXAA
 - This problem occurs if the shell_rtt thread and another thread are configured to use the same preemptible priority. In the example buildable project, I left the priority of the shell_rtt thread at its default value of 14 and I started a work queue thread that uses the same priority 
 - If I make the priority of the work queue thread higher than the priority of the shell_rtt or give it a cooperative (-) priority, then the problems seems to go away
 - Narrowed down to the `z_shell_write()` function's `while (length)` loop spinning indefinitely until I connect RTT Viewer 
 - I tested on Zephyr 2.7 and Zephyr 3.1. The issue is present on both versions

Build steps:
1. west update
2. west build app

Steps to reproduce the behavior:
1. Make sure RTT Viewer or any other terminals are NOT connected
2. Start the debugger (make sure the debugging environment does not have a built in terminal - otherwise disable the terminal)
3. Let the debugger run for a few seconds
4. Place a breakpoint in the  `work_handler()` function in the example file `main.c` 
5. The breakpoint will not be reached, indicating that the work queue's thread is not longer running. `work_handler()` should be called every 1 second but does not
6. Connect RTT Viewer or other terminal
7. The breakpoint will now be reached
8. After this step, even if the terminal is disconnected, it seems that `work_handler()` continues to be called correctly every 1 second

Expected behavior:
The thread that shares its priority with the shell_rtt thread should not stall if no terminal that can consume log messages is connected. Or is it actually an undocumented requirement that we should never configure a thread to have the same priority as the shell_rtt thread?

Impact:
Causes thread to lockup if terminal is not connected and if there are log messages to be displayed.

Environment:
 - OS: Windows
 - Zephyr 3.1
 - Zephyr SHA-1: 2ddd73feafd3316af2c547c34d6969bea637d5c6

julian-yeoh-setec, 2022-10-23:
Is it possible for someone to remove the Stale label? I can't seem to remove it probably because I lack the necessary permissions. Thanks in advance.

julian-yeoh-setec, 2023-03-05:
Can we please remove the Stale label? Thanks very much.

rene-becker-setec, 2023-05-08:
@julian-yeoh-setec , could we check if later versions still have this problem?
Thanks

julian-yeoh-setec, 2024-01-19:
Can we please remove the Stale label? Thanks very much

nordic-krch, 2024-02-13:
@julian-yeoh-setec I know It's been a while but #/68941 shall fix it.

julian-yeoh-setec, 2024-02-14:
> @julian-yeoh-setec I know It's been a while but #/68941 shall fix it.

Thanks for the update! Appreciate it

PR #68941 (2024-03-28): shell: rtt: Add detection of host presence
If host is not reading RTT data (because there is no PC connection or RTT reading application is not running on the host), thread will stuck continuously trying to write to RTT. All threads with equal or lower priority are blocked then. Adding detection of that case and if host is not reading data for configurable period then data is dropped until host accepts new data.

Fixes #49390.
PR #70851 (2024-04-03): [Backport v3.6-branch] shell: rtt: Add detection of host presence
Backport 04a74ce107e5576d7c114cfba4066e7b854bcde6 from #68941.

Fixes: #49390