zephyrproject-rtos/zephyr #49410: Bluetooth: Scan responses with info about periodic adv. sometimes stops being reported

jakkra, 2022-08-23:
**Describe the bug**
Sometimes after for example a sync termination then no new advertisments with periodic adv. are reported in the `bt_le_scan_cb recv` callback. Restarting scanning will result again with scan responses containing info about periodic adv.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? - NRF52833
 - Is this a regression? - Yes a regression. No bisect made, but it works with the Zephyr version used in nRF Connect SDK 2.0

**To Reproduce**
Build below application based on the `direction_finding_connectionless_rx` sample (the application does not enable CTE sampling by default but can still reproduce the issue).
[aoa_receiver_multiple.zip](https://github.com/zephyrproject-rtos/zephyr/files/9402234/aoa_receiver_multiple.zip)
Have 2 tags sending periodic advertisements (I used about 50ms interval and about 30 bytes per. adv. data payload, this is probably not needed and won't matter) for example use `direction_finding_connectionless_tx` sample, the `periodic_adv` should also work, but I have not tried.
Then just randomly reset the two tags until the log print `Got peridoic adv. scan rsp` disappears. 

**Expected behavior**
Scan responses keep reporting scan responses with periodic advertisements.

**Impact**
Annoyance for application using periodic advertising sync as restart of scan is needed periodically.

**Logs and console output**
```
** Booting Zephyr OS build zephyr-v3.1.0-3844-g0e560bf8e91e  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
Waiting for periodic advertising...
Scan is running...
Got peridoic adv. scan rsp.
(306) Found device sending per adv: 21:D6:DD:B7:69:62 (random)
(306) Creating Periodic Advertising Sync...
success.
Sync index: 0
Got peridoic adv. scan rsp.
Got peridoic adv. scan rsp.
(967) PER_ADV_SYNC[0]: [DEVICE]: 21:D6:DD:B7:69:62 (random) synced, Interval 0x0028 (50 ms), PHY LE 1M
Got peridoic adv. scan rsp.
(1177) Found device sending per adv: 3C:97:D6:E5:A2:40 (random)
(1177) Creating Periodic Advertising Sync...
success.
Sync index: 1
TAGS:0,sync...,-,-,-,-,-,-,-,-,
Got peridoic adv. scan rsp.
(1808) PER_ADV_SYNC[1]: [DEVICE]: 3C:97:D6:E5:A2:40 (random) synced, Interval 0x0028 (50 ms), PHY LE 1M
Got peridoic adv. scan rsp.
TAGS:0,0,-,-,-,-,-,-,-,-,
Got peridoic adv. scan rsp.
Got peridoic adv. scan rsp.
TAGS:0,0,-,-,-,-,-,-,-,-,
Got peridoic adv. scan rsp.
(3343) Found device sending per adv: 0F:46:A7:CE:90:F7 (random)
(3343) Creating Periodic Advertising Sync...
success.
Sync index: 2
Got peridoic adv. scan rsp.
(3784) PER_ADV_SYNC[2]: [DEVICE]: 0F:46:A7:CE:90:F7 (random) synced, Interval 0x0028 (50 ms), PHY LE 1M
TAGS:0,0,0,-,-,-,-,-,-,-,
TAGS:0,0,0,-,-,-,-,-,-,-,
(5460) PER_ADV_SYNC[1]: [DEVICE]: 3C:97:D6:E5:A2:40 (random) sync terminated reason: 31
Device was synced: term_cb: clear device index: 1
(5869) PER_ADV_SYNC[0]: [DEVICE]: 21:D6:DD:B7:69:62 (random) sync terminated reason: 31
Device was synced: term_cb: clear device index: 0
TAGS:-,-,0,-,-,-,-,-,-,-,
Scan is running...
TAGS:-,-,0,-,-,-,-,-,-,-,
TAGS:-,-,0,-,-,-,-,-,-,-,
TAGS:-,-,0,-,-,-,-,-,-,-,
TAGS:-,-,0,-,-,-,-,-,-,-,
(10585) PER_ADV_SYNC[2]: [DEVICE]: 0F:46:A7:CE:90:F7 (random) sync terminated reason: 31
Device was synced: term_cb: clear device index: 2
TAGS:-,-,-,-,-,-,-,-,-,-,
Scan is running...
TAGS:-,-,-,-,-,-,-,-,-,-,
TAGS:-,-,-,-,-,-,-,-,-,-,
TAGS:-,-,-,-,-,-,-,-,-,-,
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 4f84bc8f40311fb7ccd6575790c97b4e7807cc87

jakkra, 2022-08-23:
I have now run through a git bisect and it seems as this commit is the issue: 49abc04d41d811711e462d30e01a22e1bbe3e8b6

cvinayak, 2022-08-23:
@jakkra The culprit commit 49abc04 is not a bug but needed to ensure that periodic advertising sync events are not skipped due to background scan for extended advertisement. As a fix for your application where in you need overlapping auxiliary scan to take precedence over established periodic sync event, I have created this PR #49429 with a new Kconfig that will allow your application to continue to receive extended advertising reports from other peer devices that may have the AUX_ADV_IND PDU that overlap an existing established periodic sync AUX_SYNC_IND PDU.

Use the below in your application alongwith #49429 and let me know if your issue is resolved:
```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_SYNC_PERIODIC_SKIP_ON_SCAN_AUX=y
```

jakkra, 2022-08-24:
Thank you, that seems to have fixed the issue. 
Not related to this but I see much worse per. adv. sync performance on current main branch compared to nRF Connect 2.0 Zephyr version. Will open a new issue with that.

cvinayak, 2022-08-24:
There has been changes in the Periodic Advertising design to satisfy conformance tests, that said, if you can again bisect and let me know a culprit commit(s), I will analyze and provide suitable compromise between application requirement and conformance test requirement.

jakkra, 2022-08-24:
@cvinayak created https://github.com/zephyrproject-rtos/zephyr/issues/49465

closed 2022-08-30 by carlescufi