zephyrproject-rtos/zephyr #49446: STM32 SPI Set transfer direction is fixed

Cherish-Gww, 2022-08-24:
https://github.com/zephyrproject-rtos/zephyr/blob/32e2e1cb799ecbb5676a6f933cc38e8778b1ed45/drivers/spi/spi_ll_stm32.c#L533

SPI full duplex and half duplex need configured by struct spi_config.

Cherish-Gww, 2022-08-24:
@FRASTM

erwango, 2022-08-24:
Can you test https://github.com/zephyrproject-rtos/zephyr/pull/46803?

Cherish-Gww, 2022-08-30:
@erwango 
I have read and tesed #46803. 
there is an issue, I don`t understand why transmit end ( cs pin pull up), there have a word clock. 
I`m looking into the matter.

Cherish-Gww, 2022-08-31:
@erwango 
https://github.com/zephyrproject-rtos/zephyr/pull/46803. 
The author has disappeared for too long. When can this patch be merged。

erwango, 2022-08-31:
> @erwango #46803. The author has disappeared for too long. When can this patch be merged。

This patch will be merged when the review is completed, this depends on the author.
But if you're able to test and improve it (enhance factorization for instance), you can also push a new PR for this feature.

erwango, 2023-10-12:
@Cherish-Gww is this is still required, please have a check to https://github.com/zephyrproject-rtos/zephyr/pull/57993. Author is not around atm, but will show up eventually

PR #75983 (2025-01-09): drivers: spi: stm32 half duplex support
Add support for `SPI_HALF_DUPLEX ` flag for STM32 SPI driver. Tested with stm32u575 with and without `CONFIG_SPI_STM32_INTERRUPT`, and with DMA.

Fixes #49446
closed 2025-01-09 by kartben