zephyrproject-rtos/zephyr #49451: Treat carrier UP/DOWN independently to interface UP/DOWN

krish2718, 2022-08-24:
`net_if_carrier_down` sends `NET_EVENT_IF_DOWN`, and there is no API for carrier up, we only have `net_if_up`. This doesn't suffice for all cases, esp. Wi-Fi. Interface UP/DOWN means network interface is ready to accept commands or process events, where as carrier is tied to the availability of a remote end (In Ethernet cable plugged in, in Wi-Fi association completed).

E.g., If an interface is in Wi-Fi station mode and it is disassociated, driver informs using `net_if_carrier_down` but that will bring the interface down and then there is no way the applications can bring the interface back up as the for the association to be formed application has to be send commands using interface, but interface is down, a catch22.

Having a separate carrier UP/DOWN resolves this, as the interface will still be up but carrier down, application can send commands over the interface to get carrier up.

Both treatment and new events need to be supported.

mniestroj, 2022-08-26:
I think this is related: #28998

krish2718, 2022-08-30:
@rlubos FYI.

rlubos, 2022-09-01:
@mniestroj Indeed the linked issue seems to be covering the same topic, thanks for referencing it.

@krish2718 I suggest we move the discussion into https://github.com/zephyrproject-rtos/zephyr/issues/28998 to avoid duplicates.

krish2718, 2022-09-01:
Sorry, I haven't searched for existing ones. Sure, we can move to that.

rlubos, 2022-10-20:
Resolved with #51240

closed 2022-10-20 by rlubos