zephyrproject-rtos/zephyr #49465: Bluetooth: Controller: Periodic adv. sync. degraded performance on latest main branch

jakkra, 2022-08-24:
**Describe the bug**
In the below table it can be seen that current main branch of Zephyr is not working as we expect. We expect the performance to be as with nRF Connect 2.0 or **_better_**.
This is tested with https://github.com/zephyrproject-rtos/zephyr/pull/49429 code, which is a small addition to the main branch as of today.
15 Tags nRF Connect 2.0             |  15 Tags Zephyr main
:-------------------------:|:-------------------------:
[image]             |  [image]
20 Tags nRF Connect 2.0             |  20 Tags Zephyr main
[image]             |  [image]

One difference I can see in the logs are that for Zephyr main branch I get many prints as below.
That is sync_term callback but the le address is invalid (garbage). 
```
PER_ADV_SYNC[4]: [DEVICE]: 00:D9:AC:00:00:00 (public-id) sync terminated reason: 62
[00:00:17.917,205] <err> u_df_scan: Got sync_term from none queued tag!
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? - Zephyr SDK and NRF52833
 - Is this a regression? - Bisect is in progress...

**To Reproduce**
Build below application based on the `direction_finding_connectionless_rx` sample (the application does not enable CTE sampling by default but can still reproduce the issue).
[aoa_receiver_multiple_configureable.zip](https://github.com/zephyrproject-rtos/zephyr/files/9416562/aoa_receiver_multiple_configureable.zip)

There are some options in the top of the main.c file, default is using periodic adv. list. which is also from where the images above are taken using.

NOTE if you build for other commits than https://github.com/zephyrproject-rtos/zephyr/pull/49429 the line `CONFIG_BT_CTLR_SYNC_PERIODIC_SKIP_ON_SCAN_AUX=y` needs to be commented out in prj.conf

Have about 10 tags sending periodic advertisements (I used about 50ms interval and about 30 bytes per. adv. data payload, this is probably not needed) for example use `direction_finding_connectionless_tx` sample, the `periodic_adv` should also work, but I have not tried.

**Expected behavior**
Equal behaviour or better in Zephyr main branch compared to Zephyr version used in nRF Connect 2.0

**Impact**
Degraded performance when tracking many tags in an application

**Logs and console output**
[ncs_2-0-20-tags.log](https://github.com/zephyrproject-rtos/zephyr/files/9416248/ncs_2-0-log-20-tags.log)
[ncs-2-0-15-tags.log](https://github.com/zephyrproject-rtos/zephyr/files/9416249/ncs-2-0-15-tags.log)
[zephyr-main-15-tags.log](https://github.com/zephyrproject-rtos/zephyr/files/9416250/zephyr-main-15-tags.log)
[zephyr-main-20-tags.log](https://github.com/zephyrproject-rtos/zephyr/files/9416251/zephyr-main-20-tags.log)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Used the code as in https://github.com/zephyrproject-rtos/zephyr/pull/49429

cvinayak, 2022-08-24:
@jakkra Thank you again for analysing.

It does not make sense as it is only switching to new Link Layer Control Procedure implementation as default that only is used in connection oriented usecases (also, for connection oriented direction finding `BT_LL_SW_LLCP` was always the default).

Could you check if reverting https://github.com/zephyrproject-rtos/zephyr/commit/76396496b6d9ea24a1e644565c5225222e63984f help? (I am guessing here based on the only recent changes related to periodic advertising sync).

jakkra, 2022-08-24:
Ok thanks, I'll come back tomorrow with more info, perhaps it was more of a coincidence the performance change in this commit. I'll do some more investigation.

jakkra, 2022-08-25:
@cvinayak commit 76396496b6d9ea24a1e644565c5225222e63984f is indeed the problem, reverting it the behaviour is same/similar to nRF  Connect 2.0.
However I'm still getting some (currupt?):
```
(114925) PER_ADV_SYNC[9]: [DEVICE]: 00:D9:AC:00:00:00 (public-id) sync terminated reason: 62
[00:01:54.925,933] <err> u_df_scan: Got sync_term from none queued tag!
```
I'll backtrack when it started.

jakkra, 2022-08-25:
@cvinayak actually the invalid `PER_ADV_SYNC[9]: [DEVICE]: 00:D9:AC:00:00:00 (public-id) sync terminated reason: 62` is not something new, I see it now on Zephyr version used in nRF Connect 2.0 also. Just that I saw it a lot before reverting 76396496b6d9ea24a1e644565c5225222e63984f and thought it was new.

Do you have any input regarding this? Want me to open a new issue? It's not a huge problem as I for now can handle

cvinayak, 2022-08-25:
@jakkra to confirm, https://github.com/zephyrproject-rtos/zephyr/commit/76396496b6d9ea24a1e644565c5225222e63984f is the only change for which a fix is needed, right? if so, I will upstream a PR to add a Kconfig to permit your application to not use the reduced time reservations for periodic sync PDU reception.

`PER_ADV_SYNC[9]: [DEVICE]: 00:D9:AC:00:00:00 (public-id) sync terminated reason: 62` is a typical issue that can occur due to the fact that the Bluetooth Specification only permits 6 periodic interval of missed PDUs before declaring that a sync establishment has failed; If there are more than 6 peer Periodic Advertising devices that have overlapping intervals, sync established to be failed would occur. (You may confirm by only using 5 peer advertisers whether a repeated the 5 periodic sync establishments for a long duration tests scenario do not report any failures of reason 62).

jakkra, 2022-08-25:
@cvinayak yes commit 76396496b6d9ea24a1e644565c5225222e63984f fixes the issues with degraded sync performance.

About sync term and the reason 62, I'm more talking about that the MAC address is invalid/garbage and not corresponding to any tags being tracked. The `bt_le_per_adv_sync_term_info`  `bt_le_per_adv_sync` parameters are mostly just 0's.

Thalley, 2022-08-25:
> @jakkra sorry I misunderstood, yes please create a new github issue for the invalid device address in the sync terminated.
> 
> @Thalley Is there any chance the host could be filling the invalid device address in the sync terminated callback when sync fails to be established?
> 
> Could the reference to addr that is assigned to the info and then the instance is deleted be the cause? https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/scan.c#L909-L917

Hmm, the only scenario that I can see is that if the scanning fails in `bt_le_per_adv_sync_create` and that the sync is established before the host manages to call `bt_le_per_adv_sync_delete` (which would cancel it), as the address is only set if the scanning succeeds or if the device is already scanning: 
```c
	if (!atomic_test_bit(bt_dev.flags, BT_DEV_SCANNING)) {
		err = bt_le_scan_update(true);

		if (err) {
			bt_le_per_adv_sync_delete(per_adv_sync);
			return err;
		}
	}

	*out_sync = per_adv_sync;
	bt_addr_le_copy(&per_adv_sync->addr, &param->addr);
	per_adv_sync->sid = param->sid;
```

Thalley, 2022-08-25:
Hmm, actually if the address supplied is garbage, which would probably result in sync terminated with error code 62 (`BT_HCI_ERR_CONN_FAIL_TO_ESTAB`), then that would explain it. 

Please verify that the address in the sync parameters is correctly set (potentially provide an HCI log as well).

jakkra, 2022-08-25:
@cvinayak @Thalley I think I found the issue. Since I'm using perioidic adv. list the current `get_pending_per_adv_sync()` will have a none populated le addr.
```
struct bt_le_per_adv_sync_param sync_create_param;
sync_create_param.options = BT_LE_PER_ADV_SYNC_OPT_USE_PER_ADV_LIST;
sync_create_param.skip = 0;
sync_create_param.sid = 0; // Not used but needs to be 0 otherwise error
sync_create_param.timeout = SYNC_LOST_TIMEOUT_10MS_UNIT;
memset(&sync_create_param.addr, 0xFF, sizeof(bt_addr_le_t)); // <= Added this now
err = bt_le_per_adv_sync_create(&sync_create_param, &syncForList);
```

With `memset(&sync_create_param.addr, 0xFF, sizeof(bt_addr_le_t));` part and the address in sync_term is now 0xFF .. 0xFF... Previously I did not touch the `sync_create_param.addr`, hence the random bytes.

So the problem is here https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/scan.c#L910 that 
`term_info.addr = &pending_per_adv_sync->addr;` should be populated with the address coming from the HCI event instead of `pending_per_adv_sync` =>
```
term_info.addr = &evt->adv_addr;
term_info.sid = evt->sid;
```

Thalley, 2022-08-25:
Ah, yes, that makes sense. Indeed it should use the event address (the event address should be compared to the `pending_per_adv_sync->addr` when the list is not being used. 

For using the list, I guess `pending_per_adv_sync->addr` should be set to the NONE address in case of using the list.

cvinayak, 2022-08-26:
> yes commit https://github.com/zephyrproject-rtos/zephyr/commit/76396496b6d9ea24a1e644565c5225222e63984f fixes the issues with degraded sync performance.

@jakkra Please test your application with #49579 and let me know if it solves the performance issue.

You can also check if the following improves Extended Advertising Report generation by increasing `CONFIG_BT_CTLR_SCAN_AUX_SET`:
```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_SCAN_AUX_SET=2
```

jakkra, 2022-08-26:
@cvinayak I have now tested https://github.com/zephyrproject-rtos/zephyr/pull/49579 with my sample application (attached to this issue) and it looks promising. Will test with our main application on monday also.
250ms per adv interval and 45 tags.
[image]

jakkra, 2022-08-29:
@cvinayak yes I was able to sync to around ~40/45 tags simultaniously, as you can see in the plot they come and go a bit (blue dots goes to yellow means sync was lost), but around 40 seems to be the limit for 250ms per.adv interval.

> hope you are using a supervision timeout > (250 ms* 45)
Do you mean supervision timeout for an additional BLE connection, or parameters inside of `bt_le_per_adv_sync_param`?
What I do in the sample application is to just add all tags to per. adv. list and never removing them, hence no sync timeout really used (I do this in our "real" application though).

I just wrote a long reply here https://github.com/zephyrproject-rtos/zephyr/issues/39376#issuecomment-1230088110 after testing those two, at the end I wrote about a new problem I found:

> When syncing with more than about 5 tags with per.adv interval of about 10ms then after a while (not long, like 10s) no events are generated from the controller (no scan rsp, no ext. scan rsp, no sync term, no cte etc.).

cvinayak, 2022-08-29:
> When syncing with more than about 5 tags with per.adv interval of about 10ms then after a while (not long, like 10s) no events are generated from the controller (no scan rsp, no ext. scan rsp, no sync term, no cte etc.).

Ok. Let me try to debug this and if I find something will address it in #49579

cvinayak, 2022-08-30:
@jakkra I have tested 5 tags synchronized under simulation under two conditions; one, all 5 tags have overlapping periodic advertising and all the 5 tags got synchronized; two, 5 tags with a progressive offset over time and only 4 tags got synchronized. But in both cases I was using duplicate filtering, stopping and starting scanning,  saw no issues, and periodic_sync device discovered extended advertising all along (except the one that was not synchronized). In the second case 5th tag was not discovered and this is possible if the auxiliary PDU of the 5th was overlapping any of the periodic advertising sync constantly. Keeping the periodic interval at 10 ms and extended advertising at a multiple of 10 ms would cause constant overlap between extended advertising event and periodic advertising event.

_Extended Advertising auxiliary PDU transmissions are implemented as periodic events_, a design choice to support non-overlapping multiple Extended and Periodic sets as a broadcaster. This means multiple broadcaster devices with same extended advertising interval will have overlapping auxiliary PDUs and make it difficult for a Periodic Sync device in discovering these overlapping broadcasters.

My testing yesterday was under simulation in BabbleSim. But on a target I suspect the overlapping may be too much for CPU processing between overlapping events. Something that was discovered during conformance testing of mixed multiple sets of extended advertising, periodic advertising synchronization and on multiple PHYs.

On target, the temporary solution was https://github.com/zephyrproject-rtos/zephyr/commit/a40a403a0dfc8fdf573b4ba436d51d84d68a8bd1 

There is room for improvement in how ticker module schedules closely expiring timeouts. Currently, a minimum of 3 ticks is used between consecutive ticker expiry callbacks implemented by:
https://github.com/zephyrproject-rtos/zephyr/blob/be7173a49054e34ee02d1e7ec713aabb6acc89a7/subsys/bluetooth/controller/ticker/ticker.c#L2405-L2429

Use of minimal time reservation causes many just-in-time scheduling decisions being taken and consecutive ticker expiry that consume time leading to the winning event being delayed to an amount that it cancels itself not being able to meet the hard real-time instant required to start the radio event.

To keep the just-in-time scheduling mitigation at O(1), i.e. scheduling decision take only based on time reservation, we have to not use the minimal time reservations, need #49579:

```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_SCAN_AUX_SYNC_RESERVE_MIN=n
```

Please let me know if the above Kconfig setting is acceptable by your application.

jakkra, 2022-08-30:
@cvinayak thanks for the detailed explaination.

> Keeping the periodic interval at 10 ms and extended advertising at a multiple of 10 ms would cause constant overlap between extended advertising event and periodic advertising event.

I'll make sure that extended adv. int. is not a multiple of periodic adv. interval. In the tags I have extended adv. int. set to 200ms, so in case of 10ms periodic adv. int. then what you say can occur. However regarding the issue I talked about It's the same problem at 7.5ms and up to somewhere around 10ms+.

Take a look on this log:
7.5ms per. adv interval used, 200ms extended interval.
It's ok that some tag is never seen due to reasons you describe. However the problem is that something seems to break and NO events at all are genereated from controller. Even after I power of all transmitters, no sync term event or any other event is sent. I made some comments in the log wrapped around ----'s.

```
*** Booting Zephyr OS build zephyr-v3.1.0-4071-ge64c3ad1c2fc  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
[00:00:00.004,547] <inf> main: Creating Sync for tags in list
Waiting for periodic advertising...
Scan is running...
[00:00:00.111,145] <err> main: scan_recv: per. adv. interval: 6
[00:00:00.111,389] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.272,705] <err> main: scan_recv: per. adv. interval: 6
[00:00:00.272,918] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.531,311] <err> main: scan_recv: per. adv. interval: 6
(539) PER_ADV_SYNC[0]: [DEVICE]: 33:46:8A:57:AC:25 (random) synced, Interval 0x0006 (7 ms), PHY LE 1M
[00:00:00.540,069] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:00.540,069] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(540) success. CTE receive enabled.
[00:00:00.796,844] <wrn> bt_hci_core: Malformed data
[00:00:00.834,259] <err> main: scan_recv: per. adv. interval: 6
[00:00:00.834,472] <dbg> main: scan_recv: Added tag from per adv list
[00:00:00.951,110] <err> main: scan_recv: per. adv. interval: 6
Running...
TAGS:28,sync...,sync...,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 2 Synced: 1 Num CTE with err: 0/28
[00:00:01.254,364] <err> main: scan_recv: per. adv. interval: 6
(1264) PER_ADV_SYNC[1]: [DEVICE]: 21:ED:DB:E2:70:50 (random) synced, Interval 0x0006 (7 ms), PHY LE 1M
[00:00:01.264,862] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:01.264,892] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(1265) success. CTE receive enabled.
[00:00:01.464,263] <err> main: scan_recv: per. adv. interval: 6
[00:00:01.581,085] <err> main: scan_recv: per. adv. interval: 6
[00:00:01.699,951] <wrn> bt_hci_core: Malformed data
[00:00:01.952,728] <err> main: scan_recv: per. adv. interval: 6
(1970) PER_ADV_SYNC[2]: [DEVICE]: 3D:78:84:09:C1:4D (random) synced, Interval 0x0006 (7 ms), PHY LE 1M
[00:00:01.970,306] <dbg> main: sync_cb: Removed tag from per adv list
[00:00:01.970,306] <inf> main: Creating Sync for tags in list
Enable receiving of CTE...
(1970) success. CTE receive enabled.
Scan is running...
Running...
TAGS:49,0,28,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 3 Num CTE with err: 0/77
[00:00:02.211,059] <err> main: scan_recv: per. adv. interval: 6
[00:00:02.304,260] <err> main: scan_recv: per. adv. interval: 6
[00:00:02.372,589] <err> main: scan_recv: per. adv. interval: 6
[00:00:02.631,072] <err> main: scan_recv: per. adv. interval: 6
[00:00:02.724,243] <err> main: scan_recv: per. adv. interval: 6
[00:00:03.002,563] <err> main: scan_recv: per. adv. interval: 6
Running...
TAGS:26,25,23,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 3 Num CTE with err: 0/75
[00:00:03.261,047] <err> main: scan_recv: per. adv. interval: 6
[00:00:03.354,248] <err> main: scan_recv: per. adv. interval: 6
[00:00:03.503,662] <wrn> bt_hci_core: Malformed data
[00:00:03.774,261] <err> main: scan_recv: per. adv. interval: 6
[00:00:03.842,498] <err> main: scan_recv: per. adv. interval: 6
[... 72 lines cut ...]
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
Running...
TAGS:0,0,0,0,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,
Queued: 0 Synced: 4 Num CTE with err: 0/0
```

cvinayak, 2022-08-30:
>  the problem is that something seems to break and NO events at all are genereated from controller. Even after I power of all transmitters, no sync term event or any other event is sent.

I have a hunch that there is a flaw in the implementation of all `abort_cb` that is closing the radio event that is aborted by an overlapping event. The implementation is not taking the control path that should be checking for supervision/synchronization timeout.

Let me do some investigation and get back.

jakkra, 2022-08-30:
Okey great, thanks. Just a quick question regarding
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC vs. CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL I know that the biggest difference is power consumption, but in case of periodic sync with many tags, will it make a difference? 
Is my understanding correct that if internal K32RC is used then the time radio is used will be longer (due to window widening?) hence higher power consumption, BUT this should also case the radio being "wasted", or is this time negliable ?

cvinayak, 2022-08-30:
Larger window widening = higher power consumption. Radio is listening earlier and longer to detect missing PDUs than when the clock accuracy is better. Hence, lower clock accuracy means each radio event needs higher event length to use.

What does this mean for Zephyr Controller, the time reservation is fixed (minimal by default). If a PDU arrived late in the window , overflows beyond the time reservation and with a chance to overlap with another event, then the just-in-time negotiation would get used.

Also what I mean is with low clock accuracy more radio event would slide into each other to overlap often.

closed 2022-09-01 by carlescufi

jakkra, 2022-09-01:
@cvinayak Yes sure, I'll do that, thanks.

jakkra, 2022-09-05:
@cvinayak created https://github.com/zephyrproject-rtos/zephyr/issues/49915

PR #49579 (2022-09-01): Bluetooth: Controller: Add Kconfig for minimal time reservation 
Add Kconfig to use minimal time reservation for auxiliary
and sync PDU reception. A peer device could be scheduling
multiple advertising sets in a short duration with small
PDUs hence using the minimal time reservation would avoid
skipping closely scheduled reception of multiple auxiliary
PDUs.

Add implementation in Periodic Sync LLL abort to use near
fair scheduling to ensure overlapping multiple Periodic
Sync set each get to use the radio and not lead to Sync
Loss.

The implementation is used when overlap in radio event
happen [... cut]